hoglet67/PiTubeDirect #7: 32016 Core Bug "invalid shift" running CLOCKSP

hoglet67, 2016-10-18:
Also during BAS32/CLOCKSP we get overruns and repeated characters on the screen.

Cause seems to be the logging of a 32016 Core Bug:

tube reset - copro 13
pc=00005EC5: Invalid shift of 000000DF for size 32
pc=00005EC5: Invalid shift of 000000DF for size 32
OVERRUN: A=1; D=20; RNW=1; NTUBE=0; nRST=1
OVERRUN: A=1; D=20; RNW=1; NTUBE=0; nRST=1
OVERRUN: A=1; D=20; RNW=1; NTUBE=0; nRST=1
OVERRUN: A=1; D=36; RNW=1; NTUBE=0; nRST=1
OVERRUN: A=1; D=39; RNW=1; NTUBE=0; nRST=1
OVERRUN: A=1; D=2E; RNW=1; NTUBE=0; nRST=1

BigEd, 2016-10-18:
Perhaps make all warning texts conditional, only included for a debug build?

hoglet67, 2016-10-21:
Fixed in 1dfbe624b3860be894e5b33c265c4188889bac53 by allowing +- 32 without warning.

closed 2016-10-21 by hoglet67