hoglet67/PiTubeDirect #9: Switch to hardware mailboxes for the delivery of tube mailbox message

hoglet67, 2016-10-18:
Some thoughts on this....

Goal:
- increase performance
- allowing emulators to stay in a continuous loop (without the need to poll)
- make the multi-core and GPU interactions with the emulator look the same

Still need to find the address of the mailbox hardware on the GPU side

Make this one directional - detect overruns with a 4-bit sequence counter

New ISR on emulator side would:
- listen on hardware mailbox interrupt
- call tube_io_hander() C code to update tube state as quickly as possible
- test for an exception condition (reset, tube NMI, tube IRQ)
- call emulator provided exception callback to deal with reset, NMI, IRQ

I'm not so sure about the last point. It sounds good, but the emulator would have to be written to allow the callback to be called asynchronously. Which might end up anyway with the emulator still needing to poll for interrupts.

The fast 6502 "r12/ip" trick would still be possible.

BigEd, 2016-10-18:
Would an odd/even single bit sequence counter be adequate, and easier to understand? No wrapping!

hoglet67, 2016-10-18:
I guess it depends whether we want to reliably detect loosing multiple messages.

My feeling is it because it's easy, and essentially free, if we have the space in the mailbox message, then we should use it.

The message can be up to 28 bits.

Currently we have the following 15 GPIOs:
D[7:0]
A[2:0]
nRST
nTUBE
RnW
Phi2 (aka 2MhzE)

I would add A3, and a 4-bit sequence number (in bits 12..15)

That's still leaves 8 spare bits.

hoglet67, 2016-10-21:
We have are now using the GPU->ARM hardware mailbox in all builds, as from:
8bdf0d9c162dde9ed8d6e7c7062ab65dd60a5720

BigEd, 2017-01-21:
Maybe edit the title to reflect what remaining improvements are possible?

hoglet67, 2017-04-01:
Closing, as what Dominic has done over the last few weeks renders this issue obsolete.

closed 2017-04-01 by hoglet67