hoglet67/PiTubeDirect #12: GPU code needs updating to cope with Pi 1 GPIO differences

hoglet67, 2016-10-20:
I think Dominic has a plan for this....

dp111, 2016-10-20:
I do , basicly pass into GPU land pointers to the GPIO pin map.

hoglet67, 2016-10-20:
It's specifically differences in A0, A1 and A2 we need to deal with.
i.e. constants in the following code:
   btst   r7, A0
   and    r7, (1<<A1)+(1<<A2)
   lsr    r7, 1                  # A1 shift

I guess we could use registers r3 and r4 at runtime.

These constants are known at compile time, so we could build two versions of the code, and pick (at compile time) which gets sent to the GPU.

dp111, 2016-10-20:
Yep using R3 or r4 if we need more we can point to a structure and load
registers just like the driving the databus registers databus.

I did wonder if some of the GPU constants should be in the new structure as
they currently defined in two place in the code. or we need to look at
getting the build system to include one constant file.

hoglet67, 2016-10-21:
Fixed in a950a491fd8a25aa13d4754e1ad41c74507fe9de

closed 2016-10-21 by hoglet67