hoglet67/PiTubeDirect #23: Debug application on ARM2 crashes

BigEd, 2017-02-07:
`DEBUG HELLOC`
or 
`DEBUG HELLOC`
fails with
`Entering Supervisor because of branch through 0
R15 = 1F0016F8
`

As [reported on stardot](http://stardot.org.uk/forums/viewtopic.php?p=145999#p145999):
(But note that I'm using versions of these programs from a different disk image than the one attached. It's possible the error is mine.)

> The debugger of ARM Evaluation system doesn't work with B-em or BeebEm emulators.  I mentioned it in the other topics...
> 
> I am attaching a disk image.  It is a copy of disc 3 of AES with several other files:
> TEST - a program in assembler (6 lines of the plain text) to print a character 3 on the screen;
> TESTC - the previous program compiled;
> HELLO - a program in assembler (6 lines of the plain text) to print a "hello world" message;
> HELLOC - the previous program compiled ;
> PI-ARM1 - a compiled program which gives the digits of number pi.
> Just mount this disc.  Try 
> TYPE TEST
> TESTC
> TYPE HELLO
> HELLOC
> PI-ARM1
> All should work.  According to ARM Utilities Reference Manual the debugger invocation has syntax DEBUG A-PROGRAM-FILE.
> Try
> DEBUG TESTC
> DEBUG HELLOC
> DEBUG PI-ARM1
> All cannot work. :(
>

BigEd, 2017-02-07:
I just tried on the Matchbox and got
`Modula-2 exception - Bad filing system for use with ADFSDir`
and a memory dump. That's at least reasonable behaviour, but shows up that I'm using MMFS.

hoglet67, 2017-02-07:
Do we know this works on an original Arm Eval System?

hoglet67, 2017-02-07:
More useful data points in this thread:
http://stardot.org.uk/forums/viewtopic.php?p=147163#p147163

BigEd, 2017-02-07:
We can probably get our hands on the AES at CCH again, in May, if we still need to at that time. I didn't have a chance to do much with it last time, and that's the only AES I can think of getting my hands on.

dp111, 2017-02-07:
I'm planning to be along in May too.

hoglet67, 2017-02-13:
On the matchbox, using MMFS, I get the expected error:
A* DEBUG HELLOC
Modula-2 exception - Bad filing system for use with ADFSDir

using ADFS, I get:
A* DEBUG HELLOC
Modula-2 exception - HELLOC - ObjectFile Corrupt

So it looks lilke this expects an "Acorn ARM Object Format" executable:
http://www.chiark.greenend.org.uk/~theom/riscos/docs/CodeStds/AOF-1989.txt

I wonder how we get a hello world in this format?

BigEd, 2017-02-13:
Must be a linker question...
So, the codepath which fails on the Pi must be the filing system check.

hoglet67, 2017-02-13:
There is a tool on disk2 that came with the ARM Eval System called OBJASM that produces AOF output.

Unfortunately, I can't find this disk in a .adl format, just as a zip file, and I'm struggling to find a route to load the files onto my system without loosing there attributes.

The nearest I have come is to be able to view them in ADFS explorer from the hard disk archive:
http://acorn.huininga.nl/pub/disks/originals/Acorn/ARM%20Evaluation%20Kit/scsi1.dat
which originated here:
http://www.g7jjf.com/acornArm_disc_images.htm

hoglet67, 2017-02-13:
Maybe the best plan is to run OBJASM in B-Em as this includes copies of the ARM Eval system disks in .ADL format. Then just move the executable over.

What I would like to do eventually is to include the ARM Eval files in my standard ADFS hard disk image. But I just can't find the tools to do that at the moment.

hoglet67, 2017-02-13:
Documentation for the ARM DEBUG command in chapter 5 of:
http://chrisacorns.computinghistory.org.uk/docs/Acorn/OEM/AcornOEM_ARMUtilitiesRM.pdf

hoglet67, 2017-02-18:
There are some recent bug fixes to the ARM core in MAME since we took a copy:
https://github.com/mamedev/mame/commits/master/src/devices/cpu/arm/arm.cpp

Specifically, it's these 4 commits by sronco:
https://github.com/mamedev/mame/commit/6184bba010924c145ceb086cc918a8d77e46583e
https://github.com/mamedev/mame/commit/e0c4c04f0c6ac0f9e360bf932ba0746dfdb8ed48
https://github.com/mamedev/mame/commit/4cb5712cd2c86826064f0394070e8022e61a64a0
https://github.com/mamedev/mame/commit/cda99ff0701b2d9e07662fc7f552c72d558618a4

It's worth is resyncing with the last code, which I will try to do today..

hoglet67, 2017-02-18:
It was the third of the fixes that resolved this issue, specifically this one line:
-			if (pCarry) *pCarry = rm & SIGN_BIT;
+			if (pCarry) *pCarry = rm & (1 << (k - 1));

This affects the carry resulting from ROR.

commit 4cc3f335 (2017-02-18): Co Pro ARM: Ported sronco fix 4cb5712 from MAME (fixes #23)

arm: fixed carry for ROR and preserve IRQ_MASK in LDR R15,***

Change-Id: If188a0f4c27a9ad85c28b14ca477030a17f07f76
Note: the carry for ROR part of this fixes issue #23
closed 2017-02-18 by hoglet67