hoglet67/PiTubeDirect #24: ARM client should handle ROM-like header

BigEd, 2017-02-08:
See http://stardot.org.uk/forums/viewtopic.php?p=160379#p160379

jgharston, 2017-02-08:
I've posted appropriate code to StarDot, and emailed Hoglet a revised version of tube-swi.c. I don't know how to add the code to GitHub, plus don't have any way of compiling it to check it works before doing so.

fordp2002, 2017-02-09:
You can compile the code very easily on Linux or on Bash on Windows 10. The instructions are on the wiki here. If you do not have either of the above it is still possible to build the code but just takes a bit more effort.

jgharston, 2017-02-09:
Simon Ellwood wrote:
> You can compile the code very easily on Linux or on Bash on Windows

What I meant was here on my laptop, I;'m not at home at the mo. I should 
be able
to do it when I get home with my main PC.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

fordp2002, 2017-02-09:
I just started using Bash for Windows and was really pleased with how easy it is now to compile projects like this. I am not sure I would ever bother installing a Windows Cross Compiler again! The Linux Desktop has arrived ;)

hoglet67, 2017-02-13:
FYI, I just tried building the version of tube-swi.c that Jonathan PM'd me, and I getting some build errors:

/home/dmb/atom/PiTubeDirect/src/tube-swi.c:338:7: error: 'eblk' undeclared (first use in this function)
       eblk->errorMsg[0]=249;
       ^
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:338:7: note: each undeclared identifier is reported only once for each function it appears in
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:343:7: error: 'reg' undeclared (first use in this function)
       reg[0]=eblk->errorMsg;
       ^
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:355:9: warning: assignment makes integer from pointer without a cast [enabled by default]
       r0=address+off;         // R0=>module title
         ^
/home/dmb/atom/PiTubeDirect/src/tube-swi.c: In function 'tube_Mouse':
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:695:3: warning: implicit declaration of function 'tube_byte' [-Wimplicit-function-declaration]
   tube_byte(reg);      // ADVAL(7)

Jonathan, are you in a position where you can test these changes yourself yet?

jgharston, 2017-02-13:
David Banks wrote:
> Jonathan, are you in a position where you can test these changes
> yourself yet?

Not before Wednesday evening. I'm away from home again, currently in 
N'cstl.
A brief look through the errors suggests its because I wasn't sure how
to refer to the ARM registers structures, though the ADVAL(7) looks like
a typo - should it be calling tube_osbyte() ?

SYS OS_Mouse should do:
R0=ADVAL(7), R1=ADVAL(8), R2=ADVAL(9), performed by doing
R0=R1 result from OSBYTE 128,7,0
R1=R1 result from OSBYTE 128,8,0
R2=R1 result from OSBYTE 128,9,0
where OSBYTE has been updated to ensure R1 on exit is X+256*Y.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

jgharston, 2017-02-13:
David Banks wrote:
>  /home/dmb/atom/PiTubeDirect/src/tube-swi.c: In function 'tube_Mouse':
>  /home/dmb/atom/PiTubeDirect/src/tube-swi.c:695:3: warning: implicit
> declaration of function 'tube_byte' [-Wimplicit-function-declaration]
>  tube_byte(reg); // ADVAL(7)

Ah, tube_byte() should be tube_Byte() - upper case 'B'.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

hoglet67, 2017-02-14:
Jonathan,

If you like I can have a go at manually merging these changes in today. I'll probably break them into separate commits.

Dave

jgharston, 2017-02-14:
David Banks wrote:
> If you like I can have a go at manually merging these changes in
> today. I'll probably break them into separate commits.

The Z80 update should be ok, as it's identical to the working BeebEm 
update.
OS_Mouse should be ok with the tube_byte() tube_Byte() typo fixed
The correction to the OSWORD block transfer should be ok

The user_exec_raw() update could be cut down to:

//
   if (address[19]==0 && address[23]==0 && address[27]==0) {
       // RISC OS module header
       off=address[16]+256*address[17]+65536*address[18];
       r0=address+off;			// R0=>module title
     } else {
       // No header, r0 should point to *command used to run code
       r0 = (unsigned int) env->commandBuffer;
     }

   if (address[3]==0) {
     off=address[0]+256*address[1]+65536*address[2];
     address=address+off;		// Entry word is offset, not branch
   }
//

which is the minimum to correctly enter correctly formed ARM code. It's 
generating
an error for non-ARM code where I haven't got the parameter passing 
correct.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

hoglet67, 2017-02-14:
Jonathan, all of your changes to tube-swi have been merged in (as multiple separate commits)
https://github.com/hoglet67/PiTubeDirect/commits/boa-dev

I had to re-factor the error handing slightly to be able to generate an error. I tested this by trying to run execute an image of the BASIC2 ROM, and got "This is not ARM code".

I tested the module handling by changing the exec address of BAS135 to be the same as the load address, and this seemed to load correctly. 

Thanks for your fixes; please can we have any further ones as separate git pull requests (one per individual fix is ideal).

hoglet67, 2017-02-14:
Jonathan,
PS Just added your Z80 changes as well (4856b929)

jgharston, 2017-02-15:
David Banks wrote:
> I had to re-factor the error handing slightly to be able to generate
> an error. I tested this by trying to run execute an image of the
> BASIC2 ROM, and got "This is not ARM code".

That was the bit I was having difficulty working out how to structure 
the
code to get the functionality. I'm back at home now, so I'm going to see
if I can get a build setup set up.

I also really need to get my Pi set up - I took it to the last ABUG
intending to do it there with people around to give hand-on advice,
but never got around to it.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

jgharston, 2017-02-15:
I've noticed a cut'n'paste typo:
  // RISC OS module header
       off=address[16] + 256 * address[1] + 65536 * address[2];
       r0 = (unsigned int) address + off; // R0=>module title

should, of course, say:
       off=address[16] + 256 * address[17] + 65536 * address[18];

So, my task for this evening, is to get a build environment working on 
this PC,
fork the repository (is that the right verb?) insert that one-line 
correction
and see if I can put it back into the repository.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

fordp2002, 2017-02-16:
I have some spare interface boards if you need one let me knwo.

hoglet67, 2017-02-18:
Reopening as it sounds like there is an issue to fix here (see Jonathan's comment two up).

What I don't understand is why this mistake is not causing a crash when I try to run BAS135 (with exec address == load address).

Jonathan, how can we better test this?

hoglet67, 2017-02-18:
The only weirdness I'm seeing is first time after power up:
```
arm>*DIN 15
arm>*XXX
ARM BBC BASIC V

Starting with 2060540 bytes free.

Not found
>
```
(XXX is a copy of BAS135 with exec = load)

If I hit break and re-run XXX, you don't see the error.

I get this even if add the fix: off=address[16] + 256 * address[17] + 65536 * address[18];

Is there any way to tell from within BBC Basic what command/params it thinks it was invoked with?

jgharston, 2017-02-19:
David Banks wrote:
> The only weirdness I'm seeing is first time after power up:
> arm>*DIN 15
> arm>*XXX
> ARM BBC BASIC V
> 
> Starting with 2060540 bytes free.
> 
> Not found
>> 

You need to implement OS_GetEnv and return the parameters in R0. What 
you are seeing is BASIC calling OS_GetEnv, OS_GetEnv returning R0 
pointing to some random memory, and then BASIC attempting to CHAIN $R0, 
and so getting a 'File not found' error from the filing system.

Tho', I thought OS_GetEnv had been implemented. I'll check tomorrow 
after I've had some sleep.

> Is there any way to tell from within BBC Basic what command/params it
> thinks it was invoked with?

SYS "OS_GetEnv" TO A$:PRINT A$

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

hoglet67, 2017-02-19:
It does look like there is a bug in SYS "OS_GetEnv". I've created a new issue for this: #35

jgharston, 2017-02-19:
I had a ponder while drifting off to sleep.

I'm not used to a tube client written in C, so I'm still navigating 
around it to familiarise myself with it. The following needs to happen 
when code is loaded and executed:

OSCLI needs to set up an execution environment, including a local 
'command line parameters' pointer, this needs to point to:

(skip any stars or spaces)(skip any non-spaces)(skip any spaces)(HERE)
or
(skip any stars or spaces)/(skip any spaces)(skip any non-spaces)(skip 
any spaces)(HERE)
or
(skip any stars or spaces)RUN(skip any spaces)(skip any non-spaces)(skip 
any spaces)(HERE)

eg:
*  ***  *  foo.blah.jim   boo<cr>
..........................^
   ***   ***foo.blah.jim  <cr>
.........................^
**   *** /   blah.foo<cr>
.....................^
*   ***   */foo.bah  blim blam foo <cr>
.....................^
** *RUN   jim.fred.hazel  sophie.wilson  <cr>
..........................^

Where 'RUN' is case insensative, and could be abbreviated to 'RU.' or 
'R.'

OS_GetEnv then needs to return that local command line pointer in R0.

EXECUTE needs to enter code with R0 pointing to the command that caused 
the execute. This isn't the parameters, it's the whole command. In the 
case of a module it is just the module name, with no parameters. 
Executing code has to collect the command parameters by calling 
OS_GetEnv. The updated EXECUTE does look like it's entering code with 
the registers correctly set up, it looks like the problem is between 
OSCLI preparing the command line pointer and OS_GetEnv collecting it.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

hoglet67, 2017-02-19:
Jonathan, could you edit those examples in the previous post into a code block (the <> button in the tool bar) as they don't line up properly for me with a variable width font.

Thanks

Dave

jgharston, 2017-02-19:
It's in non-monospaced because it's an email reply, and it won't let me edit it. I have to reply here.

OSCLI needs to set up an execution environment, including a local  'command line parameters' pointer, which is collected later by EXECUTE, this needs to point to, eg:

<code>*** * foo.blah.jim   boo<cr></code>
<code>......^</code>

<code>*** ***foo.blah.jim  <cr></code>
<code>.......^</code>

<code>** *** / blah.foo<cr></code>
<code>.........^</code>

<code>* *** */foo.bah  blim blam foo <cr></code>
<code>........^</code>

<code>** *RUN jim.fred.hazel  sophie.wilson  <cr></code>
<code>........^</code>

Crumbs, this message editor is a bit mankey.

hoglet67, 2017-02-24:
Fixed by commit 0cafad7de (at least, the weirdness I was previously seeing that resulted in a Now Found error is no longer present)

commit d9a2c474 (2017-02-14): JGH/DMB: tube-swi: Updated code entry to check code header (fixes #24)

Change-Id: I2742bd6b98de6452abafa21082f6514907e9f781
closed 2017-02-24 by hoglet67