hoglet67/PiTubeDirect #30: Check unaligned write_x64

fordp2002, 2017-02-15:
The 64 bit read has been modified as the ARM does not support reading 64 bits non-aligned in one go.

The 64 bit write has not been modified however.

Can the Raspberry Pis support writing 64 bits non-aligned?

If not the write_x64 can cause a crash.

hoglet67, 2017-02-15:
Most of the documentation I can find states that LDRD/STRD does nor support unaligned accesses, e.g.:
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka15414.html

What's confusing me is the write_x64 crash only happens on a Pi 3, it works fine on a Pi Zero. I need to compare the generated code to see if both use STRD.

fordp2002, 2017-02-16:
I am guessing this is a duplicate of the issue 28 then. I was just looking on my phone while travelling back for Amsterdam. Still it is better a bug is logged twice rather than not at all.

dp111, 2017-02-16:
As a quick fix undefine NS_FAST_RAM

hoglet67, 2017-02-16:
Fixed this against ticket #28 

Ford, you were indeed correct that ARM doesn't allow unaligned 64-bit accessed (STRD/LDRD)

The reason it was working on the Zero was that the compiler code generation was different, and only used 32-bit stores.

closed 2017-02-16 by hoglet67