hoglet67/PiTubeDirect #34: 6809 Basic hangs on Escape

BigEd, 2017-02-19:
confirmed in latest boa-dev
-   git version: cd01f82

(Even though this change landed in Anaconda:
Synced 6809 Tube ROM with the lastest from JGH
)

hoglet67, 2017-02-19:
If I remember correctly, this was a bug in the Flex distribution, not the client ROM.

Yes, fixed here:
http://www.stardot.org.uk/forums/viewtopic.php?p=158937#p158937

Is that the Flex image you are using?

BigEd, 2017-02-19:
Ah, my mistake, surely. I've made no effort to update my Flex, it will a copy of yours.
I'll close the ticket.

closed 2017-02-19 by BigEd

hoglet67, 2017-02-19:
Still doesn't work for me on either the Matchbox or PiTubeDirect.

Instead of hanging, escape is just ignored.

But probably not our bug.....