hoglet67/PiTubeDirect #35: SYS OS_GetEnv to A$ does not return invocation string

hoglet67, 2017-02-19:
SYS "OS_GetEnv" TO A$:PRINT A$ should return the command line used to invoke Basic.

It doesn't.

Here's the current implementation:
https://github.com/hoglet67/PiTubeDirect/blob/boa-dev/src/tube-swi.c#L673

Dave

hoglet67, 2017-02-19:
It's possible SYS is broken due to the lack of SVC &39 (SWINUMBERFROMSTRING).

I need to try SYS &10 TO A$:PRINT A$ instead.

Dave

jgharston, 2017-02-19:
I seem to have successfully made a fork, but can't get it compiling at the moment. Anyway.

In my fork I've updated tube-swi.c tube_cli() to prepare the command line buffer for later collection by execute to pass to R0 and R1. At the moment it doesn't swap in the module title for module code as that requires a static buffer somewhere for the command line buffer, as on real hardware. Currently commandBuffer is just set pointing to whatever command string whereever in memory has been passed to OSCLI. That needs copying to a buffer in workspace so it can be manipulated (and able to be changed by OS_SetEnv).

hoglet67, 2017-02-20:
Jonathan. A couple of things.

Firstly, all development work is happening on a branch called boa-dev, whereas you have just pushed your changes to master. This is fine on your fork - you can do anything you like on your fork as you own it. But I won't be able to merge this directly.

Second, most of the changes already exist in boa-dev. You can see them at the top of this list:
https://github.com/hoglet67/PiTubeDirect/commits/d9a2c47416608002c94e879776f9b2f971e7aada

What I can do is to take a look at your commit and pull in the relevant bits that don't already exist.

If you want stuff to be easily mergeable, here are some tips:
1. Work on the development branch (currently boa-dev) not master
2. Always change you have the latest code (git fetch followed by git rebase should do this)
3. Several small separate commits are better than one big commit

fordp2002, 2017-02-20:
Can we add these comments to the wiki. I think you mean "check" not "change" in item 2.

I am still a git novice too, but I am using it professionally now so that will change. Git is a bit of a steep learning curve but a revelation once you have the skills I think.

hoglet67, 2017-02-20:
Thanks Simon, typo above fixed.

Ed is working on release notes (on thw wiki), currently in draft form here:
https://github.com/hoglet67/PiTubeDirect/wiki/Known-Issues-and-Future-work

Those should eventually include links to all known issues with the release.

jgharston, 2017-02-20:
David Banks wrote:
> If you want stuff to be easily mergeable, here are some tips:
>  	* Work on the development branch (currently boa-dev) not master
>  	* Always change you have the latest code (git fetch followed by git
> rebase should do this)
>  	* Several small separate commits are better than one big commit

I thought that was what I was doing. I went to 'boa-dev' and selected 
'fork'; and seeing that tubeswi.c had OS_Mouse and OS_Plot in it 
concluded I'd got things right.

I then just looked at the one issue of tube_CLI setting up commandBuffer 
to pass to exec_raw, deliberately leaving until later the issue that the 
command buffer should be a modifiable static data block somewhere.

I'll check again later after having something to eat.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

BigEd, 2017-02-20:
Your github fork will have all branches - you'll need to select the branch you want to explore and edit, if you're editing in github.

If you're using a git client to work on your own machine, you'll need something like

`git checkout boa-dev
`
to get you on the right branch.

jgharston, 2017-02-20:
Ok, I've tried again, and it now says "This branch is 1 commit ahead of hoglet67:boa-dev." as a result of submitting the tube-swi.c I worked on yesterday.

BigEd, 2017-02-20:
(If you're ahead, and you want your change adopted, you go the destination, the parent, and issue a 'pull request' - there's a button for that, just to the right of the select-a-branch pulldown.)

hoglet67, 2017-02-21:
Jonathan,

FYI, I just gave your change a quick try, and these are the compiler errors:
```
/home/dmb/atom/PiTubeDirect/src/tube-swi.c: In function 'user_exec_raw':
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:399:10: error: invalid type argument of unary '*' (have 'int')
   while (*r1 > ' ') r1++;    // Step past command
          ^
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:400:10: error: invalid type argument of unary '*' (have 'int')
   while (*r1 == ' ') r1++;    // Step past spaces, r1=>command tail
          ^
/home/dmb/atom/PiTubeDirect/src/tube-swi.c: In function 'tube_CLI':
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:540:22: error: incompatible types when assigning to type 'char[256]' from type 'char *'
   env->commandBuffer = lptr;    // Parameters for this *command
                      ^
/home/dmb/atom/PiTubeDirect/src/tube-swi.c: In function 'tube_Byte':
/home/dmb/atom/PiTubeDirect/src/tube-swi.c:583:30: error: incompatible types when assigning to type 'char[256]' from type 'char *'
           env->commandBuffer = "\x0D";  // Null command line
                              ^
make[2]: *** [CMakeFiles/tube-client.dir/tube-swi.c.obj] Error 1
make[1]: *** [CMakeFiles/tube-client.dir/all] Error 2
make: *** [all] Error 2

```
What's hold you up getting the compile environment set up? Let me know if there's anything I can help with...

jgharston, 2017-02-22:
David Banks wrote:
> What's hold you up getting the compile environment set up? Let me know
> if there's anything I can help with...

I'm not sure, I'm still fiddling with it.

I initially got loads of errors:
...src\ns32016\32016.c(5) : Cannot open include file: 'stdint.h':
...src\armc-cstubs.c(63) : Cannot open include file: 'sys/times.h':
...

I remember I initially got this when first working on BeebEm, having not 
specified the include paths. I added the include paths, but now I'm 
getting:

...\include\inttypes.h(260) : error C2054: expected '(' to follow 
'__inline__'
...\include\inttypes.h(261) : error C2082: redefinition of formal 
parameter 'imaxabs'
...\include\inttypes.h(261) : error C2143: syntax error : missing ';' 
before '{'
1
...

so it looks like I'm using the wrong includes. But they're the only ones 
I've got.

-- 
J.G.Harston - jgh@mdfs.net - mdfs.net/jgh

hoglet67, 2017-02-22:
What OS are you using? The only ones known to work are:
- Windows 10 with the bash shell
- Ubuntu 14.04 or later

fordp2002, 2017-02-22:
I used the Linaro Windows cross-compiler in the for the PiTubeSPI early days and that worked OK.
I am using Windows 10 Bash (Ubuntu Cross Compiler) these days.

BigEd, 2017-02-22:
Just to note: Ubu 14.04 (trusty) has version 4.8.2 of the package gcc-arm-none-eabi
Back in May 2016, we were using gcc-arm-none-eabi-4_7-2014q2 from https://launchpad.net/gcc-arm-embedded

jgharston, 2017-02-24:
I'm using Windows XP on my desktop and Windows 7 on my laptop. I've got to the stage where I can do a test compile of an isolated .c file with a dummy include of externals, so that lets get types and syntax correct, I seem to be missing (or missing the settings for) the correct include headers.

I've been following through http://gnutoolchains.com/raspberry/tutorial/  I'll check BigEd's link just above.

I'm proceeding carefully as I don't want to discovere I've inadvertantly killed my ability to build BeebEm. :( It may end up with me arranging to drag my PC to an ABUG meeting to get hand-on help.

hoglet67, 2021-04-08:
SYS &10 TO A$ does now work.

SYS OS_GetEnv doesn't due to the lack of SVC &39 (SWINUMBERFROMSTRING).

This is a seperate issue (#117)

So closing this issue now.

closed 2021-04-08 by hoglet67