hoglet67/PiTubeDirect #42: 80286 - suspicious findings on code review

BigEd, 2017-02-22:
dp111 has been looking through the code and noted some suspicious findings. In the absence of tests, and in preference to lots of new tickets, I though we could perhaps list them here. Or, we can delete this and raise new tickets.

1. DAA probably wrong "as we aren't using the old value of AL or CF"
1. pushf "appears to set the overflow flag"
1. We don't need to support the 286 instruction set, so is this code we need or want?
   ```
      case 0xB: /* 0B OR Gv Ev */
        modregrm()
        ;
        oper1 = getreg16(reg);
        oper2 = readrm16(rm);
        op_or16();
        if ((oper1 == 0xF802) && (oper2 == 0xF802))
        {
          sf = 0; /* cheap hack to make Wolf 3D think we're a 286 so it plays */
        }

        putreg16(reg, res16);
      break;
   ```

BigEd, 2017-02-22:
> @dp111 comments:
> could you have a quick test with the following for pushf :
> ```
> case 0x9C: /* 9C PUSHF */
> push(makeflagsword() | 0x0000);
> break;
> ```

The [current code](https://github.com/hoglet67/PiTubeDirect/blob/58ddff72b4666829cd424fb61c878abacf1caaf8/src/cpu80186/cpu80186.c#L2814) looks like this:
```
        case 0x9C: /* 9C PUSHF */
        push(makeflagsword() | 0xF800);
        break;
```

BigEd, 2017-02-22:
Looks to me like F800 is indeed an error - [reference](https://en.wikipedia.org/wiki/FLAGS_register) - should be F000 for 186 or 0000 for 286.

fordp2002, 2017-02-22:
Probably the wrong time but I considered https://github.com/adriancable/8086tiny in the early days but I guess it would have to be extended to be a 80186.

hoglet67, 2017-02-22:
Ed, I agree with the value F000 for a 186.

BTW, the client ROM uses the following test to distinguish a 186 and a 286 for the purpose of deciding whether to use NMI:
```
PUSH SP
POP  AX                 ; Push SP and pop it to AX
CMP  AX,SP
JNZ  L0A30              ; If popped value different, running on 80186
```

BigEd, 2017-02-22:
(See also #43 for this question. I think there's a simple fix.)

hoglet67, 2017-02-23:
I'm just working through the items raised in this issue, and I'm a bit stuck on DAA.

Looking at the operation, e.g. as described here:
http://www.felixcloutier.com/x86/DAA.html

```
         old_AL ← AL;
         old_CF ← CF;
         CF ← 0;
         IF (((AL AND 0FH) > 9) or AF = 1)
               THEN
                     AL ← AL + 6;
                     CF ← old_CF or (Carry from AL ← AL + 6);   //// **** ///
                     AF ← 1;
               ELSE
                     AF ← 0;
         FI;
         IF ((old_AL > 99H) or (old_CF = 1))
              THEN
                     AL ← AL + 60H;
                     CF ← 1;
              ELSE
                     CF ← 0;
         FI;
```
What I'm confused about is how the line I've marked above actually has an effect on anything, as later CF is always overwritten.

Or is it implied that it's used in AL <= AL+ 60H?

hoglet67, 2017-02-23:
For reference, here is B-Em's implementation, which never seems to actually clear CF:
```
                        case 0x27: /*DAA*/
                        if ((flags&A_FLAG) || ((AL&0xF)>9))
                        {
                                tempi=((uint16_t)AL)+6;
                                AL+=6;
                                flags|=A_FLAG;
                                if (tempi&0x100) flags|=C_FLAG;
                        }
                        if ((flags&C_FLAG) || (AL>0x9F))
                        {
                                AL+=0x60;
                                flags|=C_FLAG;
                        }
                        setznp8(AL);
                        tubecycles-=4;
                        break;
```

BigEd, 2017-02-23:
If ZEXALL tests these instructions, and if the Z80 behaves the same with DAA (and SBB), can we crib from a model which passes ZEXALL?

BigEd, 2017-02-23:
There's a simple English explanation at
http://z80-heaven.wikidot.com/instructions-set:daa

I think CF can never be cleared if it was set going in, but on the other hand, it seems you can always tell when it should be set.

hoglet67, 2017-02-23:
It would be nice to write an exhaustive test suite for the. There are only 2^10 combinations.

I might just have a got at writing one in BBC Basic, and then someone can run it on a real 80x86 Co Pro

hoglet67, 2017-02-23:
Here's a small test program:
```
   10 REM DAA TEST SUITE
   20 @%=&902
   30 DIM code% 256
   40 P%=code%
   50 [OPT 0
   60 DAA
   70 PUSHF
   80 POP DX
   90 RETF
  100 ]
  110 FOR cf=0 TO 1
  120   FOR af=0 TO 1
  130     FOR al=&00 TO &FF
  140       A%=al
  150       F%=cf+&10*af
  160       R%=USR(code%)
  170       alout=R% AND &FF
  180       cfout=(R% DIV &10000) AND 1
  190       afout=(R% DIV &100000) AND 1
  200       PRINT "cf=";cf;" af=";af;" al=",~al;" : ";
  210       PRINT "cfout=";cfout;" afout=";afout;" alout=",~alout 
  220     NEXT
  230   NEXT
  240 NEXT
```
Resuts for the current (incorrect we think) code attached:
[DAA.txt](https://github.com/hoglet67/PiTubeDirect/files/796426/DAA.txt)

dp111, 2017-02-23:
Might be worth trying on the FPGA as well.

hoglet67, 2017-02-23:
I don't think my program is correctly outputting the flags....Looking into that now.

hoglet67, 2017-02-23:
Results from the Matchbox:
[DAAMB.txt](https://github.com/hoglet67/PiTubeDirect/files/796480/DAAMB.txt)
Some interesting differences, and not just in the flags.

fordp2002, 2017-02-23:
This is the output form a zet core maybe ask on stardot for somebody to run it on a real copro?

fordp2002, 2017-02-23:
cfout is true when the first nibble is at least 10 or when the first nibble is 9 and the lower nibble is at least 10.

hoglet67, 2017-02-23:
Here is what I hope is the correct output, obtained from running on BBC Basic for Windows.
[DAAWIN.txt](https://github.com/hoglet67/PiTubeDirect/files/796702/DAAWIN.txt)
The matchbox is almost correct, but there are 6 cases where the carry flag is set when it shouldn't be:
```
cf=0 af=1 al=  0 : cfout=1 afout=1 alout=  6
cf=0 af=1 al=  1 : cfout=1 afout=1 alout=  7
cf=0 af=1 al=  2 : cfout=1 afout=1 alout=  8
cf=0 af=1 al=  3 : cfout=1 afout=1 alout=  9
cf=0 af=1 al=  4 : cfout=1 afout=1 alout=  A
cf=0 af=1 al=  5 : cfout=1 afout=1 alout=  B
```
I'll have a go now at the code in PiTubeDirect

hoglet67, 2017-02-23:
The fix I've just checked in gives identical result to when I run on a genuine x86 (BBC Basic on my windows laptop). Hopefully this will also match the behaviour of a real 186!

The B-Em code was actually 100% correct!

Our original fake86 code was nearly correct. The only mistake was ever actively clearing cf or af; that never seems to happen in reality.

Pretty much every detailed description of DAA on the internet is incorrect. Which is a bit scary!

fordp2002, 2017-02-23:
Maybe you can fix the Zet core too and ping Zeus to let him know?

fordp2002, 2017-02-23:
Yes I forgot to consider carry ins in my code above.

hoglet67, 2017-02-23:
Mame actually gets it correct as well:
https://github.com/mamedev/mame/blob/master/src/devices/cpu/i86/i86inline.h#L946

hoglet67, 2017-02-23:
Ford, good luck to figuring out how DAA is implemented in Zet :-)
https://github.com/hoglet67/CoPro6502/blob/master/src/zet/zet/zet_conv.v

fordp2002, 2017-02-23:
I will give it a go, then I will give up and contact zeus ;)

hoglet67, 2017-02-23:
I made the same fix to DAS, expecting it to just work.

I'm not getting quite the same results as I get from BBC Basic Windows:
[DASWIN.TXT](https://github.com/hoglet67/PiTubeDirect/files/797014/DASWIN.TXT)
[DASFIX1.txt](https://github.com/hoglet67/PiTubeDirect/files/797013/DASFIX1.txt)

I really need the results from a genuine 80186 Co Pro, as it's possible the behaviour changed on later Intel CPU (I'm assuming the only legal inputs are valid BCD numbers, is that correct?)

BigEd, 2017-02-23:
I'd only worry about valid BCD (on the 6502 Bruce figured out both that case and the full case)

I'd very much suppose the x86 has stayed unchanged from 8080 to today, wrt to these instructions.

Great idea to use BBC Basic with its assembler!  I imagine we can do the same with SBC on the ns32k, one day.

hoglet67, 2017-02-23:
> I'd only worry about valid BCD (on the 6502 Bruce figured out both that case and the full case)

Ed, that was my initial thought. But then I realised the input to DAA is the binary sum of two valid BCD numbers, which I think means that pretty much any input value for AL is valid. For example 0xAA can be produced by 0x99 + 0x11 both of which are valid BCD.

Now, I think some flag combinations are probably illegal. For example, the above 0x99 + 0x11 would generate cf=0 and af=1. Now, are these the only legal flag values for 0xAA?

BigEd, 2017-02-23:
You are right of course, DAA's inputs are from an ADD or ADC. But, I would limit it to the case of correcting for the addition of two legal BCD inputs to the addition. (If that makes any difference.)

I was, as ever, coming from 6502 land, where decimal addition is a fusion of the addition and the correction.

hoglet67, 2017-02-23:
I'm still surprised how many emulators get DAA wrong, evem with valid inputs.

Consider the following case:
0x77 + 0x88 = 0xFF resulting in cf=0 and af=0
apply DAA should give:
0x65 with cf=1 and af=1

which is indeed what we currently get:
cf=0 af=0 al= FF : cfout=1 afout=1 alout= 65

The following code, from SIMH of all places, I think gets this wrong:
https://github.com/simh/simh/blob/master/IBMPC-Systems/common/i8088.c#L964
```
                if (((AL & 0xF) > 9) || GET_FLAG(AF)) {
                    AL += 6;
                    SET_FLAG(AF);
                }
                if ((AL > 0x9F) || GET_FLAG(CF)) {
                    AL += 0x60;
                    SET_FLAG(CF);
                }
                break;
```
I think the output here would be 0x05 with cf=0 and af=1

The corner case seems to be not modelling a carry generated in the first stage of the DAA instruction.

Or maybe I've been staring at this for too long!

dp111, 2017-02-23:
When we get real 186 results that will be great. If you are at a loose end
you could code up the intel algorithm and feed it into your test bench.

hoglet67, 2017-02-23:
So you have a reference to "the intel algorithm"?

BigEd, 2017-02-23:
I wouldn't expect the AF after a DAA to be important - is it specified? Either way, using BBC Basic on a modern x86 should give us the info we need - that would be confirmed in the next week or so if a stardot member takes up the challenge.

hoglet67, 2017-02-23:
Ed, in my SIMH example above it's the carry flag they get wrong!

DAS is proving a bit more tricky!

BigEd, 2017-02-23:
Have you checked a 6502 emulator? They are doing the same thing, but fused with the add, and if they pass Klaus' testsuite, they are correct. Subtraction is indeed messier.

hoglet67, 2017-02-23:
OK, here's the Intel reference Dominic mentioned earlier in an email:
https://courses.engr.illinois.edu/cs426/Resources/64-ia-32-architectures-software-developer-instruction-set-reference-manual-325383.pdf#page=288

```
IF 64-Bit Mode
THEN
#UD;
ELSE
old_AL ← AL;
old_CF ← CF;
CF ← 0;
IF (((AL AND 0FH) > 9) or AF = 1)
 THEN
 AL ← AL + 6;
 CF ← old_CF or (Carry from AL ← AL + 6);
 AF ← 1;
 ELSE
 AF ← 0;
FI;
IF ((old_AL > 99H) or (old_CF = 1))
 THEN
 AL ← AL + 60H;
 CF ← 1;
ELSE
 CF ← 0;
FI;
FI;
```
It would be interested to simulate this, as I think the <= 0 assignments are at best not needed.

hoglet67, 2017-02-23:
Here's a quick and dirty simulation, and it gives the same results as on real hardware, and also the same as we currently have in PiTubeDirect. Which is re-assuring.
```
#include <stdio.h>

int main() {
  int cf;
  int af;
  int al;

  for (cf = 0; cf <= 1; cf++) {

    for (af = 0; af <= 1; af++) {

      for (al = 0x00; al <= 0xff; al++) {

        int CF = cf;
        int AF = af;
        int AL = al;

        int old_AL = AL;
        int old_CF = CF;

        CF = 0;
        if (((AL & 0x0F) > 9) || (AF == 1)) {
          AL = AL + 6;
          CF = old_CF;
          if (AL & 0x100) {
            CF = 1;
          }
          AF = 1;
        } else {
          AF = 0;
        }
        if ((old_AL > 0x99) || (old_CF == 1)) {
          AL = AL + 0x60;
          CF = 1;
        } else {
          CF = 0;
        }

        AL &= 0xFF;

        printf("cf=%d af=%d al= %2X : cfout=%d afout=%d alout= %2X\n", cf, af, al, CF, AF, AL);

      }
    }
  }
}
```
As does this simpler version which matches our current code:
```
#include <stdio.h>

int main() {
  int cf;
  int af;
  int al;

  for (cf = 0; cf <= 1; cf++) {

    for (af = 0; af <= 1; af++) {

      for (al = 0x00; al <= 0xff; al++) {

        int CF = cf;
        int AF = af;
        int AL = al;

        if (((AL & 0x0F) > 9) || (AF == 1)) {
          AL = AL + 6;
          if (AL & 0x100) {
            CF = 1;
          }
          AF = 1;
        }
        if ((AL > 0x9F) || (CF == 1)) {
          AL = AL + 0x60;
          CF = 1;
        }

        AL &= 0xFF;

        printf("cf=%d af=%d al= %2X : cfout=%d afout=%d alout= %2X\n", cf, af, al, CF, AF, AL);

      }
    }
  }
}
```

fordp2002, 2017-02-23:
I logged it on the Zet github

hoglet67, 2017-02-24:
I've reworked DAA and DAS to use common code (which is likely what the hardware does).

I've verified both of these and I'm now seeing identical results:
- in BBC Basic on Windows (on my laptop)
- in a simulation of the original intel algorithms
- in PiTubeDirect with the above commit

For reference, here are my final set of experiments:

[intel_simulation.zip](https://github.com/hoglet67/PiTubeDirect/files/799372/intel_simulation.zip)

[DAAWIN.txt](https://github.com/hoglet67/PiTubeDirect/files/799362/DAAWIN.txt)
[DAAINT.txt](https://github.com/hoglet67/PiTubeDirect/files/799361/DAAINT.txt)
[DAAFIX3.txt](https://github.com/hoglet67/PiTubeDirect/files/799360/DAAFIX3.txt)

[DASWIN.txt](https://github.com/hoglet67/PiTubeDirect/files/799364/DASWIN.txt)
[DASINT.txt](https://github.com/hoglet67/PiTubeDirect/files/799363/DASINT.txt)
[DASFIX3.txt](https://github.com/hoglet67/PiTubeDirect/files/799365/DASFIX3.txt)

I don't think there are any remaining things in this issue, so I'll close.

In the future, I think it would be best to open separate issues for each defect.

closed 2017-02-24 by hoglet67

BigEd, 2017-02-24:
(Retitled the issue.)

hoglet67, 2017-02-24:
Phil Young has posted the results from a real 186 Co Pro:
http://www.stardot.org.uk/forums/viewtopic.php?p=161842#p161842
[DAA186.txt](https://github.com/hoglet67/PiTubeDirect/files/799973/DAA186.txt)
[DAS186.txt](https://github.com/hoglet67/PiTubeDirect/files/799974/DAS186.txt)
Rather surprisingly, there are actually some differences from later x86 models, and from the current Intel documentation:

DAA (186 on the right)
```
cf=0 af=1 al= 9A : cfout=1 afout=1 alout=  0  |	cf=0 af=1 al= 9A : cfout=0 afout=1 alout= A0
cf=0 af=1 al= 9B : cfout=1 afout=1 alout=  1  |	cf=0 af=1 al= 9B : cfout=0 afout=1 alout= A1
cf=0 af=1 al= 9C : cfout=1 afout=1 alout=  2  |	cf=0 af=1 al= 9C : cfout=0 afout=1 alout= A2
cf=0 af=1 al= 9D : cfout=1 afout=1 alout=  3  |	cf=0 af=1 al= 9D : cfout=0 afout=1 alout= A3
cf=0 af=1 al= 9E : cfout=1 afout=1 alout=  4  |	cf=0 af=1 al= 9E : cfout=0 afout=1 alout= A4
cf=0 af=1 al= 9F : cfout=1 afout=1 alout=  5  |	cf=0 af=1 al= 9F : cfout=0 afout=1 alout= A5
```
DAS (186 on the right)
```
cf=0 af=1 al=  0 : cfout=1 afout=1 alout= FA  |	cf=0 af=1 al=  0 : cfout=0 afout=1 alout= FA
cf=0 af=1 al=  1 : cfout=1 afout=1 alout= FB  |	cf=0 af=1 al=  1 : cfout=0 afout=1 alout= FB
cf=0 af=1 al=  2 : cfout=1 afout=1 alout= FC  |	cf=0 af=1 al=  2 : cfout=0 afout=1 alout= FC
cf=0 af=1 al=  3 : cfout=1 afout=1 alout= FD  |	cf=0 af=1 al=  3 : cfout=0 afout=1 alout= FD
cf=0 af=1 al=  4 : cfout=1 afout=1 alout= FE  |	cf=0 af=1 al=  4 : cfout=0 afout=1 alout= FE
cf=0 af=1 al=  5 : cfout=1 afout=1 alout= FF  |	cf=0 af=1 al=  5 : cfout=0 afout=1 alout= FF
cf=0 af=1 al= 9A : cfout=1 afout=1 alout= 34  |	cf=0 af=1 al= 9A : cfout=0 afout=1 alout= 94
cf=0 af=1 al= 9B : cfout=1 afout=1 alout= 35  |	cf=0 af=1 al= 9B : cfout=0 afout=1 alout= 95
cf=0 af=1 al= 9C : cfout=1 afout=1 alout= 36  |	cf=0 af=1 al= 9C : cfout=0 afout=1 alout= 96
cf=0 af=1 al= 9D : cfout=1 afout=1 alout= 37  |	cf=0 af=1 al= 9D : cfout=0 afout=1 alout= 97
cf=0 af=1 al= 9E : cfout=1 afout=1 alout= 38  |	cf=0 af=1 al= 9E : cfout=0 afout=1 alout= 98
cf=0 af=1 al= 9F : cfout=1 afout=1 alout= 39  |	cf=0 af=1 al= 9F : cfout=0 afout=1 alout= 99
```
Now, we need to decide if these cases actually matter.

hoglet67, 2017-02-24:
Just thinking about the DAA case:
cf=0 af=1 al= 9A..9F
I don't think this should arise in practice through the addition of two BCD numbers. It is possible to find numbers that add up to 9A (e.g. 99 and 01) but none of these will have the half carry (af) set after addition.

I'm struggling to do the same analysis for DAS.

hoglet67, 2017-02-24:
Now thinking about the DAS cases:

I don't think the first block cf=0 af=1 al=  0..5 can arise.

I'm less sure about the second block: cf=0 af=1 al= 9A..9F

e.g. 0x33 - 0x99 gives 0x9A with af=1 but I think cf=1 so we're probably OK.

dp111, 2017-02-24:
My view is that we should emulate as best we can the original hardware not
what the latest x86 hardware does. The issue should be well commented.

dp111, 2017-02-24:
More intel documentation

http://soft.laogu.com/download/80c186.pdf

dp111, 2017-02-24:
DAA intrustruction pseudo code from the 186 datasheet

if ((AL) and 0FH) > 9 or (AF) = 1 then
         (AL) ← (AL) + 6
         (AF) ← 1
if (AL) > 9FH or (CF) = 1 then
         (AL) ← (AL) + 60H
         (CF) ← 1

Which is different to what we have

dp111, 2017-02-24:
DAS code :

if ((AL) and 0FH) > 9 or (AF) = 1 then
     (AL) ← (AL) – 6
     (AF) ← 1
if (AL) > 9FH or (CF) = 1 then
     (AL) ← (AL) – 60H
     (CF) ← 1