hoglet67/PiTubeDirect #43: 80286 - propose consistent renaming as 80186

BigEd, 2017-02-22:
We emulate a 186, which is accurate for emulating the Master 512, but because of a quirk in the ROM and the handling of tube transfers, it is detected and reported as a 286. We've been referring to 286 in talking about it.

It might be better to tweak the ROM we use, accurately report as a 186, and consistently refer to it as a 186. We just need to arrange to use NMI for data transfers - assuming this is something which only the ROM needs to know about.

> The client ROM was coded to support both a 186 and a 286, the difference being:
> - the 186 uses DMA for data transfers
> - the 286 uses NMI for data transfers
> 
> Now, NMIs were much easier to work with than having to implement (or emulate) the 186's DMA hardware. So we need the CPU to be detected as a 286 (which depends on a some subtlety of how a certain thing that escapes me is pushed onto the stack).
> 
> But no 286 instructions are ever used, and all master 512 software is 186 compatible.

jgharston, 2017-02-22:
It would more accurately reflect the real world if the ROM remained the same, with the ability to detect what CPU it is running on, and we actually emulate an 80186 if we want to be emulating a 80186 and emulate an 80286 if we want to be emulating an 80286. If we are emulating an enviroment where we are using the 80286's NMI transfer capability, then it is correct for the environment to declare that it is running on an 80286 as it is running on an 80286, as with real world hardware.

If we bodged the Client ROM to declare we were running on a 80186 but we were actually running an 80286 with NMI transfers we would be lying to the user.

jgharston, 2017-02-22:
However, I'd have no objection to renaming the files and bits of the project itself to something like cpu80x86, etc. instead of cpu80186, etc., to reflect that it's not just a 80186.

BigEd, 2017-02-22:
It's not accurate to call the model a 286, as it doesn't have the extra instructions. I see that you feel it's not accurate to call it a 186, if it uses NMI.

BigEd, 2017-02-22:
So, we do have a bug, in that the ROM detects 286 and announces 286, but we wouldn't be able to run 286 code. So, we shouldn't announce it, and our emulation should act as a 186 because that's the code we can execute.

It looks like a very small patch to client ROM would sort this, just NOP out the JNZ after the first check, which would then fall through to the NMI setup. The second check we leave alone, so it will detect and report 186 - as would any application software.

hoglet67, 2017-02-22:
Ed, unfortunately the processor detection is not restricted to the Client ROM. The DOS Plus XIOS also performs such a test, as you can see in the new disassembly here:
http://www.seasip.info/Cpm/software/dpgen.zip

While we could patch the client ROM, we can't patch DOS Plus!

We also do exactly the same in the Matchbox, to avoid implementing DMA and all the other 186 peripherals.

I think what we have is the lesser of two evils, especially as there is no known Master 512 software that is 80286 only.

BigEd, 2017-02-23:
Hmm. I suppose the only clean way out is to have a core model which has the extra 286 instructions and behaviour. We could probably make a note somewhere that this is where we are.

hoglet67, 2017-02-23:
As far as I understand, the additional 286 instructions all relate to protected mode and the on-chip MMU. It seems unlikely that any application software would use these. The only alternative is to implement the 186 on-chip peripherals. There's a list of those that are architecturally visible here:
http://www.cpu-world.com/Arch/80186.html

> Control registers are used to control 80186 integrated peripherals. These 16-bit registers are part of 256-byte control block that can be mapped into system RAM or I/O space. The control block includes the following registers:
> - Interrupt control - 20h - 3Eh
> - Time 0 control - 50h (counter), 52h (max count A), 54h (max count B), and 56h (control word)
> - Time 1 control - 58h (counter), 5Ah (max count A), 5Ch (max count B), and 5Eh (control word)
> - Time 2 control - 60h (counter), 62h (max count A), and 66h (control word)
> - Chip-select control - A0h (upper memory), A2h (lower memory), A4h (peripheral), A6h and A8h (mid-range memory)
> = DMA descriptors channel 0 - C0h and C2h (source pointer), C4h and C6h (destination pointer), C8h (transfer count) and CAh (control word).
> - DMA descriptors channel 1 - D0h and D2h (source pointer), D4h and D6h (destination pointer), D8h (transfer count) and DAh (control word).
> - Refresh Control Unit registers (80C186 in Enhanced mode only) - E0h - E4h
> - Power-Saving control (80C186 only) - F0h
> - Relocation register - FEh
> After reset the control block is mapped at address FF00h in I/O space.

fordp2002, 2017-02-23:
I would suggest just placing a comment block at the top explaining what it is and what it is not. There is little to be gained in changing the code. i.e. This is a emulation of 80286 real mode ONLY.

BigEd, 2017-02-23:
Good thought, real mode 286 might be about right.

BigEd, 2017-03-17:
Closing this - not clear-cut and not worth worrying about

closed 2017-03-17 by BigEd

hoglet67, 2023-03-18:
FYI, we now appear as an 80186 using DMI rather than an 80286 using NMI.

33bb8a68d

https://stardot.org.uk/forums/viewtopic.php?f=2&t=26540

Dave