hoglet67/PiTubeDirect #49: NS32K: Reset behaviour differs from real Co Pro

hoglet67, 2017-03-12:
On a real Co Pro, if you have loaded BAS32 and then hit soft break, you end up back at Basic's > prompt.

On PiTubeDirect you end up back at Pandora's * prompt.

i.e. we have lost the concept of a current language that persists across soft breaks.

I suspect this is down to:
```
void n32016_reset()
{
   n32016_reset_addr(0xF00000);
}
```
In the real  hardware, the reset address is 0x0, and the ROM is temporarily paged in over the top of this.

fordp2002, 2017-03-12:
Is this not more down to clearing the RAM?

fordp2002, 2017-03-12:
It would be pretty easy to put a more faithful emulation back in. The change as far as I remember just makes the ROM run the first set of instructions at a higher location before it jumps to the location the ROM resides normally anyway.

fordp2002, 2017-03-14:
Is this not more down to clearing RAM on reset?

hoglet67, 2017-03-14:
I don't think it's as simple as that. Looking at the code, we only clear memory once, on power on reset, not on every subsequent reset:
https://github.com/hoglet67/PiTubeDirect/blob/master/src/copro-32016.c#L22

I'll have a look at the Pandora ROM and see how it maintains the current language.

Actually, I've the only real 32016 Co Pro I have used is your one Simon. That has a different ROM version (1.00 vs 2.00 in PiTubeDirect). So it's possible it's just a difference in the ROMs. I could try 1.00 in PiTubeDirect and see what happens.

fordp2002, 2017-03-14:
If you look at the two instructions that are run at a different address I cannot see how that would break anything!

hoglet67, 2017-03-14:
At the moment neither can I.

hoglet67, 2017-03-14:
I've done a bit of debugging, with JHG's disassembly to hand:
http://mdfs.net/Software/Tube/32016/v200/Pan200.zip

This bug centres around the notion of the current program address, which (on a 1024KB system) is stored at:
```
 00FFDE0 |&24(SB) Current program address
```
After hard break, this has the value 00F00DAE, which is the supervisor prompt loop.

When *BAS32 is executed, what should happen at somepoint before basic is actually entered is the current program address should be updated with the execution address of BAS32.

This is not happening; even after BAS32 is running, if you inspect 00FFDE0 it still has the value 00F00DAE.

This must be one of two things:
- a bug in Pandora 2.00
- a bug in our 32016 emulation

I think I will try with Pandora 1.00, just to make sure...

hoglet67, 2017-03-14:
Just tried Pandora 1.00 and the bug is not present there.

This is either a bug in Pandora 2.00, or a core bug that only affects Pandora 2.00.

I'm going to "upgrade" the real 32016 to Pandora 2.00 and see what happens.

hoglet67, 2017-03-14:
The real 32016 is also now upgraded to Pandora 2.00, and the bug is present.

So this is a bug in Pandora 2.00, not in PiTubeDirect.

I'll see if Jonathan has any input on this.

fordp2002, 2017-03-14:
Nice to know my old Copro is getting some use and I a pleased the issue seems to be Acorn's and not mine!

hoglet67, 2017-03-14:
Yes, it's incredibly useful at times like this to have access to real hardware. Otherwise the afternoon would have been spent on a wild good chase!

closed 2017-04-01 by hoglet67