hoglet67/PiTubeDirect #56: Native ARM only offers 2MByte of RAM

BigEd, 2017-10-29:
If I'm not mistaken, the Native ARM model offers just 2M, whereas the emulated ARM2 offers a historically credible 4M and Sprow's ARM7TDMI board was sold with up to 32M and supported up to 64M.  Is there room for a bump in RAM for Native ARM?

hoglet67, 2017-10-29:
From a hardware perspective, pretty much all physical memory on the Pi should, in principle, available to the Native Pi Co Pro.

At the moment the build system compiles/loads the "kernel" (i.e. all of PiTubeDirect) to 0x1f00000 (31MB) and the various stacks sit just below this. Each stack is 1MB, and there can be quite a few of them:
https://github.com/hoglet67/PiTubeDirect/blob/master/src/armc-start.S#L32

So the current system layout gives ~20MB for any of the Co Pros to play with,

The Native ARM Co Pro, rather arbitrarily, sets the memory limit to 2MB here:
https://github.com/hoglet67/PiTubeDirect/blob/1f8cc361beff676650cc5a9bdd88816efea3b2cf/src/copro-armnative.c#L195  

There is no reason for this to be so low and it could be increased to 16MB without risk of bumping into the stacks.

hoglet67, 2017-12-05:
It's now 16MB in Diamondback RC2

hoglet67, 2017-12-05:
Closing until someone asks for more!

closed 2017-12-05 by hoglet67