hoglet67/PiTubeDirect #59: Boot time degradation

hoglet67, 2018-07-24:
There is anecdotal evidence that the PiTubeDirect boot time has degraded recently, such the the Master no longer recognises the Co Pro on power up.

To try to quantify this, I'm running this program on the Beeb:
```
10 P%=&2000
20[
30 LDA &FEE0
40 JMP &2000
50]
60 CALL &2000
```
I've got a scope hooked up to the Pi's reset switch, and also to GPIO Pin 24 (D0). This provides a clear indication of how long after reset is released it takes for the Pi to start responding to read requests from the Beeb.

My master probe's the tube 918ms after power on (measured from VCC reaching 4V).

All the SD Cards are freshly formatted with the SD Card Association formatter

I'll start using my Sandisk Ultra 8GB Class 10 Card and a Pi Zero v1.3:
- 1156/1024ms - Anaconda
- 1160/1023ms - Boa
- 924/798ms - Cobra
- 930/800ms - Diamondback
- 1084/952ms - Egg Eater rc0

The first number is the boot time when the SD Card has just been inserted. The second number is the boot time on subsequent reboot cycles.

Note, none of these are recognised on first power up of the Master.

To understand if the SD Card makes a difference, here is the boot time of Diamondback with three different cards:
- 930/800ms -  Sandisk Ultra 8GB Class 10 (as above)
- 732/576ms - OV 8GB Class 6
- 812/744ms - Sandisk 4GB Class 4

Out of these cards, the Sandisk Class 10 is not recognised on Boot in my Master, where as the other two are. 

Staying with the OV Card, and using the Diamondback release, try different firmware blobs
- 732/576ms - firmware blob released with Diamondback 
- 880/726ms - firmware blob released with EggEater
- 882/728ms - current firmware blob, as of 24/7/2018

Staying with the OV Card, and using the EggEater release, try different firmware blobs
- 734/584ms - firmware blob released with Diamondback 
- 884/732ms - firmware blob released with EggEater
- 886/732ms - current firmware blob, as of 24/7/2018

So this demonstrates a Firmware Blob change between Diamondback and EggEater has slowed boot down by ~150ms.

Finally, comparing with a Pi Zero and a Pi Zero W:

Pi Zero:
- 732/576ms - original firmware blob released with Diamondback 
- 882/728ms - current firmware blob, as of 24/7/2018

Pi Zero W:
- 782/584ms - original firmware blob released with Diamondback
- 1030/787ms -  current firmware blob, as of 24/7/2018
 
Conclusions:
1. The firmware blob changes post Diamondback have slowed boot by ~150ms
2. The Pi Zero W is slower to boot that the Pi Zero, by 50-150ms (depending on the firmware)
3. The sum of (1) and (2) causes my Master to not see the Co Pro on boot
4.The  PiTubeDirect kernel has little effect on Boot time between Diamondback and EggEater

Next step is to bisect the firmware blob to see which commit added 150ms.

dp111, 2018-07-24:
Could you also measure the SDCARD clock frequency, between blobs?

hoglet67, 2018-07-24:
Firmware Blob included in Diamonback Is:
20/5/2016 - https://github.com/raspberrypi/firmware/commit/ac0c0bff
fastest boot time: 576ms

Firmware Blob included in EggEater is:
21/3/2018 - https://github.com/raspberrypi/firmware/commit/145ed363
fastest boot time: 726ms

22 months difference!

30/12/2016 - https://github.com/raspberrypi/firmware/commit/4aac0c55
fastest boot time: 582ms

4/1/2017 - https://github.com/raspberrypi/firmware/commit/87edb826
fastest boot time: 584ms

10/1/2017 - https://github.com/raspberrypi/firmware/commit/e44f66a9e
fastest boot time: 721ms

12/1/2017 - https://github.com/raspberrypi/firmware/commit/6df93defc
fastest boot time: 721ms

15/2/2017 - https://github.com/raspberrypi/firmware/commit/3fb63c413
fastest boot time: 719ms

21/3/2017 - https://github.com/raspberrypi/firmware/commit/2ef9cb622
fastest boot time: 719ms

8/6/2017 - https://github.com/raspberrypi/firmware/commit/ca9116ff
fastest boot time: 719ms

1/12/2017 - https://github.com/raspberrypi/firmware/commit/9426e18b
fastest boot time: 724ms

So the commit that is responsible for the slowdown is from 12/1/2017.
```
commit e44f66a9e4e41636634501a093f9a55ceefb46d6
Author: popcornmix <popcornmix@gmail.com>
Date:   Tue Jan 10 19:29:51 2017 +0000
    firmware: Add usb/net support from next branch
```
https://github.com/raspberrypi/firmware/commit/e44f66a9e

Add usb/net support from next branch

This allows booting from USB Mass Storage:
https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/

Unfortunately, this predates adding Pi Zero W support:
```
commit f5446ba0b3bbd4241bf0ad35a2c9f299f5c447d5
Author: popcornmix <popcornmix@gmail.com>
Date:   Thu Mar 2 15:38:37 2017 +0000
    firmware: platform: Treat Pi0W like Pi0 w.r.t. clocks and voltages    
    firmware: platform: Add final Pi Zero W support
```
https://github.com/raspberrypi/firmware/commit/f5446ba0

So it's not possible to avoid this, unless there is some magic to put in config.txt to disable USB Boot which seems unlikely.

hoglet67, 2018-07-24:
> Could you also measure the SDCARD clock frequency, between blobs?

Not easily, it's almost impossible to probe the SD Card interface on a Pi Zero.

Would using a Pi One be OK?

dp111, 2018-07-24:
> 20/5/2016 - raspberrypi/firmware@ac0c0bf
> 21/3/2018 - raspberrypi/firmware@145ed36
> 30/12/2016 - raspberrypi/firmware@4aac0c5
> 4/1/2017 - raspberrypi/firmware@87edb82
> 10/1/2017 - raspberrypi/firmware@e44f66a
> 12/1/2017 - raspberrypi/firmware@6df93de
> 15/2/2017 - raspberrypi/firmware@3fb63c4
> 21/3/2017 - raspberrypi/firmware@2ef9cb6
> 8/6/2017 - raspberrypi/firmware@ca9116f
> 1/12/2017 - raspberrypi/firmware@9426e18
> raspberrypi/firmware@e44f66a
> raspberrypi/firmware@f5446ba

I have raised something similar with the RPI team
https://github.com/raspberrypi/firmware/issues/702

BigEd, 2018-08-21:
Oops, I may have muddied the water over there, but there are some ideas and some questions in that issue.

philpem, 2020-05-12:
I've noticed this on my Master 128 a few times. Is there a spare GPIO which could be used to pull the Master's reset line low while the PiTube boots?

Seems like that'd be a quick fix (admittedly requiring a hardware tweak) which would remove the dependency on the Pi team fixing this bug.

dp111, 2020-05-12:
unfortunately that reset line is an output from the Master and so we can't
drive it.

philpem, 2020-05-12:
I was thinking more along the lines of adding a transistor to the PiTube and an optional wire to the Master's reset circuit. The aforementioned wire could terminate in a clip-lead, allowing it to be connected without solder.

hoglet67, 2021-04-08:
Closing as this is mostly out of out control.

The Pi 4 boot time is a few seconds!

closed 2021-04-08 by hoglet67