hoglet67/PiTubeDirect #60: Working with Rapsberry Pi 3A+

TomDDG, 2018-12-15:
More of a comment than an issue but it will require some changes to the release files. I've managed to get the coprocessor working on a 3A+, running same speeds as a 3B+ but with the size advantage. All I had to do was download the latest start.elf, bootcode.bin & fixup.dat from the raspberry pi site, rename them to match and it works fine.

hoglet67, 2018-12-16:
Tom,

Thanks for the bug report.

I've updated the firmware to the latest firmware, and pushed a new release (Egg-eater-rc1):
https://github.com/hoglet67/PiTubeDirect/releases

It would be great if someone with a Pi 3A+ could validate this works correctly out of the box.

I'll leave this issue open pending that feedback.

Dave

TomDDG, 2018-12-16:
Just tested the new release with my Pi 3A+ and all working fine

hoglet67, 2018-12-16:
Many thanks Tom.

closed 2018-12-16 by hoglet67