hoglet67/PiTubeDirect #61: 6502 and 65C102 boot strings and modes

simoninns, 2019-04-02:
As a suggestion for a feature enhancement, it would be good if there was some way of configuring the PiTubeDirect to either an external Acorn wedge 6502 or an internal Master 65C102 board (especially showing the correct string on boot so the Master with internal 65C102 looks authentic with "Acorn TUBE 65C102 Co-Processor").

I'm not sure how it's done at the moment, but it looks like both CoPro mode 0 and 1 are 65C102 and there is no option for pure 6502 support.  I realize it's not a big issue, but it might be nice for compatibility to disable 65C102 extensions in 6502 mode.

Making the copros separate would also allow a more 'natural' configuration of 3 or 4MHz speeds depending on the mode (The Domesday AIV system requires a 65C102 running at 4MHz, otherwise the mouse pointer doesn't move correctly).

hoglet67, 2019-04-03:
There never was a vanilla 6502 Second Processor. 

The Acorn 6502 Second Processor (cheese wedge) typically contains a 3MHz R65C02.

The Master Internal 6502 Co Pro typically contains a 4MHz R65C102.

From an application software perspective, these are identical.

Do you know how the Master Turbo (65C102) Tube ROM differs from the original cheese wedge ROM? 

Other than the boot message, are there any code changes.

I can check on JGH's site.

simoninns, 2019-04-03:
I'm told (by JGH) that the only real difference in the ROMs is the boot string and the *HELP string.  It still would be nice (for completion) to be able to choose between them simple for a more 'authentic' experience :)  

I'm not sure if there are functional differences between the 65C102 and 65C02; I will dig up the datasheet and take a look.  I know there are physical differences, but it's probably just a hardware thing.

It would be a 'nice to have' if there was a 65C102 mode with the correct boot string running at a default of 4MHz - especially for Domesday, as the original software exhibits mouse pointer issues with a Pi CoPro running at >4Mhz speeds.

dp111, 2019-04-03:
In config.txt you can select the variable speed 6502 core and set the speed
to 4MHz.

I have no objection to swapping to the 65C102 rom but I see no reason to
support both roms. As it is only textual and possibly osword something to
access shadow ram.

Dominic

simoninns, 2019-04-04:
I have it set up in the way you suggest @dp111 and it works fine, except that the boot up and help strings are incorrect.  I realise that functionally it doesn't matter which ROM image you're using, but I would counter that there are two ROMs and they are different - so the emulation would be more accurate if it supported both.  Generally it should be the case that the host can't tell the difference between emulation or real hardware... with a Pi CoPro on the internal tube of the Master (with a 65C02 boot string), you could simply check the ROM and know it's a Pi since it would not be possible to connect a real 65C02 second processor that way.  Swapping one ROM for the other would probably just make BBC model B users ask the same question (of why the boot string is incorrect) :)

Again, it's just a suggestion.  With multiple ROM support you could even add one for the high-speed 65C102/65C02 copro and have it report the speed as part of the boot string; that would make it obvious what setting you have (and that it is not directly compatible with the 'real thing' unlike the 3/4Mhz modes).

fordp2002, 2019-04-04:
I could be wrong but I was under the impression a lot more effort has been put in to making the fastest 6502 CoPro in the world. I remember when I was in Egypt on holiday the fantastic gains in speed Dave and Dominic made seemingly hour by hour!

The slower versions are not cycle accurate?

I am pretty sure this could be done however. This is an open source project so I guess forking and contributions are usually encouraged!

Unlike some people who took their projects closed source  on Monday ;)

simoninns, 2019-04-04:
I'm not sure why Domesday doesn't work correctly with the high-speed copro emulation; probably something timing related I guess @fordp2002 (you'd have to ask Logica ;) )

Of course I can make the changes myself; but, as I stated before, it's just a suggestion for improvement rather than a 'must have now!' thing (I'm busy writing GPL code for the Domesday86 project or I would have looked into it myself).

Speed is all very nice, but accuracy of emulation is important too imho.  One does not preclude the other.

As for all my projects; I can assure you they all remain GPLv3, even my April fools post on FB was CC licensed as far as I was concerned :)

julianregel, 2019-09-04:
I'd echo this request. Having the ability to easily and clearly switch between a 3Mhz 6502 second processor and the 4Mhz 65C102 co-processor, with different boot strings/code by specifying a different PiTube co-pro, would be very useful.

hoglet67, 2019-09-04:
Julian,

Thanks for your comment.

We have some ideas about how to do this, but it's important we understand exactly what users actually want.

Could you explain abit more about why you would find this useful?

For example, is it the speed difference thay you care about? Or the accuracy of the boot message?
Or some other difference?

Dave

julianregel, 2019-09-04:
Hi Dave

For me, it's a combination of authenticity and simplicity.

Authenticity in that if I'm emulating a Master Turbo, I want to see the correct boot string and for it to run at the correct 4Mhz, and simplicity in that switching between that and a 3Mhz second processor is as easy as issuing an FX command and pressing CTRL-Break to select the correct processor.

As the second processor and the co-processor are different devices,  it seems logical (to me) that they would be separate Pi co-processors.

Thanks for the consideration.

Julian

dp111, 2019-09-05:
*fx 151,226,0 followed by *fx 151,228, (speed in MHz)
will set the speed of the variable speed 6502

simoninns, 2019-09-06:
> As the second processor and the co-processor are different devices, it seems logical (to me) that they would be separate Pi co-processors.

This is something I really agree with.  There are two different ROMs, the fact they are functionally equivalent isn't really the point, they are still different and therefore supporting both is more 'correct' than just one. 

From a end-user perspective being able to select between BBC model B second processor and Master co-processor would be great - variable super-fast versions are a separate option (so I would advocate 4 modes; Beeb original, Master original, Beeb speed-you-like and Master speed-you-like) - in each case selecting the right ROM image for the choice.

hoglet67, 2019-11-26:
Done some work on this in bramch fer-de-lance-dev:

We now have:
0 - "ACORN TUBE 6502 64K" @ 280MHz
1 - "ACORN TUBE 6502 64K" @ 3MHz
2 - "ACORN TUBE 65C102 Co Processor" @ 280MHz
3 - "ACORN TUBE 65C102 Co Processor" @ 4MHz

The old lib65c02 Co Pro is now at (16), and the OPC 5/6/7 Co Pros are now at (20,21,22)

The ROM part of this may be subsumed by future work on loading client ROMs off the SD Card.

julianregel, 2019-11-27:
Sounds great! Thanks for adding this functionality. I'll look forward to trying it out.

closed 2020-01-21 by hoglet67