hoglet67/PiTubeDirect #62: Enhancment: PDP11 - Emulate a 4010 terminal via VDU?

afarlie, 2019-05-05:
For reference : https://en.wikipedia.org/wiki/Tektronix_4010

This was a widely used terminal on the PDP11 and other mincomputer systems of that era.

It was also coincidentally one of the terminal emulations provided in Termulator, which means it should in theory be possible to provide some kind of translation in the Pi Tube code?

BigEd, 2019-06-30:
I see Termulator is a ROM for the Beeb. Can that ROM be activated on the host and then a second processor used? That might do the trick without any change to PiTubeDirect.

Indeed, it feels to me that doing this emulation layer is a job for the host. It's possible for a client program running on the second processor to upload code to the host and hook it into the VDU vector, and that might well be a good way to do this, assuming there's enough RAM to do it, or sideways RAM.

Once the VDU vector is hooked, any subsequent program would be able to use 4010 commands to control the display.

jgharston, 2021-02-21:
I agree, I/O functionality a job for the I/O system. Something like the STEM terminal in conjunction with the VideoNULA would give you a colour VT terminal. Years ago I wrote a serial terminal that hooked onto WRCHV to provide monochrome ANSI controls.

hoglet67, 2021-04-08:
Hognose will include a full-featured BBC VDU driver for the Pi's HDMI interface.

So a terminal emulator could be done in software, and would output either to the Beeb display or the Pi display.

That terminal emulator is out of scope for PiTubeDirect, so closing this issue.

closed 2021-04-08 by hoglet67