hoglet67/PiTubeDirect #63: Software reset (bit 5 of FEE0) no longer level sensitive

hoglet67, 2019-05-16:
Setting bit 5 of FEE0 should hold the Co Pro in the reset state. It used to, but that's no longer the case.

This is the commit the broke things: aa6a4a13

This doesn't cause an issue on the Beeb, because the MOS never uses this bit. But the Atom Tube Host code does, and this particular bit of code works with Boa, but fails with Egg-Eater:
https://github.com/hoglet67/AtomTubeHost/blob/master/src/AtomHost.asm#L190

hoglet67, 2019-05-26:
Hi Dominic,

I wonder if you have any thoughts in this issue.

Reverting the change to tube-ula.c does resolve it.

It's easy to work around in the AtomTubeHost, but it does seem a genuine bug in PiTubeDirect compared to the way the Tube ULA works.

Dave

BigEd, 2020-09-13:
(Just checking: should this still be open?)

hoglet67, 2020-09-13:
Yes, it's not a biggie, but it's not actually been fixed yet.

dp111, 2023-09-09:
Fixed in indigo-alpha2

closed 2023-09-09 by dp111