hoglet67/PiTubeDirect #64: 80286 co-processor won't load DOS

jamie-alnasir, 2019-07-23:
Hi there,

Congratulations on this project. I've found it amazing. However, I am having trouble getting any of the co-processors to load.

For the 80286 when I issue the FX 151,230,8 command and reboot, it takes several attempts to recognise, then finally shows 80286. Then, when I use the @dos command, nothing happens, it just hangs.

I've also tried Shift-Break to boot from GOTEK DOS disk images, again without any luck.

I'd be most grateful for any help or pointers as to what I'm doing wrong, or what the issue is.

Many thanks,
Jamie

hoglet67, 2019-07-23:
Can you summarise your Beeb setup?

Is it a Model B or a Master?

What version of ADFS are you using?

jamie-alnasir, 2019-07-23:
I have DFS 2.2 (1770) which is a Solidisk interface card - tested working with GOTEK (Flash Floppy) and tested with 5 1/4" floppy drive.

hoglet67, 2019-07-23:
Ah, I think that's the problem. I'm pretty sure the the DOS Plus disk images need ADFS to run, as they are ADFS format.

jamie-alnasir, 2019-07-23:
So the *dos command runs an image on the SD card? and that's stored in memory, but the BBC ROM needs to be ADFS?

hoglet67, 2019-07-23:
It's a complicated boot sequence.

*DOS is built in to the 80x86 Client ROM in the Co Pro

It then issues the command *FADFS to switch to ADFS, and then searches for ADFS floppy and hard disks that contain the DOS Plus boot loader. This is a file called DOSBOOT.

So yes, the BBC needs to have the ADFS file system fitted.

jamie-alnasir, 2019-07-23:
Great, thanks for the explanation. I now understand! :-)

Keep up the fantastic work!

closed 2019-07-23 by hoglet67