hoglet67/PiTubeDirect #66: Core 1 - 3MHz 6502 not loading BASIC

RetroClinic, 2019-10-05:
Hi.

I've just had someone point out that Core 1 isn't loading BASIC on a 3A or 3B with Egg Eater? It was working fine on a Zero with Diamondback. I've tested it on a Model B and he's right, it just gets stuck after BASIC and doesn't return a prompt. Core 0 is working fine.

Is this an issue that's known about, or just not been picked up on?

Thanks, Mark.

hoglet67, 2019-10-05:
I'm not aware of this issue.

What happens with Core 1, Egg Eater, Pi Zero?

Which 3A / 3B model are you using:
https://elinux.org/RPi_HardwareHistory

RetroClinic, 2019-10-05:
Hi.

I’ve not tried Egg Eater with it on a Zero, I can do some more testing when I’ve got this Model B repair off the bench, but with Diamondback it works fine (on a Zero).

The boards I’m using are 3A+ and 3B+.

Thanks, Mark.

From: David Banks 
Sent: Saturday, October 05, 2019 8:06 PM
To: hoglet67/PiTubeDirect 
Cc: RetroClinic ; Author 
Subject: Re: [hoglet67/PiTubeDirect] Core 1 - 3MHz 6502 not loading BASIC (#66)

I'm not aware of this issue.

What happens with Core 1, Egg Eater, Pi Zero?

Which 3A / 3B model are you using:
https://elinux.org/RPi_HardwareHistory

RetroClinic, 2019-10-06:
Now I’m not sure this is a firmware issue. It was reported by a customer who has one of my 3A adapters. I tried it on the machine I had on the bench using a 3B and it did the same thing with Core 1, froze after the BASIC prompt and didn't return a cursor. However testing it this morning, for the life of me I can’t repeat the issue, using exactly the same hardware, it’s all functioning fine. I tried both RC1 and the full version of Egg Eater.

I can’t imagine it’s a hardware fault, as Core 0 works perfect for both of us, but it might be a very slight timing of the 2MHzE signal coming from the CPU perhaps? I tried several different CPUs but all were fine, and I changed the main 16MHz timing crystal (as I do for all machines now).

I’ll just keep testing Core 1 on various machines and configs as I go with repairs and upgrades, keep an eye on it. Thanks for getting back so quick.

Regards, Mark.

From: David Banks 
Sent: Saturday, October 05, 2019 8:06 PM
To: hoglet67/PiTubeDirect 
Cc: RetroClinic ; Author 
Subject: Re: [hoglet67/PiTubeDirect] Core 1 - 3MHz 6502 not loading BASIC (#66)

I'm not aware of this issue.

What happens with Core 1, Egg Eater, Pi Zero?

Which 3A / 3B model are you using:
https://elinux.org/RPi_HardwareHistory

hoglet67, 2019-10-06:
Mark, I can think of two possibilities...

1. It is just a bad connection on the tube connector. A few people have had very strange/intermittent issues that were fixed by cleaning the tube connector. But as you say, strange that switching to Core 0 resolved the issue. Also strange that both you and your customer experienced the same issue.

2. It is actually an issue with the speed that Core 1 is running at. The 3MHz is only approximate, and if it dips much below this during the language transfer (of BBC Basic on Ctrl Break) then it's possible that Basic woiuld get corrupted.  The delay loop that slows down the core to 3MHz is meant to adapt to what ever clock speed the Pi is running at. It's possible with the faster Pi models that something is going wrong here.

Can you try something like fifty successive Ctrl-Breaks? Each one should do a language transfer and get to the > basic prompt.

Are the 3A+ and 3B+ both running at 1.4GHz?

HamsterRICK, 2019-10-06:
Hi,
I have tried both of these methods to get my co-pro working.
By unplugging the the cable from the tube and then back in again and i have also tried the successive Ctrl+breaks. Now on the first try i got basic after first 10 successive presses. Then on the second try after turning it off and on, couldn't get anything. Then on the Third try nothing. However one thing I did get after trying it was occasionally it would start printing the divide sign in white and green repeatedly, then go to start up. Printing Acorn Tube and the star command. I tried to see if i could get this again to record. But it didn't do it may of just been a hiccup.

hoglet67, 2019-12-13:
Mark (RetroClininc),

A couple of months have passed, and I'm wondering if you have been able to reproduce this issue.

I've just got hold of a Pi 3A+ and can't recreate it myself.

HamsterRICK, 2019-12-13:
Hi. When I next get my BBC micro out. Which will be around late Sunday, Monday time. I will try Core 1 again and see if the problem still there. Since it was still doing it the last time I used the machine. But in the mean time thanks for looking into the problem!

HamsterRICK, 2019-12-14:
Hi. I am still having the problem.
[image]
Then if I control and break:
[image]
Anything you want me to try?

hoglet67, 2019-12-14:
> Anything you want me to try?

There is one test that would be very useful. On the Pi's SD Card is a file called cmdline.txt. Can you edit it so it looks exactly like:
```
copro=1 copro1_speed=4 tube_delay=0 elk_mode=0
```
The copro parameter causes it to boot into copro 1 by default.

The copro1_speed sets the default speed to 4MHz. Important: make sure this parameter contains an underscore. You'll probably find the version on your SD card misses this out.

I'd like to see if changing the speed from 3MHz to 4MHz resolves the issue.

Dave

HamsterRICK, 2019-12-14:
Hi edited the text file didn't work. However I noticed in my text file it didn't have a underscore on the copro1_speed=4, I tryed with both. But no difference.

hoglet67, 2019-12-14:
Thanks for trying this.

(The underscore needs to be there or the parameter is simply ignored)

Is there any chance you could post a photo of your setup, so that I can see the Pi, the level shifter and the cabling?

I don't have any other suggestions at the moment.

HamsterRICK, 2019-12-14:
He is the overall setup (The blue and grey cable are only there to extend the wires going to the switch):
[image]
The Pi and what I think is the level shifter:
[image]
And the cabling to the tube port:
[image]

dp111, 2019-12-14:
Could you try a jumper directly across the PWR connector instead of the
switch ?

HamsterRICK, 2019-12-14:
Hi. How strange is it that now there is jumper over the PWR and it now works. I have jumped around from other cores to 3mhz 6502 and it is working. It sometime come up with the star but after the ctrl+break it is fine. Test with elite works fine. Now I just have the problem that how am I going to use the switch if this is the case? It work fine before I was getting this problem with the switch in uses a couple of months before it started doing this problem.

HamsterRICK, 2019-12-14:
Also did I hit an Easter egg routine or something.
[image]

hoglet67, 2019-12-14:
That was a good suggestion Dominic, thanks!

The Pi 3 does need a good power supply, and those "dupont" cables are notoriosly dodgy. It does seem possible this is the cause.

Dave

HamsterRICK, 2019-12-14:
Hi, Thanks for the fix. So it possible that the switch is causing an issue with power getting to the Pi.
However if this is cause why do the other cores work with the switch and how would be able to get it running with the switch?
Also did i hit some sort of test mode on the Pi or on the RC Data centre?

dp111, 2019-12-14:
I suggest a low ESR >=100uF capacitor to ground on the Pi side by the PWR
connector.

hoglet67, 2019-12-14:
> Also did i hit some sort of test mode on the Pi or on the RC Data centre?

The 6502 Co Processors have a small BASIC program called SPHERE pre-loaded in memory.

If you LIST the program, the first three lines give useful information about the configuration, such as the PiTubeDirect version, the Pi model, and the current temperature.

Dave

HamsterRICK, 2019-12-14:
> I suggest a low ESR >=100uF capacitor to ground on the Pi side by the PWR connector.

I have some capacitor from an electronics kit but don't know if they are a low ESR capacitor. How would i find out? Also the Capacitor i have at the moment from my electronics Kit are 100uF at 50V. As well as 10uF at 50V. Are these ok since i am better off you since my electronic isn't that good.

> The 6502 Co Processors have a small BASIC program called SPHERE pre-loaded in memory.
> If you LIST the program, the first three lines give useful information about the configuration, such as the PiTubeDirect version, the Pi model, and the current temperature.

That is very Useful. Thanks!

dp111, 2019-12-14:
I would suggest giving the 100uF a try nothing to loose, except a bit of
time. You could also try using thicker cable to the switch

PS without the picture I wouldn't have have guessed the switch cables was
the issue.

hoglet67, 2019-12-14:
For reference, here is a photo of the level shifter without the Pi:
[image]

HamsterRICK, 2019-12-14:
@hoglet67 That is the same board i have in my machine.
@dp111 I will give the capacitor a go. Here is Picture of the cables:
The Yellow and Orange cables are from the Switch, are about the same thickness as the 5V cable from the BBC Power Supply.
[image]

HamsterRICK, 2019-12-14:
The Capacitor didn't work. I think i need a low rate cap in Volts and in ESR if i am correct. Please correct me if i am wrong.

hoglet67, 2019-12-14:
Can you post of photo of where/how you have connected the capacitor?

HamsterRICK, 2019-12-14:
Grey > Orange > Switch > Yellow > Red > into breadboard then into capacitor positive > out by negative into Blue into the board.
Sorry I don't have that much experience with electronic if I have wired the board up wrong.
[image]
[image]

hoglet67, 2019-12-14:
Actually, I woiud suggest you stop at this point, as the capacitor is not wires correctly.  It's possible you will damage your Pi and/or Level Shifter. For it to be effective, the capacitor needs to be soldered directly across the pins 4 and 6 of the Pi connector. I would not suggest you try to do this if you are not expecienced with electronics.

What I would suggest is that Mark takes a look at this, given the new information in this thread. Possibly he can test a switched configuration and supply you with a modified level shifter.

don't know 

, and see if he can also reproduce the issue.

HamsterRICK, 2019-12-14:
Ok. Thanks for the help. I will put the Pi back in with the Jumper and i will e-mail Mark.

HamsterRICK, 2019-12-21:
Hi, I got some new parts from Mark and put them together and now everything seems to be working again. Thanks for all the help and have a Happy Christmas!

hoglet67, 2020-01-21:
Closing, as this was an electrical (power) issue.

closed 2020-01-21 by hoglet67