hoglet67/PiTubeDirect #67: 80286 emulation buggy (pauses loading dos, wont fully load gem)

piemmm, 2019-12-06:
Not sure if this is a known thing, but if not I'll mention it just in case!

Master 512k, took the pc board out, stuck a RPI3 and pi tube adapter in place instead of it. fx 151,230,8 to stick it as a 286

Symptoms: 
 There's a 'pause' loading dos that is just after the logo is displayed (there is no pause using the actual acorn 0243,200 186 board).  Also, GEM never loads, it gets as far as sitting on a blank white screen and occasionally tries to do something with the disk.  Suspect whatever is causing the 'pause' happens lots with gem, slowing things down a lot.

Other differences/observations: 
 When the 'real' 186 board is in place, on reset, the master will immediately try to boot from disk.
 When using the pitube, the master won't boot, and I have to prod it to boot.  Swap back around, boots fine from the disk on reset.

hoglet67, 2019-12-06:
So, you are able to successfully boot DOS Plus, but then running GEM fails?

We've certainly had GEM working well in the past, so this should work.

Can I ask for a few more details please:
- What version of the PiTubeDirect software is being used?
- What adapter is being used?
- What Pi Model exactly?
- What version of DOS Plus is it?

hoglet67, 2019-12-06:
I've just done a quick test myself, and it's not totally broken:
[image]
The setup is not the same as yours, but if you can provide mode details I can try to replicate this.

piemmm, 2019-12-07:
Hi, thanks for the fast reply!  

PiTubeDirect software is from the github page, taken yesterday morning 'EggEater'
The adapter is the long 'thin' internal adapter that fits inside the master
Tested with a 3B+, and a 3B, same issue.
Dos plus 2.1 (Xios 1.03)
Standard 5&1/4" floppy, ADFS, Retroclinic DualOS (same using either MOS)

Sometimes the pause happens loading dos plus, sometimes in GEM, to varying degrees of 'pausiness'. In the video below you can see we have maximum pausage just loading dos plus
 
Video of the fault in action (just loading dos plus): https://youtu.be/mG_l6CJcZE0
Video of it working with the original acorn copro board: https://youtu.be/puP5hnRt0wg

And I've attached a pic of the setup, as basic as I can get it to rule out everything

[image]

hoglet67, 2019-12-07:
I hope you don't mind me asking a few more questions - remote debugging is never easy!

1. It would be really help if I could see what was happening on the BBC screen (with the floppy noise audible in the background). Any chance of re-doing the videos a bit more zoomed in on your monitor? Also, neither video seems to show running GEM. Maybe there is some confusion with terminology here.  GEM is the graphical interface I in the screen shot I posted above. So when you say GEM, is that really what you mean?

2. Does the 6502 Co Pro (Co Pro 0) work correctly? i.e. is something like the Tube Elite splash screen stable? 

3. It seems you have the IFEL Co Pro adapter - I have one of these as well. I have never myself used a Pi 3 connected via a cable with one of these. Unfortunately, I don't have a Pi 3 at the moment, but I will try a Pi 2 on a cable later.

Dave

piemmm, 2019-12-07:
Hi I'll try to do another video tonight as I'm out and about a the moment - when I say gem I mean the gem desktop, the video shows it also pausing just loading dos plus (I was going to try gem after it loaded but it was taking so long I simply posted an example video of it doing the same just loading dos).   

 I haven't tried the other copros yet (other than the FX and break to select them). Getting elite will take a day or so as I will have to resurrect my A540 to get it onto something the master can read

If I can find a decent header I'll try to get the pi on without the ribbon cable just in case it's that (I think I have a stack of them in one of my mouser boxes)

For clarity, the pauses in gem always started after the screen was cleared to white (after starting gem.bat).  
The pauses in dos plus (if they were going to start at all) always happened after the green digital logo had been displayed on screen and the screen mode was changed back

hoglet67, 2019-12-07:
Ian,

On a larger monitor in Full HD I can see what's going on in the Beeb Window, so no need to re-do the videos.

I've hooked up my Pi 2 to the IFEL adapter with a 30cm long cable. Using the EggEater firmware it works fine, and there are no problems booting DOS Plus 2.1 then onto GEM:

[image]
[image]
[image]
[image]

I have some suggestions for you:

1. Booting and switching between Co Processors will be more reliable if you *CONFIGURE EXTUBE. I know this seems counter intuitive, but there are sound reasons for this.

2. Some versions of DOS Plus require a certain amount of memory to boot; any more or any less would cause a crash (I assume your real 80186 Co Pro has 512KB). If you issue the following commands, the Pi should emulate a 512KB system:
```
*FX 151,226,1
*FX 151,228,8
<press break>
```
You should then see ACORN TUBE 80286 512KB in the boot message.

3. It's just possible your DOS Plus boot disk is non-standard, and is somehow triggering an emulation bug that's I'm not able to reproduce. Is there any way you could image it, so I can try it here. Alternatively, I've just made one that I know works from M512_1.ADL in this package:
http://www.cowsarenotpurple.co.uk/bbccomputer/master512/m512-sysdisks-1.zip

4. It's definitely work testing one of the other Co Processors, because hardware problems rarely affect just one Co Processor. I would recommend Co Pro 0 with Tube Elite. Here is a disk image I know works.
[tubeelt.zip](https://github.com/hoglet67/PiTubeDirect/files/3935085/tubeelt.zip)

You run it with *TUBEELT.

Dave

piemmm, 2019-12-07:
Hi, thanks for taking the time to look at this.  I've managed to get some time to make another video showing the disk drive and screen (and if you turn the volume up you'll hear the bus noise from the speaker as well). Hopefully it will be of more use.  Unfortunately I haven't had the time to get the disk images sorted yet.

In the video I have the Pi directly on the IFEL board now, so the ribbon cable is gone, unfortunately this seems to have made little difference. 

The FX commands appear to make no difference, the issue is still present if it is configured to use 512k (tried this just after I made the video).  Configuring for external tube also made no difference.

I have had one 'not ready error reading drive A, abort or retry?' message from dos plus whilst there was one paticular pause (it's the only message I've had out of about 25 of these pauses so far).

The video is at: https://youtu.be/CBIUAu2_Q6E   

Because you're not having issues your end, and I don't have issues using the original acorn copro board, I'm suspecting it may be a bad ifel board.  I'm going to get another one (non-ifel) just to be sure and a pi zero as well for good measure.

hoglet67, 2019-12-07:
Yes, this is proving a bit of a mystery....

If you get the chance, do try running Tube Elite on Co Pro 0. That's a good test of the hardware.

Trying a Pi Zero would also be useful - it's possible this is a problem with the Pi 3. It could even just be power related.

I'll probably order a Pi 3A+ this week - I've been meaning to get one for a while.

Dave

piemmm, 2019-12-13:
Quick update: Pi zero arrived yesterday, tried but unfortunately still had the same issue, so it's not that.  Still waiting for the alternate level shifter board to arrive, will post back once I've managed to test with it!

hoglet67, 2019-12-13:
Thanks for the update.... I think that rules out it being a Pi 3A+ specific issue.

I've also just bought a Pi 3A+ and it's working fine.

So I think we are now down to:
- you have a faulty level shifter
- somthing perculiar about your particular Master that affects PiTubeDirect but not the real Co Pro

It will be interesting when you get the new level shifter.

piemmm, 2019-12-13:
RetroClinic's level shifter (Issue 2C) arrived today, just had a moment to try it and it's working without any problems with the zero, 3B and 3B+.    The only difference is that this is plugged into the 40pin tube idc instead of the internal tube connector in the master (though given the original co-pro board works, I don't think this is an issue).

I think it may be fairly safe to say at this point that the IFEL board is at fault, though it's strange how it works, pause a bit, works (and doesn't completely crash out as I would expect with a bad level shifter)

closed 2019-12-14 by piemmm

hoglet67, 2019-12-14:
> I think it may be fairly safe to say at this point that the IFEL board is at fault, though it's strange how it works, pause a bit, works (and doesn't completely crash out as I would expect with a bad level shifter)

Are you planning to return the IFEL board to Steve Picton, so he can take a look at what the problem was?

piemmm, 2019-12-14:
Hi! Probably won't return - if it's just the shifter that's bad then I'll just replace it when I get around to having a closer look (the shed is a mess at the moment and my bbc 'scsi' pcb arrives on Monday so I'd like to get on with populating that first) - I will post back here if I find the issue though!