hoglet67/PiTubeDirect #72: PCCE doesn't work on the 80x86 Co Pro

hoglet67, 2020-07-06:
It does on the original 80186.

glvas, 2021-01-01:
I think PCCE does not work, not due to PiTubeDirect, but due to incompatibility with IDE Devices and / or ADFS IDE-patched ROMs.

I do not know what the testing setup is, so I may be completely wrong.

It used to freeze my machine when I was using an IDE HDD, and tried to access the HDD after running it. Now, after I switched to a BeebSCSI-emulating Pi 1Mhz interface, it seems to work fine. This solution is using the standard ADFS ROMs, so the issue may be incompatibility with ADFS IDE patches.

I am using a BBC Master with Ken Lowe's on-port PiTube and 1Mhz interfaces. Both are connected to Pi Zeros, and the PiTube is running the latest Gecko image.

hoglet67, 2021-02-09:
Thanks for that response glvas, this is good input.

I'll close this issue then.

closed 2021-02-09 by hoglet67

hoglet67, 2023-03-18:
FYI, we now appear as an 80186 using DMI rather than an 80286 using NMI.

https://github.com/hoglet67/PiTubeDirect/commit/33bb8a68dd4c34fdecb5a5becd8e73b284b2bce4

https://stardot.org.uk/forums/viewtopic.php?f=2&t=26540

Dave