hoglet67/PiTubeDirect #77: Native ARM - *BASICxxxx bug

beebmaster, 2020-09-29:
Native ARM traps star commands beginning with "BASIC" and assumes the host BASIC language ROM is being requested.  E.g. trying to load BASIC 135 from disc with *BASIC135 will attempt to invoke 6502 BASIC and exit with the "This is not ARM code" error.

hoglet67, 2020-09-29:
The Native ARM Co Pro doesn't special-case BASIC in any way, so this is probably the *BASIC command in the MOS being greedy.

Is this on a Master?

beebmaster, 2020-09-29:
Yes, it's on a Master.  It may be the MOS although I would have thought it would check for end of line after "BASIC". I'll do some testing without a co-processor active to see what it does.

beebmaster, 2020-09-29:
Mmm...looks to be a bug in MOS 3.20 but only with numbers. *BASIC9999 will just do *BASIC.  *BASICALLY will look in the filing system.

Can the Native ARM Tube trap *BASIC before it goes to the MOS?

hoglet67, 2020-09-29:
> Can the Native ARM Tube trap *BASIC before it goes to the MOS?

And do what with it?

beebmaster, 2020-09-29:
Sorry, what I meant was can it distinguish between *BASIC<cr> which it would pass to the MOS of course (or invoke ARM BASIC if it's been included in the build) and *BASICnnnn which it would pass to the filing system.

hoglet67, 2020-09-29:
I'm not sure it can, at least not easily and cleanly.

The Native ARM Co Pro can either handle a command itself, or pass it to the host via OSCLI.

It's then upto the host how the command is handled.

What's wrong with typing */BASIC135?

beebmaster, 2020-09-29:
That doesn't work, still tries to invoke BASIC.  (But it doesn't do that on other co-pros).

BigEd, 2020-10-03:
For me, it's BAS135 not BASIC135.  I wonder if that's not merely to save a bit of typing.

hoglet67, 2020-10-03:
Good point Ed - BASIC135 isn't even a legal filename on DFS (1 character too many)

beebmaster, 2020-10-04:
Very true, and I am sure we would want to keep the filename within a 7 character limit for maximum filing system compatibility. But there is still a difference between the way the Native ARM handles commands beginning with "BASIC" and other processors.

I've attached a few screenshots.

[image]
[image]
[image]
[image]
[image]

It looks like in all cases, any numeric characters after BASIC are ignored, and *BASIC is executed.
Also in all cases, any alphabetic character after BASIC are taken into account and the full command line is acted up.
Prefixing with a slash character always passes the full command line to the filing system, except with the Native ARM running where it still operates as if the slash doesn't exist: numeric suffix does *BASIC, alpha suffix does *commandline.

hoglet67, 2020-10-04:
This is the code I think that's causing a difference in behavior with handling /
https://github.com/hoglet67/PiTubeDirect/blob/master/src/tube-swi.c#L512

JGH had a hand in this bit, and so did I, so I'll need to take another look.

beebmaster, 2020-10-04:
It looks like *RUN and */ should be treated as a special case and always passed to the filing system, otherwise it's impossible to run a file which is the same name as an MOS command:

[image]
[image]

hoglet67, 2020-10-11:
> It looks like *RUN and */ should be treated as a special case and always passed to the filing system, otherwise it's impossible to run a file which is the same name as an MOS command.

Agreed, and now fixed in gecko-dev

closed 2020-10-11 by hoglet67

beebmaster, 2020-10-13:
Confirmed fixed, thanks very much.

commit 0f7ac985 (2020-10-11): Native ARM Co Pro: Fix #77 (*BASICxxxx bug)

Change-Id: I8d80a8282c3ff79078a68531a5684d4e806e2264