hoglet67/PiTubeDirect #80: Current clock speed incorrect with latest firmware:

hoglet67, 2020-10-12:
This messes up the serial port speed setting.

Benchmarks show same speed in both cases.

Old firmware fetched from github on 21/04/2020:
```
    FIRMWARE_VERSION : 5e9dc2a4 
         BOARD_MODEL : 00000000 
      BOARD_REVISION : 00900093 
   BOARD_MAC_ADDRESS : 80eb27b8 5f381171 
        BOARD_SERIAL : 11807111 00000000 
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :   1000.000 MHz   1000.000 MHz   1000.000 MHz
           CORE_FREQ :    400.000 MHz    400.000 MHz    400.000 MHz
            V3D_FREQ :    300.000 MHz    300.000 MHz    300.000 MHz
           H264_FREQ :    300.000 MHz    300.000 MHz    300.000 MHz
            ISP_FREQ :    300.000 MHz    300.000 MHz    300.000 MHz
          SDRAM_FREQ :    450.000 MHz    450.000 MHz    450.000 MHz
          PIXEL_FREQ :      0.000 MHz  -1894.967 MHz  -1894.967 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  40.08 °C
        CORE VOLTAGE :   1.35 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
     SDRAM_I VOLTAGE :   1.20 V
```
New firmware fetched from github on 12/10/2020
```
    FIRMWARE_VERSION : 5f7f443d 
         BOARD_MODEL : 00000000 
      BOARD_REVISION : 00900093 
   BOARD_MAC_ADDRESS : 80eb27b8 5f381171 
        BOARD_SERIAL : 11807111 00000000 
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :    700.000 MHz   1000.000 MHz   1000.000 MHz
           CORE_FREQ :    250.000 MHz    400.000 MHz    400.000 MHz
            V3D_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
           H264_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
            ISP_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
          SDRAM_FREQ :    400.000 MHz    450.000 MHz    450.000 MHz
          PIXEL_FREQ :      0.000 MHz  -1894.967 MHz  -1894.967 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  38.47 °C
        CORE VOLTAGE :   1.35 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
     SDRAM_I VOLTAGE :   1.20 V
```

dp111, 2020-10-12:
might be worth trying setting in config.txt : arm_freq_min and core_freq_min

hoglet67, 2020-10-12:
Setting (only) core_freq_min=400 gives:
```
    FIRMWARE_VERSION : 5f7f443d 
         BOARD_MODEL : 00000000 
      BOARD_REVISION : 00900093 
   BOARD_MAC_ADDRESS : 80eb27b8 5f381171 
        BOARD_SERIAL : 11807111 00000000 
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :    700.000 MHz   1000.000 MHz   1000.000 MHz
           CORE_FREQ :    400.000 MHz    400.000 MHz    400.000 MHz
            V3D_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
           H264_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
            ISP_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
          SDRAM_FREQ :    400.000 MHz    450.000 MHz    450.000 MHz
          PIXEL_FREQ :      0.000 MHz  -1894.967 MHz  -1894.967 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  29.32 °C
        CORE VOLTAGE :   1.35 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
     SDRAM_I VOLTAGE :   1.20 V
```
Setting only (arm_freq_min=1000) causes the system to fail to boot.

Setting only (arm_freq_min=800) gets this:
```
    FIRMWARE_VERSION : 5f7f443d 
         BOARD_MODEL : 00000000 
      BOARD_REVISION : 00900093 
   BOARD_MAC_ADDRESS : 80eb27b8 5f381171 
        BOARD_SERIAL : 11807111 00000000 
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :    800.000 MHz   1000.000 MHz   1000.000 MHz
           CORE_FREQ :    250.000 MHz    400.000 MHz    400.000 MHz
            V3D_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
           H264_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
            ISP_FREQ :    250.000 MHz    300.000 MHz    300.000 MHz
          SDRAM_FREQ :    400.000 MHz    450.000 MHz    450.000 MHz
          PIXEL_FREQ :      0.000 MHz  -1894.967 MHz  -1894.967 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  30.40 °C
        CORE VOLTAGE :   1.35 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
     SDRAM_I VOLTAGE :   1.20 V
```
Confused!

BigEd, 2021-03-14:
(Just checking - is this still an issue?)

hoglet67, 2021-03-14:
I was a "bug" in the Pi firmware that was fixed by the Pi team.

In the hognose-dev branch we are using the firmware from 2021/01/09 and don't see any issues,

I'll close.

closed 2021-03-14 by hoglet67

hoglet67, 2021-08-25:
OK, it seems there may still be an issue here, this time with the Pi 1.

On Gecko we get:
```
    FIRMWARE_VERSION : 5e208855 
         BOARD_MODEL : 00000000 
      BOARD_REVISION : 00000002 
   BOARD_MAC_ADDRESS : 73eb27b8 5f38b417 
        BOARD_SERIAL : b27317b4 00000000 
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :    700.000 MHz    700.000 MHz    700.000 MHz
           CORE_FREQ :    350.000 MHz    350.000 MHz    350.000 MHz
            V3D_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
           H264_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
            ISP_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
          SDRAM_FREQ :    400.000 MHz    400.000 MHz    400.000 MHz
          PIXEL_FREQ :      0.000 MHz  -1894.967 MHz  -1894.967 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  21.79 °C
        CORE VOLTAGE :   1.20 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
     SDRAM_I VOLTAGE :   1.20 V
```

On hognose we get
```
    FIRMWARE_VERSION : 5ff753eb
         BOARD_MODEL : 00000000
      BOARD_REVISION : 00000002
   BOARD_MAC_ADDRESS : 73eb27b85f38b417
        BOARD_SERIAL : b27317b400000000
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :    700.000 MHz    700.000 MHz    700.000 MHz
           CORE_FREQ :    250.000 MHz    350.000 MHz    350.000 MHz
            V3D_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
           H264_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
            ISP_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
          SDRAM_FREQ :    400.000 MHz    400.000 MHz    400.000 MHz
          PIXEL_FREQ :      0.000 MHz   2400.000 MHz   2400.000 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  32.01 °C
        CORE VOLTAGE :   1.20 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
     SDRAM_I VOLTAGE :   1.20 V
```
The main difference here is different Pi firmware blobs.

Just focus on the CORE_FREQ lines.

So even though we are trying to force a core clock speed of 350MHz, it looks like the system is only running at 250MHz. 

This upsets the UART rate, because we are clamping within the min/max range, see commit a091cabe

I'm thinking that commit (which was a work around before the Pi people fixed the broken firmware) is actually unhelpful in this case. But there is a new issue where for some reason we can't now force the Pi 1 to increase it's core_freq to 350MHz.

Dave

dp111, 2021-08-25:
Core freq lines look the same to me.

hoglet67, 2021-08-25:
No, because it's now returning 0 for the clock rate.

Looking at the mailbox docs, there is now a new tag: GET_CLOCK_RATE_MEASURED
https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#get-clock-rate-measured

I had a quick play with this, and it's also returning 0.

dp111, 2021-08-25:
Does the latest firmware break any other Pi s?

hoglet67, 2021-08-25:
OK, with the latest firmware if I use GET_CLOCK_RATE_MEASURED rather than GET_CLOCK_RATE, I get non-zero values:
```
    FIRMWARE_VERSION : 611e4e73
         BOARD_MODEL : 00000000
      BOARD_REVISION : 00000002
   BOARD_MAC_ADDRESS : 73eb27b85f38b417
        BOARD_SERIAL : b27317b400000000
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :    700.000 MHz    700.000 MHz    700.000 MHz
           CORE_FREQ :    250.000 MHz    350.000 MHz    350.000 MHz
            V3D_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
           H264_FREQ :      0.000 MHz    250.000 MHz    250.000 MHz
            ISP_FREQ :      0.000 MHz    250.000 MHz    250.000 MHz
          SDRAM_FREQ :      0.000 MHz    400.000 MHz    400.000 MHz
          PIXEL_FREQ :    148.500 MHz   2400.000 MHz   2400.000 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  31.48 °C
        CORE VOLTAGE :   1.20 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
```
I think this reflects reality.

So the issue is now, for some reason our config file results in a Pi 1 running at 250MHz rather than 350MHz.

dp111, 2021-08-25:
Try setting core_freq_min as well. That should fix it.

hoglet67, 2021-08-25:
It does....
```
    FIRMWARE_VERSION : 611e4e73
         BOARD_MODEL : 00000000
      BOARD_REVISION : 00000002
   BOARD_MAC_ADDRESS : 73eb27b85f38b417
        BOARD_SERIAL : b27317b400000000
           EMMC_FREQ :    200.000 MHz    200.000 MHz    200.000 MHz
           UART_FREQ :     48.000 MHz   1000.000 MHz   1000.000 MHz
            ARM_FREQ :    700.000 MHz    700.000 MHz    700.000 MHz
           CORE_FREQ :    350.000 MHz    350.000 MHz    350.000 MHz
            V3D_FREQ :    250.000 MHz    250.000 MHz    250.000 MHz
           H264_FREQ :      0.000 MHz    250.000 MHz    250.000 MHz
            ISP_FREQ :      0.000 MHz    250.000 MHz    250.000 MHz
          SDRAM_FREQ :      0.000 MHz    400.000 MHz    400.000 MHz
          PIXEL_FREQ :    148.500 MHz   2400.000 MHz   2400.000 MHz
            PWM_FREQ :      0.000 MHz    500.000 MHz    500.000 MHz
           CORE TEMP :  32.55 °C
        CORE VOLTAGE :   1.20 V
     SDRAM_C VOLTAGE :   1.20 V
     SDRAM_P VOLTAGE :   1.20 V
     SDRAM_I VOLTAGE :   1.20 V
```
I'll push this shortly....