hoglet67/PiTubeDirect #107: What is the best location to load native ARM programs for execution?

markdryan, 2020-12-22:
I'm trying to figure out what the best location is to load ARM machine code programs for execution on the native ARM copro (15).  Someone on startdot suggested 0000f000, and this does work, but after the program exits my datacentre no longer seems to work until I do a CTRL-BREAK.  Is there a recommended program load location that is compatible with ADFS and the datacentre?

hoglet67, 2020-12-22:
F000 should be fine.

What release of PiTubeDirect are you using?

With the latest Gecko RC0 release, 8000-C000 is no longer corrupted by a language transfer on Control-BREAK. So you could go much lower.

I'm not sure what the DataCentre issue is. I know some versions of Data Centre RamFS doesn't work correctly with files > 64KB in size. But it sounds like you are using ADFS which should be fine. How large is your excutable?

Dave

markdryan, 2020-12-22:
I'm using Fer De Lance (on a master).  The binary size is 8kb (8136).  The compiler that produced the binary is a little experimental, so it's almost certainly a problem on my end, now I know that the load address I'm using is okay.  Thanks for the confirmation.

closed 2020-12-22 by markdryan

hoglet67, 2020-12-22:
I would f try Gecko-RC0, there have been lots of fixes to the Native ARM Co Pro since Fer-De-Lance.

How are you exiting your application? I think the correct way is to call the OS_Exit (SWI &11),

markdryan, 2020-12-22:
> I would try Gecko-RC0, there have been lots of fixes to the Native ARM Co Pro since Fer-De-Lance.

Okay, will do.  I'll upgrade tonight.  I'll also make sure that the data centre still works fine after the program is loaded but before it's executed.  In addition, I'll try some other addresses.

> How are you exiting your application? I think the correct way is to call OS_Exit (SWI &11),

I'm using SWI &11 and control returns back to the command prompt after the program finishes executing.  It's just that when I get control back, the data centre (well the RAM drive, haven't tested the harddisk) isn't working.  The binary is calling some unsupported APIs however, e.g., OS_RemoveCursors.  I'll get rid of these to see whether it makes any difference.

markdryan, 2020-12-22:
I tried to reproduce the problem this evening on Fer de Lance without success.  Everything is working fine now at f000, I get the console back after the program finishes executing (as before) but now the datacentre works.  I think the issue must have been with my binary (there are still some things which don't work, e.g., escape handling, and which may be corrupting things.)  So sorry for the noise.  Anyway, it's good to get confirmation about f000 as a good execution address.

> With the latest Gecko RC0 release, 8000-C000 is no longer corrupted by a language transfer on Control-BREAK. So you could go much lower.

What would be the recommended start address on Gecko? 8000 or could we go lower than that?