hoglet67/PiTubeDirect #108: gcc 10.2 -pedant warnings for tube-swi.c

dp111, 2020-12-22:
/mnt/c/linux/PiTubeDirect-gecko-dev/src/tube-swi.c: In function 'user_exec_fn':
/mnt/c/linux/PiTubeDirect-gecko-dev/src/tube-swi.c:348:20: warning: ISO C forbids conversion of function pointer to object pointer type [-Wpedantic]
  348 |   ret = _user_exec((unsigned char *)f, param, 0, 0);
      |                    ^
/mnt/c/linux/PiTubeDirect-gecko-dev/src/tube-swi.c: In function 'user_exec_raw':
/mnt/c/linux/PiTubeDirect-gecko-dev/src/tube-swi.c:378:22: warning: cast discards 'volatile' qualifier from pointer target type [-Wcast-qual]
  378 |       generate_error((void *) address, 249, "This is not ARM code");

hoglet67, 2020-12-22:
This one should be fixed now.

dp111, 2023-09-06:
Fixed in Indigo-alpha1

closed 2023-09-06 by dp111