hoglet67/PiTubeDirect #111: Native ARM implementation of OS_ReadPoint returns an incorrect value for off screen coordinates

markdryan, 2020-12-24:
It returns 255 when it should return -1.  For example, on Gecko RC0

```
*armbasic
P. POINT(-1,-1)
```

prints 255.

If you try the same command with the copro disabled, or on a RiscPC, you get -1.

Looking at the SWI docs ( https://www.riscosopen.org/wiki/documentation/show/OS_ReadPoint ), it seems that R4 should be set to -1 if the point cannot be read.  The current code only seems to initialise R2  (https://github.com/hoglet67/PiTubeDirect/blob/master/src/tube-swi.c#L881).

hoglet67, 2020-12-24:
This should be easy to fix.

One question: if thie point is off screen, then R4=-1, but what happens to R2/3?

hoglet67, 2020-12-24:
Does this seem correct to you?
```
  int pt = receiveByte(R2_ID);  // logical colour of point, or 0xFF is the point is off screen
  if (pt == 0xff) {
     reg[2] = 0;
     reg[3] = 0;
     reg[4] = -1;
  } else {
     reg[2] = pt;
     reg[3] = 0;
     reg[4] = 0;
  }
```

markdryan, 2020-12-24:
> One question: if thie point is off screen, then R4=-1, but what happens to R2/3?

Helpfully, un-documented.  However, I tried the following on RPCEmu

```
> SYS "OS_ReadPoint", -1, -1 TO ,, r2%,r3%,r4%
> P. r2%, r3%, r4%
-1        0       -1
```
So that would suggest that in the error case, 

reg[2]=-1;

Perhaps BASIC doesn't even bother to read the value from r4.

markdryan, 2020-12-24:
So just to clairfy,

```
  int pt = receiveByte(R2_ID);  // logical colour of point, or 0xFF is the point is off screen
  if (pt == 0xff) {
     **reg[2] = -1;**
     reg[3] = 0;
     reg[4] = -1;
  } else {
     reg[2] = pt;
     reg[3] = 0;
     reg[4] = 0;
  }
```

hoglet67, 2020-12-24:
OK,  that fix will be included in the final Gecko release next week.

Thanks

hoglet67, 2020-12-24:
For reference, this is the implementation of POINT is ARM Basic (v1.35):
```
POINTB  STR     R14,[SP,#-4]!
        BL      EXPR
        BL      INTEGY
        CMP     R10,#","
        BNE     ERCOMM
        STR     IACC,[SP,#-4]!
        BL      BRA
        BL      INTEGZ
        MOV     R1,R0
        LDMFD   SP!,{R0,R14}
        SWI     READPOINT
        MOV     R0,R2
        B       SINSTK
```
So you are right, it does just look at R2.

Dave

markdryan, 2020-12-24:
> OK, that fix will be included in the final Gecko release next week.

Excellent.  Thanks.

For reference, I stumbled across the issue playing with the mounts example from creative graphics, which does a

```UNTIL POINT(x%,y%) = -1.```

hoglet67, 2020-12-24:
I've just pushed a further fix to OS_ReadPoint

Turns out the original was broken, even if the coordinates were on screen.

I'd neglected to take account of OSWORD blocks being serialized backwards!

If you get the chance, I would appreciate it if you could test the latest version.

markdryan, 2020-12-26:
Sorry, I only just noticed your last comment.  I've just tested POINT on the latest build that you uploaded to #112 and it seems to be working okay for me.  I plotted 16 pixels of different colours using GCOL/POINT to the screen and read them back with POINT.  The returned values were correct, so it seems to be working.

closed 2021-02-09 by hoglet67