hoglet67/PiTubeDirect #112: Workspace pointer not passed to Escape Handler in Native ARM copro

markdryan, 2020-12-24:
At least I don't think it is.  I'm not quite sure how to debug this with my setup but looking at the code it doesn't seem to be passed.  The escape handler wrapper seems to initialise R11, which is good, but not R12.

https://github.com/hoglet67/PiTubeDirect/blob/697b4c038fa5cb79ea98398e7da871f286642db0/src/copro-armnativeasm.S#L217

The workspace pointer is however passed to the exit handler

markdryan, 2020-12-24:
Here's the RiscOS documentation for the escape handler.

https://www.riscosopen.org/wiki/documentation/show/Escape%20Handler

hoglet67, 2020-12-24:
I've pushed a fix for this to the gecko-dev branch.

Would you be able to test it for me?

(if you need me to upload a PiTubeDirect build for testing, then I can do that)

markdryan, 2020-12-24:
The fix looks good to me.

> Would you be able to test it for me?

I can indeed.  It would be great if you could upload a test build.

hoglet67, 2020-12-24:
[PiTubeDirect_20201224_1828_dmb.zip](https://github.com/hoglet67/PiTubeDirect/files/5740322/PiTubeDirect_20201224_1828_dmb.zip)

This includes the fixed version of OS_ReadPoint

markdryan, 2020-12-24:
My escape handler still isn't working but looking at the code I think it's getting passed the EXIT handler's R12.

POINT(-1,-1) works find however.

Also, escape still works in general on the *prompt and also in BASIC.

markdryan, 2020-12-24:
So

> unsigned int r12 = env->handler[EXIT_HANDLER].r12;

should be

> unsigned int r12 = env->handler[ESCAPE_HANDLER].r12;

in the two places it's used.

hoglet67, 2020-12-24:
Sorry, that was a stupid cut/paste error

Here's another build:
[PiTubeDirect_20201224_2041_dmb.zip](https://github.com/hoglet67/PiTubeDirect/files/5740482/PiTubeDirect_20201224_2041_dmb.zip)

markdryan, 2020-12-24:
> Here's another build:

Excellent.  It works!  I can now escape out of my binaries without any issues.  I briefly tested escape on the * prompt and in ARMBASIC and everything seems to be working.  Thanks for the quick fix!

hoglet67, 2020-12-24:
Thanks for testing this.

closed 2020-12-24 by hoglet67