hoglet67/PiTubeDirect #126: JIT 6502: Overlapped Data Transfers causes random basic crashes (from Mincebert)

hoglet67, 2021-09-04:
See this post:
https://stardot.org.uk/forums/viewtopic.php?p=333305#p333305

I now have a test case that reproduces Mincebert's bug.

On the host side it uses his TUBEXFR code from github:
https://github.com/mincebert/TubeXfer/blob/main/TubeXfr

On the parasite side it uses the following Basic:
```
   10 MODE 0
   20 length%=1024*8
   30 DIM block% 32
   40 DIM buffer1% length%, buffer2% length%
   50 FOR I%=0 TO length%-1
   60 buffer1%?I%=I%
   70 buffer2%?I%=I% EOR &FF
   80 NEXT
   90 block%?0=11
  100 block%?1=0
  110 block%?2=0
  120 block%!7=&3000+8*640
  130 block%!9=length%
  140 A%=&E0
  150 X%=block% MOD 256
  160 Y%=block% DIV 256
  170 I%=0
  180 REPEAT
  190 block%!3=buffer1%
  200 CALL &FFF1
  210 FOR A=0 TO 3600
  220 B=SIN(2*PI*2/360)
  230 NEXT
  240 block%!3=buffer2%
  250 CALL &FFF1
  260 FOR A=0 TO 3600
  270 B=COS(2*PI*2/360)
  280 NEXT
  290 I%=I%+1
  300 PRINT CHR$(30);I%
  310 UNTIL FALSE
```
Attached is an SSD disk image:
[xfrtst.zip](https://github.com/hoglet67/PiTubeDirect/files/7109765/xfrtst.zip)
Eventually the client will either crash or produce a random error:
[image]

hoglet67, 2021-09-04:
Crash happens even if vdu=0 in cmdline.txt, so not caused by interaction with the ARM IRQ handler.

hoglet67, 2021-09-04:
Tested an older version (e3f0e870) from before I started messing with the ARM interrupt code.

This fails the test cases as well.

hoglet67, 2021-09-04:
Slightly shorter test case, with only one buffer:
```
   10 MODE 0
   20 length%=1024
   30 DIM block% 32
   40 DIM buffer% length%
   50 FOR I%=0 TO length%-1
   60 buffer%?I%=I%
   70 NEXT
   80 block%?0=11
   90 block%?1=0
  100 block%?2=0
  110 block%!3=buffer%
  120 block%!7=&3000+8*640
  130 block%!9=length%
  140 A%=&E0
  150 X%=block% MOD 256
  160 Y%=block% DIV 256
  170 REPEAT
  180 CALL &FFF1
  190 FOR A=0 TO 360
  200 B=SIN(2*PI*2/360)
  210 NEXT
  220 VDU &2E
  230 UNTIL FALSE
```
[xfrtst.zip](https://github.com/hoglet67/PiTubeDirect/files/7109840/xfrtst.zip)
[image]

hoglet67, 2021-09-04:
Also happens on a Pi Zero.

hoglet67, 2021-09-04:
Dominic,

This is getting interesting....

I've got some evidence that the issue involves PLP.

I wrote a test case that looked for evidence of the 6502 state getting messed up.
```
   10 *XFR
   20 PROCassemble
   30 MODE 0
   40 length%=8192
   50 DIM block% 32
   60 DIM buffer% length%
   70 FOR I%=0 TO length%-1
   80 buffer%?I%=I%
   90 NEXT
  100 block%?0=11
  110 block%?1=0
  120 block%?2=0
  130 block%!3=buffer%
  140 block%!7=&3000+8*640
  150 block%!9=length%
  160 A%=&E0
  170 X%=block% MOD 256
  180 Y%=block% DIV 256
  190 REPEAT
  200 CALL &FFF1
  210 !&80=-10000
  220 CALL test
  221 R%=!&80
  222 IF R%<>0 PRINT "Fail:";~R%, ~?&84,~?&85:END
  240 VDU &2E
  250 UNTIL FALSE
  260 DEF PROCassemble
  270 DIM code% 256
  280 FOR I%=0 TO 2 STEP 2
  290 P%=code%
  300[OPT I%
  310 .test
  320 TSX
  330 STX &84
  340 LDA #&12
  350 LDX #&34
  360 LDY #&56
  370 CLV
  380.loop
  381 PHP
  382 PLP
  390 CMP #&12
  400 BNE fail
  410 CPX #&34
  420 BNE fail
  430 CPY #&56
  440 BNE fail
  450 BVS fail
  451 TSX
  452 CPX &84
  453 BNE fail
  454 LDX #&34
  460 INC &80
  470 BNE loop
  480 INC &81
  490 BNE loop
  500 INC &82
  510 BNE loop
  520 INC &83
  530 BNE loop
  540 BEQ exit
  550.fail
  560 PHP
  570 STA &80
  580 STX &81
  590 STY &82
  600 PLA
  610 STA &83
  611 TSX
  612 STX &85
  620 .exit
  630 LDX &84
  640 TXS
  645 CLI
  650 RTS
  660]
  670 NEXT
  680 ENDPROC
```
[xfrtst.zip](https://github.com/hoglet67/PiTubeDirect/files/7110062/xfrtst.zip)
Without 381/382 (PHP/PLP) the test runs undefintely.

With them you will sometimes get:
[image]

It the value of the flags saved by the test post-amble looks like junk to me.

I'm pretty sure the problem is PLP (I tested them individually)

Could you take a look at the implementation of PLP and see if you can spot any obvious problems if it gets interrupted?

Dave

dp111, 2021-09-04:
Good find . I thought that it might be an instruction. Let me investigate .

hoglet67, 2021-09-04:
Here's a smaller test case that doesn't involve 6502 NMI, just 6502 IRQ (the event interrupt):
```
   10 DIM code% 100
   20 FOR I%=0 TO 2 STEP 2
   30 P%=code%
   40[OPT I%
   50 .test
   60 CLV
   70 LDX #&00
   80 LDY #&00
   90 STX &80
  100 STY &81
  110 .loop
  120 PHP
  130 PLP
  140 BVS exit
  150 DEX
  160 BNE loop
  170 DEY
  180 BNE loop
  190 .exit
  200 STX &80
  210 STY &81
  220 CLI
  230 RTS
  240]
  250 NEXT
  260 *FX 14,4
  270 REPEAT
  280 CALL test
  290 count=!&80 AND &FFFF
  300 VDU &2E
  310 UNTIL count<>0
  320 PRINT '"count = ";count
```
The file is MINTST2 here:
[xfrtst.zip](https://github.com/hoglet67/PiTubeDirect/files/7110139/xfrtst.zip)

Half the time the test hits the failure case, the other half of the time it crashes.
[image]

Replace the PLP at 130 with PLA and it runs correctly for ever.

hoglet67, 2021-09-04:
Maybe the same bug, maybe a different one.... PLP can push an incorrect value onto the stack:
```
   10 DIM code% 100
   20 FOR I%=0 TO 2 STEP 2
   30 P%=code%
   40[OPT I%
   50 .test
   60 CLV
   65 CLC
   70 LDX #&00
   80 LDY #&00
   90 STX &80
  100 STY &81
  110 .loop
  120 PHP
  130 PLA
  131 AND #&41
  140 BNE exit
  150 DEX
  160 BNE loop
  170 DEY
  180 BNE loop
  190 .exit
  200 STX &80
  210 STY &81
  220 CLI
  230 RTS
  240]
  250 NEXT
  260 *FX 14,4
  270 REPEAT
  280 CALL test
  290 count=!&80 AND &FFFF
  300 VDU &2E
  310 UNTIL count<>0
  320 PRINT '"count = ";count
```
The file is MINTST3 here:
[xfrtst.zip](https://github.com/hoglet67/PiTubeDirect/files/7110166/xfrtst.zip)

[image]

hoglet67, 2021-09-04:
I can also get a failure with:
```
LDA #&A5
.loop
PHA
PLA
CMP #&A5
```
So maybe it's stack pushing in general?

dp111, 2021-09-04:
If I was to suspect anything it would be popping the stack. maybe
interrupts need to be disabled around the stack point adjustment and the
load of the byte from the stack ?

hoglet67, 2021-09-04:
Yes, looking at the current POP code, it's definitely not atomic
```
macro SPOP reg sign
   mov     \reg,#1
   uadd8   regSP,regSP,\reg
 <<<<< FIQ HERE IS BAD >>>>>>
.ifc \sign,SIGN
   ldrsb   \reg, [regSP]
.else
   ldrb \reg,[regSP]
.endif
.endm
```
If an ARM FIQ happens at the instruction parked above, then 6502 IRQ state will clobber the value we are in the middle of popping off the stack before we pop it.

I think there is generally a safe order to do things in:

PUSH:
-  update the stack pointer
-  write the value

POP:
-  read the value
-  update the stack pointer

If this ordering can be followed, a 6502 interrupt in the middle is harmless

But that might be more difficult that wrapping with CPID f / CPIE f

hoglet67, 2021-09-04:
Last thought for today....

If this is the issue, then I would expect if left long enough that SPHERE would crash with *FX 14,4 events active.

I'll leave it running for a few hours....

dp111, 2021-09-04:
on the hognose-arm64 branch I think I have made the update of regSP atomic,
but this doesn't fix the problem unless there is more than one issue.

hoglet67, 2021-09-04:
I've just looked over the updated code and it looks fine to me. I'll do some more testing tomorrow morning. I'm sure we are close to resolving this....

hoglet67, 2021-09-05:
It's certainly fixed the issue with PHA then PLA - I no longer ever see incorrect results with that pair

The following sequences seem reliable with *FX 14,4 interrupts running and don't give unexpected results
- JSR/RTS
- PHA/PLA
- PHP/PLA
- SEI/PHP/PLP/CLI
- LDA #&37/PHA/PLP/CLI

However the following still crash quickly
- LDA #&33/PHA/PLP
- PHP/PLP

In some ways, testing with *FX 14,4 is quite a benign test, because you should only evey get isolated tube interrupts, nicely spaced 20ms apart.

The above does suggest the problem is when PLP is interrupted.

What's perculiar now is that the issue appears to be a consistent crash of the 6502 execution. I never see and ARM stack trace, so it's not a hard crash in that respect. And pressing BREAK always cleanly recovers. If I have the VDU driver running, then when a crash happens the cursor stops flashing, so ARM IRQs are disabled.

Is there any way we could dump the existing 6502 registers (inc PC) when a reset is detected?

dp111, 2021-09-05:
thats some good data points.

I suspect the issue is in : check_irq_plp . we need to have FIQs disabled
during the check for if IRQ happened during execution or it will be lost.

hoglet67, 2021-09-05:
The critical section seems to be the second half of the PLP macro
```
 .macro  PLP
   mov   temp,#1
   uadd8 temp2,regSP,temp
   ldrb  temp2, [temp2]
   uadd8 regSP,regSP,temp
   CPSID f
   and  flags, flags, #0xffffff00
   orr  flags, flags, temp2        // preserve the other bits in flags, e.g. the slowdown bit
   and  temp, temp2, #N_FLAG6502
   and  temp2, temp2, #Z_FLAG6502+C_FLAG6502
   mov  temp, temp, lsl #24
   bic  flags, flags, #N_FLAG6502+X_FLAG6502+B_FLAG6502+Z_FLAG6502+C_FLAG6502
   orr  temp, temp, temp2, lsl #29
   CPSIE f
   //   Need to update the ARM flags later with  msr     CPSR_flg, temp
   // temp1 return address
   // temp has 6502 flags to be restored to the ARM
.endm
```
Adding CPSID/E seems to move the PHP/PLP sequence into the reliable camp.

This is quite unexpected!

hoglet67, 2021-09-05:
Or more specifically, just this bit needs to apear atomic:
```
   CPSID f
   and  flags, flags, #0xffffff00
   orr  flags, flags, temp2        // preserve the other bits in flags, e.g. the slowdown bit
   bic  flags, flags, #N_FLAG6502+X_FLAG6502+B_FLAG6502+Z_FLAG6502+C_FLAG6502
   CPSIE f
```

dp111, 2021-09-05:
okay for a short window the IRQ enable flag becomes set .

hoglet67, 2021-09-05:
Can the be re-written using temp registers so that the update to flags is atomic?

hoglet67, 2021-09-05:
e.g.
```
   and  temp, flags, #0xffffff00
   orr  temp, temp, temp2        // preserve the other bits in flags, e.g. the slowdown bit
   bic  flags, temp, #N_FLAG6502+X_FLAG6502+B_FLAG6502+Z_FLAG6502+C_FLAG6502
```

hoglet67, 2021-09-05:
That still passes MINTST2 (PHP/PLP) without needing CPSID/CPSIE

dp111, 2021-09-05:
> Just tried exactly the same code sequence here!

still fails my PHP/PLA test here on a Pi4. maybe we have some other
differences.

hoglet67, 2021-09-05:
Can you post that test?

hoglet67, 2021-09-05:
Bugger, XFRTST has crashed after 604 iterations, but with slightly different symtoms: the VDU cursor is still flashing, and the Beeb Host is not dead-locked (caps lock still works). So maybe a different/new issue?

dp111, 2021-09-05:
my test that failed was  MINTST3

hoglet67, 2021-09-05:
I'm running MINTST3 now on a Pi 4 (vdu=0) and it's run for at least a minute

hoglet67, 2021-09-05:
The only change is to macro PLP:
```
.macro  PLP
   mov   temp,#1
   uadd8 temp2,regSP,temp
   ldrb  temp2, [temp2]
   uadd8 regSP,regSP,temp

   and  temp, flags, #0xffffff00
   orr  temp, temp, temp2        // preserve the other bits in flags, e.g. the slowdown bit
   bic  flags, temp, #N_FLAG6502+X_FLAG6502+B_FLAG6502+Z_FLAG6502+C_FLAG6502
   and  temp, temp2, #N_FLAG6502
   and  temp2, temp2, #Z_FLAG6502+C_FLAG6502
   mov  temp, temp, lsl #24
   orr  temp, temp, temp2, lsl #29
   //   Need to update the ARM flags later with  msr     CPSR_flg, temp
   // temp1 return address
   // temp has 6502 flags to be restored to the ARM
.endm
```

hoglet67, 2021-09-05:
Just hit 3000 iterations of the original XFRTST on a Pi 4 (with vdu=0)

hoglet67, 2021-09-05:
Dominic, could you re-test on a Pi 4 with the PLP macro from two posts up.

This seems entirely reliable for me, so I'm interested if you can make it fail.

dp111, 2021-09-05:
I still see and issue with PHP/PLA  so not even using the POP macro.

hoglet67, 2021-09-05:
Can you attach your current Pi 4 Kernel?

dp111, 2021-09-05:
attached

hoglet67, 2021-09-05:
Don't see it... You might need to ZIP it.

dp111, 2021-09-05:
try this

hoglet67, 2021-09-05:
Still no attachement visible.

hoglet67, 2021-09-05:
You need to drag/drop the ZIP file in using a browser.

dp111, 2021-09-05:
[kernel8-32.zip](https://github.com/hoglet67/PiTubeDirect/files/7111953/kernel8-32.zip)

hoglet67, 2021-09-05:
Hmmm, seems to MINTST3 seems to run for me.

How does it fail for you, and how quickly?

Are you using a Master or a Model B?

Internal or External Tube?

Value of the vdu property in cmdline.txt?

dp111, 2021-09-05:
Master, internal tube , vdu = 1 
fails within a minute.

dp111, 2021-09-05:
most with count =  errors . even vdu =0 fails. I could have some dodgy hardware

dp111, 2021-09-05:
okay I might have some dodgy hardware actually . I've had copro 0 not load a file correctly.

dp111, 2021-09-05:
I'll leave PHA /PLA running

dp111, 2021-09-05:
copro 0 has just crashed. So I probably have a hardware issue.

hoglet67, 2021-09-05:
I can't get your attached kernel to boot at all. It just crashes the Pi 4 early in the startup. Maybe it's got corrupted somehow?

But I can reproduce the issue....

I've just rebuilt from yout latest hognose-arm64 branch (a1906ac) and after a while (a couple of minutes) MINTST3 fails, even with the PLP macro patch applied.

So I guess there is another issue somewhere.

This time it seems to be dependent on having vdu=1; it hasn't yet crashed with vdu=0.

I'll soak test for while longer with vdu=0.

hoglet67, 2021-09-05:
> I'll soak test for while longer with vdu=0

A bit more testing of a190a6c and MINTST3 on Pi4 (non-debug build):

vdu=0:
- Without the PLP patch it has run now for over an hour

vdu=1:
- Without the PLP patch it fails after 3s, 75s, 57s, 27s, 23s 
- With the PLP patch it fails after 3s, 75s 15s, 95s, 107s

This suggests the PLP patch is not making a difference here (but it does elsewhere, in the PHP/PLP test)

Failures are all clean (i.e. exit with non-zero count), rather than any kind of crash. VDU cursor keeps flashing.

Even though the test doesn't include PLP, I assume BBC Basic does, so it's possible even this test might have need the PLP patch.

Done for today....

hoglet67, 2021-09-05:
OK, not quite finished...

When the test fails, the value incorrectly popped off the stack by the test program is always &FF, which I suspect means the fake interrupt address has somehow clobbered the valid stack data,

hoglet67, 2021-09-05:
Tests to do tomorrow:
- check vdu=1 failure also occurs on Pi 0 and Pi 3A (it should)
- check vdu=1 failure goes away if ARM IRQ interrupts are disabled (i.e. confirm it is the IRQ causing the problem, not some other aspect of the VDU driver)

dp111, 2021-09-05:
I'm running a pi3B now with vdu =1 and its done over 15minutes now. So we could have a Pi4 specific bug .

hoglet67, 2021-09-06:
I believe this is a fix for the vdu=1 issue: 3620a96

If an ARM FIQ occurred shortly after an ARM IRQ, but before the first instruction of the ARM IRQ handler (str r0, _main_irq_r9_regsp) ran, then the ARM FIQ handler would restore a stale value for regSP.

In out test case, this could result in the 6502 IRQ hander executing with a 6502 stack pointer that was off by one.

What surprised me was that this failure was happening so frequently.

But here's a quick calculation, given the interrupt rates:
- The ARM FIQ rate is 20ms (BBC VSync Event)
- The ARM IRQ rate is 1ms (Timer) plus 20ms (VSYNC) which we will ignore for now

For a failure to be possible, both FIQ and IRQ have to occur within a very tight window (<10 ARM clocks I would imagine)

In 1ms there are ~1,000,000 ARM clocks, so the probability of a given FIQ hitting that 10 cycle window is 1 in 100,000

As FIQ's are occurring at 50Hz, then this should ~2,000 seconds.

OK, maybe that isn't too far off what we are seeing.

I've had the fix running for about half an hour now, and no errors.

For testing, I've stagedthe three IRQ related fixed to alpha2 on their own branch:
https://github.com/hoglet67/PiTubeDirect/commits/hognose_irq_fix

Dave

hoglet67, 2021-09-06:
Actually, looking at the BCM2835 datasheet, it's possible the ARM Timer is running much more frequently this I thought.
```
   rpiArmTimer->Load = 1000;
   rpiArmTimer->Control = RPI_ARMTIMER_CTRL_PRESCALE_1 | RPI_ARMTIMER_CTRL_INT_ENABLE | RPI_ARMTIMER_CTRL_ENABLE;
```
For some reason I thoght that the base clock for the timer was 1MHz, but it seems it may actually be the system clock.

Yikes!

dp111, 2021-09-06:
can I try yours ?

dp111, 2021-09-06:
Just tried exactly the same code sequence here!

hoglet67, 2021-09-06:
> Just tried exactly the same code sequence here!

I hope it worked as well!

It's now been running for two hours without hitting the issue.

I'm still surprised at the frequency of the failure though.

hoglet67, 2021-09-06:
Fixed by these three commits:
- ba2aeb2b
- 48c8cd2
- 3720a96

closed 2021-09-06 by hoglet67

hoglet67, 2021-09-06:
As I'm still not completely understanding why the vdu=1 issue happened so frequenctly on a Pi 4, I measured the ARM timer interrupt rate with a telltale GPIO and a scope:
- Pi 0 - 1ms - as expected
- Pi 3A - 1ms - as expected
- Pi 4 - 22us - ~45x faster than expected

Why is this?

We rely on the firmware to set the predivider, so the nominal base clock rate is 1MHz.

Lets look at how the timer has been set up then....

Pi 0:
- ARM Timer Control    = 003e0020
- ARM Timer Predivider = 0000018f

Pi 3A
-ARM Timer Control    = 003e0020
-ARM Timer Predivider = 0000018f

Pi 4B:
- ARM Timer Control    = 003e0020
- ARM Timer Predivider = 0000000a

On the Pi 0 and 3A the system clock is 400MHz and the Predivider is 0x18F = 399. This gives a base clock rate of 400(399+1) = 1MHz exactly.

On the Pi 4B the system clock is 500MHz and the Predivider is 0xA = 10.  This gives a base clock rate of 500(10+1) = 45.45MHz.

I guess that's the problem then. We need to set the Predivider to 499 (0x1F3).

Seems like a firmware bug to me...

hoglet67, 2021-09-06:
Timer issue fixed in 30807ae