hoglet67/PiTubeDirect #127: 80x86 Co Pro: Possible issue with IDIV8

hoglet67, 2021-09-06:
From Dominic:

Just looking over warnings for various code checkers . Does anyone know an x86 checker like dormann ? I think cpu80186.c line 1124 has a bug. it looks as though it is written for a 16 bit divide where as it should be an 8 bit divide. line 1143 I think needs to be reinstated bit with (s2 < 0x80)? s2 :....

https://github.com/hoglet67/PiTubeDirect/blob/6e10b31991d06468f07a98120dcbdba19f73dc6b/src/cpu80186/cpu80186.c#L1143

This would be a long-standing fake86 bug.

hoglet67, 2021-09-06:
There is also a bug in this line:
https://github.com/hoglet67/PiTubeDirect/blob/6e10b31991d06468f07a98120dcbdba19f73dc6b/src/cpu80186/cpu80186.c#L1141

I wonder if the best solution is to just sign-extend the divisor to 16 bits.

i.e. instead of
```
  s2 = divisor
```
have
```
  s2 = divisor;
  if (s2 & 0x80) {
    s2 |= 0xff00;
  }
```
Then both s1 and s2 are 16 bits, and the code should be the same as the 16-bit version.

dp111, 2021-09-06:
looks sensible : https://www.felixcloutier.com/x86/idiv

hoglet67, 2021-10-30:
There may be a second issue with idiv.

The documentation states:

> Non-integral results are truncated (chopped) towards 0. 

I believe the implication of this is that the sign of the remainder should depend only on the sign of the dividend:

```
 13 / 4 : quotient =  3 remainder =   1    (checking:  3* 4+1= 13)  
 13 /-4 : quotient = -3 remainder =   1    (checking: -3*-4+1= 13)
-13 / 4 : quotient = -3 remainder =  -1    (checking: -3* 4-1=-13)
-13 /-4 : quotient =  3 remainder =  -1    (checking:  3*-4-1=-13)
```
I don't believe the current code is correct in this respect...
```
  if (sign)
  {
    d1 = (~d1 + 1) & 0xff;
    d2 = (~d2 + 1) & 0xff;
  }
```
It looks like the current code always makes the sign of the quotient (d1) and remainder (d2) the same.

I really need to try write some test cases, and maybe get someone to run them on a real 80186 Co Pro.

hoglet67, 2021-10-31:
Here's a small test case for IDIV8:
[image]
Here's the (correct) result from the Matchbox Co Pro:
[image]
Here's the (incorrect) result from PiTubeDirect Alpha4:
[image]
And here's the result after the fix:
[image]

hoglet67, 2021-10-31:
Here's a small test case for IDIV16:
[image]
Here's the (correct) result from the Matchbox Co Pro:
[image]
Here's the (incorrect) result from PiTubeDirect Alpha4:
[image]
And here's the result after the fix:
[image]

hoglet67, 2021-10-31:
Dominic, if you get the chance, could you run the updated code through cppcheck?

commit 334e734f (2021-10-31): Co Pro 80x86: Fix #127 (Issues with IDIV)

Change-Id: Id1be16e5a55197de45b8e8614748d5b9b77ae6a7
commit 6b14320c (2021-11-14): Co Pro 80x86: Fix #127 (Issues with IDIV) - overflow cases

Change-Id: I27e3b47a89c990add1764afbc450f2734176682f
closed 2021-10-31 by hoglet67