hoglet67/PiTubeDirect #130: Frame Buffer: *CAT output from ADFS looses final line feed

hoglet67, 2021-09-08:
(reported by Mince)

There is an odd problem on the last line of ADFS directory listings: if the last line is not complete (e.g. you're in a 40 column mode and there are an odd number of entries), so the last item is on a line of its own, the prompt is displayed overwriting the last line. This doesn't happen if the last line is complete, nor does it seem to happen with other filesystems (although I've only tried MMFS) and I've not seen it elsewhere.

*CAT uses the host side redirect, so it's possible we are getting characters lost (or re-ordered)

hoglet67, 2021-09-08:
OK, this is a ADFS wierdness.

The *CAT implementation on ADFS actually reads the cursor position:
```
;;
;; FSC 5 - *CAT
;; ============
.L93D5 JSR LA50D
       JSR L9478
.L93DB JSR L9331
       LDA #&04
       STA &C22B
.L93E3 LDY #&00
       LDA (&B6),Y
       BEQ L940C
       JSR L92E5
       DEC &C22B
       BNE L93FC
       LDA #&04
       STA &C22B
       JSR LA03A
       JMP L93FF
;;
.L93FC JSR LA036
.L93FF CLC
       LDA &B6
       ADC #&1A
       STA &B6
       BCC L93E3
       INC &B7
       BCS L93E3
.L940C LDA &C22B
       CMP #&04
       BEQ L9423
       LDA #&86
       JSR &FFF4  ;; Read cursor POS
       TXA
       BNE L9420
       LDA #&0B ;; Move back up a line of POS=0 !!!
       JSR LA03C
.L9420 JSR LA03A
.L9423 JMP L89D8
```

hoglet67, 2021-09-08:
This is tricky to fix, but there is a worl around.

If the host cursor is left in a column other than colomn 0, then the issue becomes there is sometimes an extra line feed at the end, which is much less noticable. This will only occur in 20 and 40 col modes, never in 80 col modes.

mincebert, 2021-09-10:
For interest more than helpful — does anything happen to support the redirector on the host side, or is it all on the copro side?  In other words, if anything on the host side calls VDU functions, does it just hit the normal host VDU driver?

mincebert, 2022-01-11:
I've just noticed dffadc1ebf728d4c67bbeb42e12d8bfff608d9eb still has a problem with this: it leaves the cursor overwriting the last line if the '(number of entries MOD 2) = 2' (i.e. 2, 6, 10, 14, etc.).

This seems to only affect 80 column or wider modes (e.g. MODE 0 or MODE 21).

hoglet67, 2022-01-11:
Yes, I was aware of this.

The fix for the *HELP MOS bug effectively reintroduced this bug.

One work around is to switch the host to an 80-column mode.

hoglet67, 2022-01-11:
I'll reopen in case I have a flash of inspiration!

hoglet67, 2022-01-13:
@mincebert I have a (hopefully) fairly robust fix for this, which you can try out here:
https://www.dropbox.com/s/t8wrd124k3h26i5/PiTubeDirect_20220113_1213_dmb.zip?dl=1

There's also been a change in the way VDU redirection works in the Native ARM Co Pro (specifically in *PIVDU 3 mode). I'm not sure which Co Pro you are using, so if you are only using the 6502 then just ignore this.

The *PIVDU 3 mode is really just used for testing the Pi and the Beeb implementations of things match. When entering this mode, It's advisable to immediately do a MODE command, so that both displays become synchronized. Otherwise it's easy to end up with a situation where this bug recurs.

Dave

mincebert, 2022-01-14:
I've tried this on the 6502 and ARM copros (under *PIVDU 2) and it looks good: sometimes you get the extra blank line but that's fine: I never got the last line overwritten again.  Thanks!

hoglet67, 2022-01-14:
Yes, the extra line is expected, and was present in the earlier fix.

Closing again...

closed 2022-01-14 by hoglet67