hoglet67/PiTubeDirect #146: tube_SWI_NumberFromString doesn't work for OS_WriteI

dp111, 2021-12-17:
in tube-swi.c tube_SWI_NumberFromString  doesn't work : 
 
      // Special case OS_WriteI as they are not in the handler table
      if (!strcmp(mod_name, "OS") && !strcmp(swi_name, "WriteI")) {
         reg[0] = x_flag + 0x100 + i;
         return;
      }

In the above code "i" is effectively uninitialised

closed 2021-12-17 by hoglet67

dp111, 2021-12-17:
Does this mean the OS SWI table could be extended to have OS_WriteI at the end and the special case isn't needed ?

hoglet67, 2021-12-17:
I initially did that, then changed to the current pattern for a reason I don't recall!

Looking at the code now, I don't see why we couldn't do that.

But possibly we have more pressing bugs, e.g. #147