hoglet67/PiTubeDirect #147: Possible Malloc Bug

hoglet67, 2021-12-17:
The GCC implementation of malloc places the heap at the _end symbol.

In our builds, this ends up about here:
```
0e55ff70 B _end
```
Looking at our memory map for a 512MB Pi:
```
# 0x00000000   0MB - Start of RAM
# 0x0D800000 216MB - ARM Basic (allow 4MB)
# 0x0DC00000 220MB - Stack space (allow 16 x 0.5MB)
# 0x0E400000 228MB - Pi Kernel (allow 4MB)
# 0x0E800000 232MB - Top of cached RAM
# 0x1F000000 492MB - Start of GPU Memory
# 0x20000000 512MB - Top of physical RAM on 256MB Pi Model
```
So that gives only 2.6MB of heap within cached memory.

This is what currenty uses dynamic memory:
```
./debugger/linenoise.c:    copy = malloc(len+1);
./debugger/linenoise.c:    cvec = realloc(lc->cvec,sizeof(char*)*(lc->len+1));
./debugger/linenoise.c:    char *new = realloc(ab->b,ab->len+len);
./debugger/linenoise.c:            line = realloc(line,maxlen);
./debugger/linenoise.c:        history = malloc(sizeof(char*)*(size_t)history_max_len);
./debugger/linenoise.c:        new = malloc(sizeof(char*)*(size_t)len);
./tube-commands.c:   uint8_t *a = calloc(sx * sy, 1);
./tube-commands.c:   uint8_t *b = calloc(sx * sy, 1);
./framebuffer/primitives.c:      sprite->data = malloc(size);
./framebuffer/fonts.c:      font->buffer = (uint16_t *)calloc(size, 1);
./rpi-aux.c:    tx_buffer = malloc(TX_BUFFER_SIZE);
./lib6502.c:  M6502 *mpu= calloc(1, sizeof(M6502));
./lib6502.c:  if (!registers)  { registers = (M6502_Registers *)calloc(1, sizeof(M6502_Registers));  mpu->flags |= M6502_RegistersAllocated; }
./lib6502.c:  if (!memory   )  { memory    = (uint8_t         *)calloc(1, sizeof(M6502_Memory   ));  mpu->flags |= M6502_MemoryAllocated;    }
./lib6502.c:  if (!callbacks)  { callbacks = (M6502_Callbacks *)calloc(1, sizeof(M6502_Callbacks));  mpu->flags |= M6502_CallbacksAllocated; }
```
To summarize that:
- linenoise - just the history I think, which is limited to 100 pointers
- tube-commands: the PILIFE command uses 2 * X * Y bytes (but frees it at the end)
- primitives: all sprite memory is malloced, and there can be 256 sprites
- fonts: all font memory is malloced, and there are 42 fonts, each using between 8K and 14K, so ~512KB at a guess
- the serial transmit buffer, currently 64KB
- lib6502 - non turbo - 64KB (memory) + 64KB * 3 * 4 (callbacks) = 64KB * 13 = 832KB
- lib6502 - turbo - 4x as much, so 3328KB

So the turbo version of lib6502 will use all of the cached heap!

We also have the following code in arm-cstubs.c
```
/* Increase program data space. As malloc and related functions depend on this,
 it is useful to have a working implementation. The following suffices for a
 standalone system; it exploits the symbol _end automatically defined by the
 GNU linker. */
caddr_t _sbrk(int incr)
{
  extern char _end;
  static char* heap_end = 0;
  char* prev_heap_end;

  if (heap_end == 0)
    heap_end = &_end;

  prev_heap_end = heap_end;
  heap_end += incr;

  return (caddr_t) prev_heap_end;
}
```

dp111, 2021-12-17:
Slightly related I did spot last night if BENCHMARK is defined in
performance.c 2Mbytes of stack is allocated ( ie. we overflow the stack)

Some of the mallocs could just become statically allocated e.g. serial
buffer.

The lib6502 stuff could easily be at fixed addresses from 0x00000000
upwards. e.g. items on 256K boundaries. It would probably make it faster
too.

dp111, 2021-12-26:
Unfortunately the fix means the heap now over laps the tables used by the 6502 Jitter

hoglet67, 2021-12-26:
Oh bugger.....

hoglet67, 2021-12-26:
These:
```
#define JITLET 0x0d000000
#define JITTEDTABLE16 0x0d100000
```
Can they just be moved down to 0x0c000000?

dp111, 2021-12-26:
I don't think so as that takes the branches out of range of the kernel, but
worth a try.

dp111, 2021-12-26:
Another option is the Jitter copies the functions it needs to closer to the
tables. I do think the Lib6502 should be using fixed memory areas just like
it does for the ram.

hoglet67, 2021-12-26:
> I don't think so as that takes the branches out of range of the kernel, but worth a try.

You're right, it didn't work

(I did spot that JITLET and JITTEDTABLE16 are defined in two places)

I've deleted the RC0 release until we have resolved this.

(there had only been one download)

hoglet67, 2021-12-26:
I wonder if the best best option is just to move the kernel down, so the 16MB heap can immediately follow it:
```
# 0x00000000   0MB - Start of RAM
# 0x0C800000 200MB - ARM Basic (allow 4MB)
# 0x0CC00000 204MB - Stack space (allow 16 x 0.5MB)
# 0x0D400000 212MB - Pi Kernel (allow 4MB)
# 0x0D800000 216MB - Heap (16MB)
# 0x0E800000 232MB - Top of cached RAM
# 0x0E800000 232MB - Frame Buffer (allow 8MB)
# 0x0F000000 240MB - Start of GPU Memory
# 0x10000000 256MB - Top of physical RAM on 256MB Pi Model
```
Then the Jitter can use 0x0C000000 and 0x0C100000 I think.

Dave

dp111, 2021-12-26:
Sounds good to me.

hoglet67, 2021-12-26:
I've just noticed there is some MMU code (cache.c) that is causing these tables to wrap.

How's that going to affect ARM Basic's use of that memory for the BASIC heap? (HIMEM will be &0C80000 and the BASIC heap will grow downwards).

(if this is an issue, it would also have been an issue before...)

dp111, 2021-12-26:
I don't think it is an issue as the wrapping only happens when the JIT is
initialised. I do need to check the exit code cleans up correctly.

hoglet67, 2021-12-26:
How big are each of the JIT tables? Your comments were a bit ambiguous.

dp111, 2021-12-26:
The jitter does clean up after itself. Jit comments updated and Jit tables
moved in latest commit

hoglet67, 2021-12-26:
Crap, we now have a big conflict....

Do you mind if I overwrite your commit?

The results should be pretty much the same.

hoglet67, 2021-12-26:
Note to self.... this is what happens when you try to rush to meet a deadline....

I'll finish this off tomorrow.

dp111, 2021-12-26:
More than happy for you to do what is required.

hoglet67, 2021-12-27:
I ended up keeping your commit and adding in the missing bits in a seperate commit.

The complete set of changes for this work is here:
https://github.com/hoglet67/PiTubeDirect/compare/2c1ea897bdd2...aa9b0703b8af3ae42

Please do check there is nothing I've missed.

I'm looking at the lib6502 memory allocation now....

hoglet67, 2021-12-27:
lib6502 fix here: 8a482108

commit 8d805a2e (2021-12-26): Add a 16MB heap (fixes #147)

Change-Id: I61e62c565c65adf7e7c7fed6ee6d00d370be2d54
commit aa9b0703 (2021-12-27): Move 16MB heap to after the kernel (fixes #147)

Change-Id: I6e4bbb4b2777f1131b427e90570aa0b2aba34979
closed 2021-12-27 by hoglet67