hoglet67/PiTubeDirect #148: Add to Call &2000 to some method to highlight which 65C02 is running.

dp111, 2022-01-04:


dp111, 2022-01-04:
Also add copro number to the PiVDU screen and the sphere REM statements

hoglet67, 2022-01-05:
How do the following look?

[image]
[image]
[image]
[image]

dp111, 2022-01-05:
Looks excellent to me.

hoglet67, 2022-01-05:
The splash screen is now redrawn when you switch Co Pros (previously it was only drawn once)

BTW, I tried cherry picking 2cc44faf and  0fa9f08f

When I did ?&FEF8=65 the system hung.

Any thoughts?

closed 2022-01-05 by hoglet67