hoglet67/PiTubeDirect #152: JIT 6502: Failure of Dormann Test Case 0x28 (self modiying code)

hoglet67, 2022-01-06:
Introduced in f98b642

Affects all Pi Models (I think... certainly 3 and 4)

[image]

Test Case 28:
```
0028 =                 >test_num = test_num + 1
6b92 : a928            >            lda #test_num   ;*** next tests' number
6b94 : 8d0032          >            sta test_case
                       >            ;check_ram       ;uncomment to find altered RAM after each test
                        
                        
                        ; OR
6b97 : a203                     ldx #3          ;immediate - self modifying code
6b99 : b560             tora    lda zpOR,x
6b9b : 8d1032                   sta ex_orai+1   ;set ORA # operand
                                set_ax  absORa,0
                       >            load_flag 0
6b9e : a900            >            lda #0             ;allow test to change I-flag (no mask)
                       >
6ba0 : 48              >            pha         ;use stack to load status
6ba1 : bd6732          >            lda absORa,x    ;precharge accu
6ba4 : 28              >            plp
                        
6ba5 : 200f32                   jsr ex_orai     ;execute ORA # in RAM
                                tst_ax  absrlo,absflo,0
6ba8 : 08              >            php         ;save flags
6ba9 : dd7332          >            cmp absrlo,x    ;test result
                       >            trap_ne
6bac : f003            >        beq skip3199
                       >        trap            ;failed not equal (non zero)
6bae : 208574          >        jsr report_error
```
Modifies the operand of ORA #immediate

dp111, 2022-01-06:
The both of build in Dormann tests (6502 and 65C02) pass for me on a Pizero
and Pizero2

hoglet67, 2022-01-06:
Interesting....

What git commit of PiTubeDirect are you running?

What size is you D6502 file? I'm wondering if the self modifying tests are disabled.

hoglet67, 2022-01-06:
I'm using the tests here:
https://github.com/hoglet67/6502_65C02_functional_tests/tree/master/beeb

(At least, I think I am....)

dp111, 2022-01-06:
It might be worth updating the Dormann tests I notice there is a new
version with a couple of outstanding pull requests

I'm using *GO 3400 and building program.c with either version

hoglet67, 2022-01-06:
I've retested with latest hognose-dev and I'm only seeing the Dorman failure on the Pi 4 (the Pi Zero 2 passes)

I must have made a mistake when testing before.

dp111, 2022-01-06:
Is this your dorman tests or the built in version ?

hoglet67, 2022-01-06:
With my dorman 6502 tests (disk 497):
- Pi Zero 2 Passes
- Pi 4 fails (test case 28 @ &6BB0; 9 more failures of same test case)

With dorman 6502 tests in programs.c
- Pi Zero 2 Passes
- Pi 4 fails, but it's a different test case (test case 2A @ &6F08; many many more failures)

Just trying to work out what the difference is.

hoglet67, 2022-01-06:
Really struggling to work out what version of the dormann 6502 tests we included (on 26th Oct 2016).

Comparing it to my dormann repo then, it looks quite different.....

... after lots of messing, it's this version:

https://github.com/Klaus2m5/6502_65C02_functional_tests/tree/04de059478c9b51404df92267fd13242d75bb276

And the assembly listing so we can understand what's going wrong...
[D6502.txt](https://github.com/hoglet67/PiTubeDirect/files/7825030/D6502.txt)

This 6f08 failure is also down to self modifying code in the decimal mode ADC/SBC test:
```
6ef6 : 28                       plp
6ef7 : 08                       php             ;save carry for next add
6ef8 : a558                     lda sb2
6efa : 8d006f                   sta chkdsbi     ;self modify immediate
6efd : a553                     lda ad1
6f00 =                  chkdsbi = * + 1         ;operand of the immediate SBC
6eff : e900                     sbc #0          ;perform subtract
6f01 : 08                       php
6f02 : c555                     cmp adrl        ;check result
                                trap_ne         ;bad result
6f04 : f003            >        beq skip3338
                       >        trap            ;failed not equal (non zero)
6f06 : 201b73          >        jsr report_error
```
Later versions of the Dorman tests don't use self modifying code in the decimal mode ADC/SBC test case.

hoglet67, 2022-01-06:
> Later versions of the Dorman tests don't use self modifying code in the decimal mode ADC/SBC test case.

This is incorrect; they do still use self modifying code for ADC/SBC #imm but the code has changed, and for some reason is passing on the Pi 4 (or these tests are not being run).

Sorry this is so confusing!

hoglet67, 2022-01-06:
What's very strange here is both the binary and decimal ADC/SBC #imm tests use the same self-modifying code pattern, but only the decimal mode ones are failing, and then only on the Pi 4, and then only with the older version of the dorman tests.

dp111, 2022-01-07:
I think this little hack fixes the issue with Pi4
: a27680bfd9c19d2fc00d783a22532cccdc52849c

dp111, 2022-01-07:
Correct fix : c13261916d4789a481b58fb57802550a06050bc4

hoglet67, 2022-01-07:
That brilliant Dominic; thanks so much for sorting this!

I've cherry picked  a27680b and c132619 into hognose-dev

Both versions of the D6502 tests now pass on the Pi Zero 2 and the Pi 4.

The Elite glich (#151) is also resolved.

Just need to try to figure out the Bank Select issue (#150) now.

closed 2022-01-07 by hoglet67