hoglet67/PiTubeDirect #153: LIB6502: Latest Dorman 6502 tests fail on new shift tests

hoglet67, 2022-01-07:
We are now using the 2020/01/05 version of Klauses tests, which include extra shift test cases.

We see ~8 failures (test cases 1E, 1F, 20 and 21)

Here's the first:
[image]

Will investigate now...

hoglet67, 2022-01-07:
```
001e =                 >test_num = test_num + 1
5c2c : a91e            >            lda #test_num   ;*** next tests' number
5c2e : 8d0032          >            sta test_case
                       >            ;check_ram       ;uncomment to find altered RAM after each test
                        
                        
                        ; shifts - zeropage
5c31 : a205                     ldx #5
5c33 :                  tasl2
                                set_z zps,0
                       >            load_flag 0
5c33 : a900            >            lda #0             ;allow test to change I-flag (no mask)
                       >
5c35 : 48              >            pha         ;use stack to load status
5c36 : b559            >            lda zps,x    ;load to zeropage
5c38 : 8552            >            sta zpt
5c3a : 28              >            plp
                        
5c3b : 0652                     asl zpt
                                tst_z rASL,fASL,0
5c3d : 08              >            php         ;save flags
5c3e : a552            >            lda zpt
5c40 : dd2132          >            cmp rASL,x    ;test result
                       >            trap_ne
5c43 : f003            >        beq skip2401
                       >        trap            ;failed not equal (non zero)
5c45 : 208574          >        jsr report_error
                       >
5c48 :                 >skip2401
                       >
5c48 : 68              >            pla         ;load status
                       >            eor_flag 0
5c49 : 4930            >            eor #0|fao         ;invert expected flags + always on bits
                       >
5c4b : dd3932          >            cmp fASL,x    ;test flags
                       >            trap_ne
5c4e : f003            >        beq skip2404
                       >        trap            ;failed not equal (non zero)
5c50 : 208574          >        jsr report_error
```

hoglet67, 2022-01-07:
This case is testing ASL zp (opcode 06)

The test values are stored at  0059:
```
0059  80 01 c3 82 41 00
```
The expected results are stores at 3221:
```
3221  00 02 86 04 82 00
```
The expected flags are stored at 3239:
```
3239  03 00 81 01 80 02
```
The case that's failing is X=00

The expected flags at 03

The actual flags are 01

So the zero flag is not being set!

Looking at the code:
```
#define asl(ticks, adrmode)			\
  adrmode(ticks);				\
  {						\
    byte B= getMemory(ea);                      \
    unsigned int i= B << 1;                     \
    putMemory(ea, (uint8_t)i);				\
    fetch();					\
    setNZC(i & 0x80, !i, i >> 8);		\
  }						\
  next();
```
The problem is that i is greater than 8 bits, so still includes the 9th (carry bit) when testing for zero.

hoglet67, 2022-01-07:
Seeing a total of 8 similar failures:
- ASL zp (2 failures)
- ASL abs (2 failures)
- ASL zp, X (2 failures)
- ASL abs,X (2 failures)

closed 2022-01-07 by hoglet67

BigEd, 2022-01-07:
Very interesting... the upstream source is, I think, on Ian Piumarta's site at
https://www.piumarta.com/software/lib6502/

Other projects which might be affected:
https://github.com/ZornsLemma/lib6502-jit
https://github.com/ShonFrazier/lib6502
and their forks (including https://github.com/MorrisMA/lib6502 and mine)

commit d5109631 (2022-01-07): lib6502 Co Pro: fix Z flag bug in ASL <memory> (fixes #153)

Change-Id: Idcea4a7b3e14afd51f3ec3a732d7466013ee0bda