hoglet67/PiTubeDirect #154: Crash in Hognose (various versions, inc. rc1) when temperature hits above around 55ºC but not Gecko

mincebert, 2022-01-07:
I'm having an erratic crashing problem that appears to just affect Hognose but the situation is not clear cut...

Over the past few months, I've been trying out various versions of Hognose: betas, release candidates and the occasional intermediate commit (to test particular fixes), compiled under Raspberry Pi OS. Coincidentally, I've been having my Master crash sometimes, and I've been trying to get to the bottom of what's causing it. By "crash", I mean it hangs or does something odd (like print an error loading a file - often 'Channel') — jabbing Ctrl+Break recovers things (although sometimes hangs after the 'Acorn 1770 DFS', presumably during language transfer across the Tube, and needs a second press).

The crash seems to take a little time to occur, from a cold power up: I tend to get 20+ minutes of good operation and then, once it starts to happen, it happens much more frequently. When it does, I push Ctrl+Break, load everything up from scratch again (I deliberately avoid OLD, in case anything's corrupt in BASIC) and then it will do the same within 5 minutes or so.

I had a similar issue about a year ago which seemed to only occur when I had a couple of external ROM cartridges in the slots on the Master and Dave helped resolve that — I think it was a combination of things: I changed the internal Master level shifter (from a Sundby to IFEL) + reseated my Retroclinic MultiOS as that was a bit askew and Dave tinkered with PiTubeDirect. It was then all working.

Thinking it was something similar, I've gone on a spree of reseating and changing things: I've removed all the ROM cartridges — no improvement. I've also swapped the Retroclinic MultiOS for a new IFEL one, as I had a GoSDC in the socket for bank 8 and the two were physically overlapping, so I had to build up the MultiOS with a few turned pin sockets to get it higher: the IFEL is smaller and avoids this. I've also removed the GoSDC in favour of a Pi1MHz with one of Ken's level shifters and a Pi Zero underneath. This was all with the aim of removing the amount of stuff inside the machine, in case it was causing the problem (and the Pi1MHz avoids wasting SRAM banks, loading in the special GoSDC version of ADFS), but it made no difference.

I've also taken the internal IFEL PiTubeDirect level shifter out and just gone over the solder joints, just in case there was a bad connection (as it was shipped flat and I had to originally solder them on), but still no improvement.

I then wondered if Hognose was causing the issue as things only started happening in the past few months, so I stuck the Gecko Fix 1 card back in the Pi (3A+) and I haven't had any issues. Swapping the Hognose rc1 card back in causes the problem to reoccur again (including various builds further along from that, up to the updates last night (but not this evening's).

Changing the copro type doesn't affect things: I get the same crashing under copros 0, 2 and 24.

I've not had any problems with the machine with the Tube disabled (*CONFIGURE NOTUBE), but the PiTubeDirect still in there physically.

I did have the Pi framebuffer enabled, but I've tried turning that off and it makes no difference.

I've tried powering the machine off, waiting a few seconds and powering on but the problem seems to remain and reoccur fairly quickly.

OK, now the particularly odd bit: the problem doesn't seem to occur if I leave the lid off, although that's a bit annecdotal — it sometimes does but only if the lid was removed fairly recently (in the past 2-3 minutes). Also, I left it running a test whilst downstairs this evening and it ran for about 3 hours OK; I came upstairs and replaced the lid and it crashed about 15 minutes later. So it could be a temperature thing although nothing inside feels particularly warm, except the PSU.

I don't know what a good test is but I thought something that exercised the Tube bus interactively, so wrote a simple assembly program to (inefficiently) compare two identical files using BGET. This program runs repeatedly and prints a "." every 256 or so bytes so I can spot it crash easily. If it crashes, it either hangs in the middle of printing the dots, or crashes out printing "Channel". This I can leave running for hours.

I've tried fiddling with tube_delay - setting it to 5 and then 10, but with no success.

So — this could be something getting warm but why does it not appear to affect Gecko? Sorry for this being quite vague but I'm not sure where to go next with this! Any suggestions? Thanks in advance!

I should also say I'm running Hognose rc1 (I think - might be a slightly later build, but not as up to date as the Master) on my B with a Zero 2W and I haven't had a single issue with that: that machine is much more loaded, with an Integra B and lots of EPROMs and EEPROMs!

mincebert, 2022-01-07:
Stardot responses:

---

 by dp11 » 07 Jan 2022 03:24

Thanks for the detailed information. A few questions

Has the master had the extra PSU caps replaced ? Not just the usually 3 caps ?

Can you possibly after 3 hours with the lid on and off give the pi's temperature. Power cycle, Ctrl break , old and list will give the pi's temperature.

You might try a little over voltage in config.txt e.g. over_voltage =2 to see if that helps.

---

by BigEd » 07 Jan 2022 08:58

Just to summarise and check: this is a Master with an internally fitted Pi 3A+ using an IFEL level shifter?

Good idea to read off the Pi's temperature!

I posted a tiny tube stress test here. That might be worth a go.

A couple of ideas - are these worth trying Dominic?

If the shifter has a power jumper which allows you to power the Pi over USB that might possibly be worth trying. But never use USB power without changing the jumper. If you have an official Pi power supply, even better: they have thick wires and provide a slightly high voltage.

Swapping in your Pi Zero for the 3A+ might be a useful test too. Maybe. It seems you've already demonstrated that Gecko works and Hognose rc1 is showing symptoms.

mincebert, 2022-01-07:
Thanks both, for the suggestions...

Yes, the Pi is a 3A+ on the IFEL internal level shifter board (which I soldered the pin blocks to, as it was flat for posting). When I bought the Master about 3 years ago, I was told it had been recapped but I don't know what this entailed. If you can tell me what to look for, I can remove the PSU and have a look, but...

I switched things on about 09:50 this morning and the temperature on the Pi (as reported in the REM in the Sphere program) was 23.6ºC. I used my multimeter to measure the voltages on the pins in the level shifter: 4.963V, the spade terminals behind the level shifter: 4.983V, and front and rear on the left side: 4.999V and 5.023V.

I then left my test running with the lid on. The machine started to misbehave consistently about 10:50, an hour later (i.e. it crashed, I did Ctrl+Break and restarted the test and it crashed within a couple of minutes).

The temperature has now risen to 55.8ºC. The voltage on the level shifter is 4.945V, spade terminals behind 4.971V and the front/rear left: 4.997V and 5.021V.

Taking the lid off it crashed again after a couple of minutes, so I powercycled and checked the temperature: 52.4ºC (it does seem to cool very quickly with the lid off, though, so it was probably a few degrees higher, when it did crash). I then restarted the test and didn't get any trouble for a few minutes; powercycled and it was 49.9ºC. I've put the lid back on and restarted — it crashed again and the temperature was 56.4ºC.

I've now quickly whipped off the lid and stuck the Gecko fix 1 release card back in and left it running for 20 minutes and no issues; the temperature was 59.1ºC when I stopped it. Swapping the card back again and I've had a crash within a couple of minutes with the temperature 56.9ºC.

I'm assuming that, given these voltages, the PSU should be fine, but I have a jumper on the level shifter and have tried that with an official Pi 3 power supply: I still get crashes with that; temperature 56.4ºC.

So it does look like temperature is a major factor, but I'm unclear on why it doesn't affect Gecko!

I'll try BigEd's Tube stress test to see if it trips the problem more or less than mine. I haven't tried over_voltage=2 yet and will test that too — is that going to have any effect, given the test with the separate PSU?

mincebert, 2022-01-07:
Mince wrote: ↑07 Jan 2022 13:32
I'll try BigEd's Tube stress test to see if it trips the problem more or less than mine. ...
That seems to crash with similar consistency/frequency: took a few minutes to get up to temperature with the lid back on, then crashing after a minute or two, after each power cycle.

One time it crashed displaying a large negative number and the other it just hung in the middle of printing the 1s.

That was under Hognose rc1 + updates; I'll leave it going with Gecko.

mincebert, 2022-01-07:
BigEd wrote: ↑07 Jan 2022 13:50
Here's a thought: not only is Hognose a new codebase, but also it has been built with a newer toolchain. I wonder, if Gecko were built with the newer toolchain, would it show the same symptoms? (Or, if possible, if Hognose were built with the older toolchain, but it mightn't be possible.)

Mincebert> That's easy for me to try as I can just checkout a Gecko release and rebuild with the same toolchain and we'll see what happens.

BigEd wrote: ↑07 Jan 2022 13:50
The overvoltage is still worth trying - the CPU's internal supply rail will not be affected by minor changes on the 5V rail, but all else being equal, the CPU will run better at a higher internal supply. It will also put out even more heat, which might or might not scupper it.

Mincebert> Thanks — I'll try that next, when I get back from my bike ride (leaving Gecko on your test whilst I'm out).

mincebert, 2022-01-07:
by dp11 » 07 Jan 2022 13:56

Just to check when you are using hognose you are also using the bootcode.bin, start_cd.elf fixup.dat from hognose ? if so can you try these files with the Gecko release.

mincebert, 2022-01-07:
by hoglet » 07 Jan 2022 13:58

BigEd wrote: ↑07 Jan 2022 13:50
Here's a thought: not only is Hognose a new codebase, but also it has been built with a newer toolchain. I wonder, if Gecko were built with the newer toolchain, would it show the same symptoms? (Or, if possible, if Hognose were built with the older toolchain, but it mightn't be possible.)

Hoglet>>>

My gut feel is this is this is very unlikely to be a software issue.

I suspect it is related to this issue:
https://github.com/hoglet67/PiTubeDirect/issues/70

Is there any correlation with screen mode?

It might also be better to create a new issue for this in github, with the info you have posted in this thread.

Edit: I'd also like to see gecko with the results of with the hognose version of the firmware blob files:
bootcode.bin  fixup4cd.dat  fixup_cd.dat  start4cd.elf  start_cd.elf

mincebert, 2022-01-07:
hoglet wrote: ↑07 Jan 2022 13:58
My gut feel is this is this is very unlikely to be a software issue.

I suspect it is related to this issue:
https://github.com/hoglet67/PiTubeDirect/issues/70

Is there any correlation with screen mode?

It might also be better to create a new issue for this in github, with the info you have posted in this thread.

Edit: I'd also like to see gecko with the results of with the hognose version of the firmware blob files:
bootcode.bin  fixup4cd.dat  fixup_cd.dat  start4cd.elf  start_cd.elf

Mincebert>>>

Thanks for the reply -- I've made issue 154 in GitHub: https://github.com/hoglet67/PiTubeDirect/issues/154, and pasted this all in there. I'll report further things there, after this (until we hopefully resolve it).

This is the same machine as issue 70 but that was working after the changes at the time.

I haven't noticed any correlation with screen mode but I'm pretty much always in MODE 7, certainly during my test program and BigEd's. I could try other modes but they were less reliable than MODE 7, before. Do you want me to try it in other modes to see if it makes it more or less crashy?

I've copied the files off the Gecko card and then the 5 firmware blob files over the top from Hognose and set the machine running the test. I'm just waiting for it to warm up: it's currently at 55.6ºC but no crash so far, so shouldn't be long, if it is going to.

mincebert, 2022-01-07:
Gecko code but with Hognose's firmware blobs (as per commit 6228eef) has been running over 30 minutes, temperature is 59.1ºC and no crashes.

Just for completeness, I'm going to try compiling from tag gecko-fix1 (dc67fd8) and see what that does.

I'll then try different screen modes (initially with Gecko in MODE 0) and see what that does (I assume that's the best place to start - to see if it starts to crash, rather than trying Hognose in MODE 0 to see if it's more or less crashy)?

hoglet67, 2022-01-07:
> Just for completeness, I'm going to try compiling from tag gecko-fix1 (dc67fd8) and see what that does.

This only affects the Pi 4; there is no change on the other Pi models.

hoglet67, 2022-01-07:
Do you have any freezer spray?

mincebert, 2022-01-07:
> > Just for completeness, I'm going to try compiling from tag gecko-fix1 ([dc67fd8](https://github.com/hoglet67/PiTubeDirect/commit/dc67fd8a7588799586ef4cbb5a4a9931061fec9f)) and see what that does.
> 
> This only affects the Pi 4; there is no change on the other Pi models.

I meant more as a version of Gecko (as opposed to Hognose) rather than a different version of Gecko (following BigEd's suggestion; my working Gecko release is already fix1, but your build) — I just thought it was worth a try (as I can leave that going whilst I'm still 'at work').

mincebert, 2022-01-07:
> Do you have any freezer spray?

No — the closest I can get is an ice pack in the freezer I could hold against specific things.  What did you have in mind?

hoglet67, 2022-01-07:
There clearly is a temperature dependency here. It would be interested to determine if really is the Pi temperature that matters. But I wouldn't recommend using an ice pack!

hoglet67, 2022-01-07:
It would also be interesting to see if the issue is Pi 3A+ specific, or whether a Pi Zero shows it as well.

mincebert, 2022-01-07:
I have another Pi 3A+ on a Retroclinic board with a ribbon cable that I use with the ATI cartridge on my Electron.  It's also in a plastic box I can take the lid off and on of, so I can separately let that get hot or not, relative to the Master.

Obviously it would have to be connected to the Tube connector underneath, which might have different behaviour.  Also, without the Pi running inside the case, it might not get the Master so hot inside but I could measure that compared to now and then with the temperature probe in my multimeter.

We could start by trying to get it to also crash and then work from there.

I can also swap in a Pi Zero — shall I start with that inside the Master as it's the easiest thing.

mincebert, 2022-01-07:
gecko-fix1 built from the same toolchain didn't crash, despite reaching 60.1ºC.

I've now popped in a Pi Zero on the internal board and I've left it running the Hognose and BigEd's test to warm up.

mincebert, 2022-01-07:
No problems with the same Hognose card on a Pi Zero, despite reaching 57.4ºC so far.  I'll leave it going so more.

The general temperature inside the case (with the lid on) measured in the space between the level shifter and the north end of the keyboard is 35ºC, according to my probe.

Dinner in a mo: I'll try the Pi 3A+ from my Electron later, inside the Master again, to see if it's a general 3A+ problem.  I'll then move on to the external box test.

hoglet67, 2022-01-07:
This is great data... I'm running Ed's stress test on my 3A+ with latest hognose-dev build and the lid closed on the Master.

It's  been running an hour now, up to 58 degrees C, and 4.92V across the Pi.

mincebert, 2022-01-07:
No crash after about 3 hours on the Pi Zero — didn't get any hotter.  I've swapped in the 3A+ from my Electron to see if it's something specific about the one in the Master, left it running the test...

mincebert, 2022-01-07:
OK.  That was quick — the other 3A+ is crashing in the same way: 56.4ºC.

I've moved it outside the BBC case into the external box and left it running the test.  The lids are on the Master and the Tube box just to see if I can still get it to crash.

(One change here, aside from moving the Tube is that I can't connect the power ot the Pi1MHz shifter, so that is connected but not running.)

mincebert, 2022-01-07:
Crash with the 3A+ in the external shifter.  Temperature was 56.9ªC.

I've now set it going again with the lid off the level shifter but the lid still on the Master.

mincebert, 2022-01-07:
No crash with the lid off the external box: temperature only got up to 48.3ºC.

I'm now trying with the Master lid off but the level shifter box lid on.

It seems pretty consistent about the temperature going over around 55-56ºC.  Is there some sort of throttling or other behaviour built into the Pi?

Also, is there firmware in the Pi aside from what's loaded from blobs on the SD card?  I'm wondering as neither of these Pis have been used for anything other than PiTubeDirect, so if something like Raspberry Pi OS would have updated something, they wouldn't have ever had the chance.

mincebert, 2022-01-07:
Righty — that crashed, so the BBC has the lid off but the Pi has the lid on and was 55.8ºC when it first crashed.

(Bear in mind all these temperatures are a few seconds or minutes after the crashes, when I notice and power cycle the machine to get it.)

I think the old test left to do is try a MODE other than 7 with Gecko to see if it will crash.

I think, however, that this has shown it doesn't matter whether the Pi is on the internal or external connector.  I've also tried two level shifter boards.  Do you think it's OK to do all further testing back using the internal level shifter + Pi?

Just for completeness, I have an IFEL MMFS Turbo board on the User Port and a Pi1MHz with one of Ken's level shifters.  The only other stuff is the IFEL MultiOS, an Acorn Econet board (0194 005 Issue 1 - not sure what version that is: I've not found much about it) and a 27128A EPROM in bank 8.

mincebert, 2022-01-07:
BTW, I was reading this: https://www.raspberrypi.com/news/thermal-testing-raspberry-pi-4/ — does that have any bearing (it mentions the 3B+, too, which I assume is similar to the 3A+).

dp111, 2022-01-08:
From this page :
https://www.raspberrypi.com/documentation/computers/config_txt.html

temp_soft_limit

*3A+/3B+ only*. CPU speed throttle control. This sets the temperature at
which the CPU clock speed throttling system activates. At this temperature,
the clock speed is reduced from 1400MHz to 1200MHz. Defaults to 60, can be
raised to a maximum of 70, but this may cause instability.

mincebert, 2022-01-08:
Is it worth setting that to 65 to see if the problem doesn’t occur, or at least the Pi gets hotter before it does?

dp111, 2022-01-08:
I think that would be worth it maybe even 70C

mincebert, 2022-01-08:
Another data point — I get the same crash on my Electron with ATI and the same Pi 3A+ with Hognose.  The only change was to set elk_mode=1.  Temperature was 57.5°C.

(When I’m using the external box, I put a blanket over it to get it to warm up more quickly.)

I’m now trying gecko-fix1 on the Electron, too.  After that, I’ll try adjusting temp_soft_limit back on the Master.

hoglet67, 2022-01-08:
That's great, because it completely exonerates the particular Master.

Let me see if I can get my Pi 3A+ above 60°C; I don't think I have managed that so far.

mincebert, 2022-01-08:
The Electron got up to 67.5°C wi5 gecko-fix1 and no crash before I stopped it.

I’ve gone back to the Master with the internal Pi and Hognose with temp_soft_limit=65 now.  I’ll see what it gets up to before a crash (if it does).

hoglet67, 2022-01-08:
OK, I think I've reproduced this with a hair dryer!

I'm running the debug build and Co Pro 16 so I can keep typing info on the console to see the temp, and use the debugger when it crashes.

As soon at the temp hit 60°C, stress test hangs, repeatable multiple times.

When it "hangs" the debugger is still running, and it's looping here:
```
fb11 2c fa fe bit FEFA
fb14 50 fb    bvc FB11
fb11 2c fa fe bit FEFA
fb14 50 fb    bvc FB11
fb11 2c fa fe bit FEFA
fb14 50 fb    bvc FB11
fb11 2c fa fe bit FEFA
fb14 50 fb    bvc FB11
fb11 2c fa fe bit FEFA
fb14 50 fb    bvc FB11
```
That's in the OSWORD code:
```
\ OSWORD - Various functions
\ ==========================
\ On entry, A =function
\           XY=>control block
\
.osWORD
STX &F8:STY &F9          :\ &F8/9=>control block
TAY:BEQ RDLINE           :\ OSWORD 0, jump to read line
PHA:LDY #&08
.LFB09
BIT TubeS2:BVC LFB09     :\ Loop until Tube R2 free
STY TubeR2               :\ Send command &08 - OSWORD
.LFB11
BIT TubeS2:BVC LFB11     :\ Loop until Tube R2 free
```
So it looks like somehow the tube protocol between the client and host is getting out of step, and the client has sent a message, but for some reason the host is not reading it.

I'm going to repeat this with the ICE-65C02 debugger in the host, so I can examine the tube FIFO state from the host side.

hoglet67, 2022-01-08:
This time is was fb38 on the client:
```
fb38 2c fa fe bit FEFA
fb3b 50 fb    bvc FB38
```
FEFA (R2 status on the client) shows BF (10xxxxxx)

On the host it is:
```
03.789810 : 05E0 : 2C E2 FE : BIT $FEE2
03.789814 : 05E3 : 50 FB    : BVC $05E0
```
And FEE2 (R2 status on the host) reads: BF (10xxxxxx)

Bit 7 = 1 means data is available
Bit 6 = 0 means FIFO is full

So somehow both sides are in the middle of sending a message to the other side, and are deadlocked waiting for the other side to make space.

This situation should never arise on a error free channel.

Here's a more complete dump of the FIFO state:

Host:
```
FEF0 - 4F (01xxxxxx)
FEF2 - BF (10xxxxxx)
FEF4 - FF (11xxxxxx)
FEF6 - 7F (01xxxxxx)
```
Client:
```
FEF8 - 4F (01xxxxxx)
FEFA - BF (10xxxxxx)
FEFC - 7F (01xxxxxx)
FEFE - 7F (01xxxxxx)
```

mincebert, 2022-01-08:
No crash yet, with the temperature up to 59.1°C, which isn’t over 60, but I don’t think I could have got it that high before.

I’ll keep running it, but I guess I can probably stop now, unless you want to confirm it’s related to the soft threshold?

If it is, I assume it’s some sort of race condition caused by the slower/erratic clock getting things out of step?

hoglet67, 2022-01-08:
Here's another run, another hang....

According to the tube analyzer, the host did:
```
R2: OSWORD: A=01 
R2: OSWORD response: BLOCK=00 00 ff 20 f7 
R2: OSWORD: A=02 BLOCK=00 00 ff 20 f7 
R2: OSWORD: A=01 
R2: OSWORD response: BLOCK=00 00 ff 20 f7 
R2: OSWORD: A=02 BLOCK=00 00 ff 20 f7 
R2: OSWORD: A=01 
R2: OSWORD response: BLOCK=00 00 ff 20 f7 
<deadlocked>
```

Client ended up at:
```
fb67 10 fb    bpl FB64
fb64 2c fa fe bit FEFA
```
Which is part of:
```
.WordRecvLow
LDY WordLengthsRx-1,X    :\ Get receive length from table
CPX #&15:BCC WordRecv    :\ Use this length for OSWORD 1 to &14
LDY #&10                 :\ Receive 16 bytes for OSWORD &15 to &7F
.WordRecv
BIT TubeS2:BVC WordRecv  :\ Wait for Tube R2 free
STY TubeR2               :\ Send receive block length
DEY:BMI LFB71            :\ Zero of &81..&FF length, nothing to receive
.LFB64
BIT TubeS2:BPL LFB64     :\ Wait for Tube R2 data present
LDA TubeR2:STA (&F8),Y   :\ Get byte to control block
DEY:BPL LFB64            :\ Loop for number to receive
.LFB71
LDY &F9:LDX &F8:PLA      :\ Restore registers
RTS
```
A bit more client state:
```
>> regs
       A = f7
       X = 01
       Y = 00
       P = N:0 V:1 *:1 B:0 D:0 I:0 Z:0 C:0 
       S = e9
      PC = fb64
```
X=1 indicating the client is in the middle of an OSWORD A=01 call, where the expected response block length is 05

This response is being written to a buffer pointed to by f8/f9:
```
>> rd f8
Rd Mem: 00f8 = 2a ( * )
>> rd f9
Rd Mem: 00f9 = 00 ( . )
>> m 002a
002a  50 f7 20 00 00 00 84 00 00 00 00 01 01 50 04 04  P. ..........P..
>> 
```
The block that was sent was: 00 00 ff 20 f7 

The block that was received was 00 00 20 f7, and the client is waiting for one more byte (which will overwrite the 50)

But that byte never comes, because the host has already sent it.

Somehow the 0xFF byte in the message was dropped.

This glitch corresponds to the Pi clock changing from 1400MHz downto 1200MHz.

(I've also seen it happen when the clock changes from 1200MHz back up to 1400MHz as the temperature falls again)

hoglet67, 2022-01-08:
Robert, you could try setting temp_soft_limit=50, that should bring on the failure much more rapidly.

What I'm thinking is happening is that the soft temp limiting is being handled by the firmware blob on GPU core 0. When this runs, it possibly causes GPU core 1 which runs our tube handler to block for long enough to miss data on the tube.

If the same firmware block is used on Gecko, and the soft temp limit is reached, then it also should show this issue.

hoglet67, 2022-01-08:
The main reason you don't see this in Gecko is that in that release only one GPU core is used. The tube handler code effectively replaces the firmware blob. So no soft thermal throttling occurs at all.

In Hognose we run both GPU cores, so soft thermal throttling can occur.

To add to the confusion, in the past the default temp_soft_limit value was 70°C, but in the firmware blob included with Hognose this now seems to be 60°C.

If you try temp_soft_limit=70 it's unlikely that temperature will ever be reached.

Maybe we should include that in the config file Dominic?

Alternatively, you could just underclock your Pi 3A+ to 1200MHz.

Dave

hoglet67, 2022-01-08:
BTW, the crash is not guaranteed to happen.

When the Pi clock switches between 1200MHz and 1400MHz (as the temp passes through the 60°C) threshold, it looks like there is a short period where both GPU cores pause (maybe waiting for the clock PLL to lock?)

If this pause happens to correspond to a tube data transfer, then data will be lost and you'll get a crash of some sort.

If this pause happens to correspond to an idle period on the tube, then you might get away with it, then the Pi will happily run forever at > 60°C. But if the system cools down there is a second opportunity for a crash.

At least, I think that's what's happening.

hoglet67, 2022-01-08:
Here's an interesting experiment to see how long the glitch lasts:
[image]
[image]
[image]

This loops reading &FEE0 on the host until the default value (&4E) is no longer seen, then it dumps 256 samples into memory.

It looks like only one sample (the first) is corrupt.

So this "outage" as the Pi clock changes lasts < 10us.

Dave

mincebert, 2022-01-08:
That fits with what I’m seeing.  Lowering the soft limit to 50 caused a crash at 51.5°C.  If I then restart, it doesn’t appear to crash, because it’s already over 50°C.  If I let i5 cool down, I get a crash again (obviously I have to turn it off for a bit - it doesn’t usually drop when it’s running).

mincebert, 2022-01-08:
I suppose whatever setting is there by default just has to ensure the throttling never kicks in (unless the Pi is about to melt itself) and has to be reliable for most users (if  people want to increase it with some minor risk, that’s up to them), but I think it’s your choice,

I think 1.2GHz is probably always fast enough for me, so I can just stick arm_freq=1200 in?  I assume that would stop it doing anything at 60°C because the speeds are the same - I’ll try with that the soft limit at 50 and start below that temperature but, as you said, it’s not always going to fail, so we’ll need to try a few cycles.

This only affects the 3 and 3+ we think?  Do we need a default override for the 3 too?  I have one of those to try but it sits on top of a wall cabinet in the kitchen, acting as my inbound SSH gateway and getting the temperature via BLE from my home sensors!

hoglet67, 2022-01-08:
The temp_soft_limit thing only affects the 3A+ and 3B+ (as they are overclocked by 200MHz by default)

With my test program (above) I can also induce this glitch about 1 time in 5 by running the info command:
[image]

The info command makes lots of calls to the Firmware Blob, so it's not surprising that it's disruptive.

I haven't yet managed to induce a failure in Ed's stress test program by running info.

mincebert, 2022-01-08:
So the difference between the 3 and 3+ is just this overclock to 1400MHz and slowdown to 1200, when we hit temp_soft_limit, but they should otherwise generate the same amount of heat?

If so, I guess we only need to limit things on the 3+?  Also, If I’m going to set arm_freq=1200, I assume I’m basically just getting a regular 3.

hoglet67, 2022-01-09:
The SoC in the 3A+/3B+ is in a new packages with a better thermal spreader, so it should run cooler.

I've just done some quick current measurements on the Pi 3A+.

I measured in Co Pro 15 (Native ARM) running ARM Basic and REPEAT UNTIL FALSE

- Gecko - 340mA
- Hognose dev latest - 330mA
- Hognose dev latest + arm_freq=1400 - 330mA
- Hognose dev latest + arm_freq=1200 - 270mA
- Hognose dev latest + arm_freq=1000 - 260mA

Enabling/Disabling the VDU on Hognose makes no difference to the current consumption.

I'm going to update the config.txt file to include arm_freq=1200 for the 3A+/3B+ as this saves ~20% of the power.

I might also try to take some temperature measurements in a closed Master.

Dave

dp111, 2022-01-09:
Just to add pre VDU driver the HDMI output was disabled which also
would have saved a little bit of power.

hoglet67, 2022-01-09:
Here's a couple of quick temperature graphs, measured over 90 minutes with the Master lid closed.

There is about a 5°C difference between 1200MHz and 1400MHz:
[image]
[1400.txt](https://github.com/hoglet67/PiTubeDirect/files/7835313/1400.txt)
[1200.txt](https://github.com/hoglet67/PiTubeDirect/files/7835314/1200.txt)

The room temperature was 18°C and the Beeb case temperature was 32°C (measured on the level shifter next to the Pi).

mincebert, 2022-01-09:
Those temperatures fit what I’m seeing, roughly.  My 3A+ @ 1200MHz seems to settle at 58°C and the case’s ambient temperature is about 34-35°C.

I haven’t had a crash since setting arm_freq=1200 when running the test not just using the Master generally.  Same goes for my Electron one, in the separate box.  I also did the “blanket” test, in both cases and got them up to just over 60°C, without incident.

So, I think we’re sorted here, now — thanks, both, for your help!

closed 2022-01-11 by hoglet67