hoglet67/PiTubeDirect #157: PDP11: Hittinng panic() will hang the Pi

hoglet67, 2022-01-10:
Hit this by switching to the PDP11 with out doing a Ctrl-BREAK

For some reason (a bug in 0.28 of the PDP Tube ROM?) this eventually hits a HALT instruction (at 000000).

The instruction hit is 177700 which is diassembled as ??? but executed as HALT. This seems like an inconsistency.

The HALT instruction calls panic()

panic() ends up doing:
```
    cpu.halted = 1;
    printstate();
    while (1);
```
Doing a while (1) like this is bad, as reset is not seen.

Doing this would be better:
```
while (tubeContinueRunning());
```

So three things to look at:
1. Replace  while (1) with while (tubeContinueRunning()
2. Investigate what instructions act as HALT
3. Investigate possible bug in 0.28 of the PDP11 Tube ROM

hoglet67, 2022-01-11:
I've done (1) - probably going to leave the rest to the next release (indigo)

jgharston, 2024-04-28:
On a real PDP11 HALT stalls the CPU until a hardware reset. On emulators such as E11 and SIMH it drops out to the emulator command line usually saying something like:
HALT at 123456
E11>

The instruction 177700 is in the block reserved for floating point instructions, if the F11 is not present it should trigger an Unknown Instruction trap via &0008. The PDP11 Tube Client sets all the unused vectors to NULLIRQ which just returns, so executing 177700 should be "long" null operation. increment PC, trap to 8, stack PC and PSW, vector to NULLIRQ, RTI, pop PC and PSW, continue executing at the next word.