hoglet67/PiTubeDirect #158: 6502 3 MHz issue in Gecko

mcnalu, 2022-01-16:
The symptoms of this issue are identical to those described in #66 in that after *FX 151,230,1 there is no cursor after a ctrl-break and no basic prompt after a break. All other co-pros seem to work just fine, including the default *FX 151,230,0.

I used the latest stable Gecko_fix1 and am using the level shifter board I just received from Retroclinic/Mark with my BBC Model B rev3 with TurboMMC and DNFS ROMs installed. The pi is connected only with a 40pin IDC cable and I checked and reseated all connections and the tube pins on the BBC are in near pristine condition.

I found the same problem with both my Pi 3B+ and Pi 2 and repeated ctrl-breaks never got it to work.

I then tried it with Hognose-rc2 and the problem was gone! Everything I've tried seems to work perfectly in Hognose-rc2 in fact.

hoglet67, 2022-01-16:
Please can you pos a couple of photos of your setup, to save me asking loads of questions.

mcnalu, 2022-01-16:
Voilà

[image]
[image]

RetroClinic, 2022-01-16:
Not sure if it's related, but I have had issues in the past, on the Elite attractor screen, where the occasional line was not removed after the next frame being drawn, leading to a build up of random lines on the screen. This has happened in Gecko sometimes as well, and under HN RC1. The fix before was to change the CPU in the beeb, as it seemed like it was a very slight timing issue with the PHI 1 Pin 3 output. The problem only occurred on the Model B, not the Master.

However, HN RC2 seems to have cleared it, and the CPUs that were causing the problem before are now working perfectly. Was this a known issue beforehand?

hoglet67, 2022-01-16:
> Was this a known issue beforehand?

Not, I've not seen that before Mark.

RetroClinic, 2022-01-16:
> Not, I've not seen that before Mark.

Ok, it didn't happen often, and before I just assumed it was bad timing off one particular CPU, as a swap would fix it. Next time I get a chance, I'll try and replicate it with RC1 and do a video, but as I said, it's fixed in RC2, so you've done something to sort it!

hoglet67, 2023-06-21:
Whatever caused this seems to be fixed in hognose so closing.

closed 2023-06-21 by hoglet67