hoglet67/PiTubeDirect #161: Hang on entering commands in COMAL on copro=24 but not copro=0-3

mincebert, 2022-06-03:
This is with hognose-rc5: I don't run COMAL very often so I'm not sure if it was on rc4 or any other previous revision.  I don't have the frame buffer enabled on either of these machines.

With copro=24 selected, after starting Acornsoft COMAL with *COMAL, type in any command (valid or not) and it hangs, needing a BREAK to recover (the exception seems to be *-commands, which work fine) — "NEW" is a simple one.  This doesn't, however, happen with copro=0-3, which seem to work fine.

I've tried this on a Model B with a Pi Zero 2 W and an Electron with a Pi 3A+ and both behave the same.  This happens straight away after turning on the machine — it doesn't need to be warmed up (just checking as we've been there before).

I've tried unplugging all the ROMs and the problem remains with just BASIC, COMAL and a filing system ROM remaining (doesn't matter which one, but I need one for the Tube Host), including the SWRAM version of MMFS that doesn't raise PAGE.

dp111, 2022-06-03:
Can you provide a link to the version of COMAL you are using ?

hoglet67, 2022-06-04:
Thanks for testing... this is likely an old issue, because nothing in Co Pro 24 has changed fora while, but we'll take a look.

mincebert, 2022-06-04:
I'm using this one: https://mdfs.net/System/ROMs/Language/COMAL/COMAL100 (it doesn't display a version number with *HELP, so I'm guessing there is only one).

I'll try downgrading PiTubeDirect to an earlier hognose rc and see if that changes anything.

Also, to confirm, this is also happening on my Master running MOS 3.50 with a Pi 3A+ and rc5 also.

hoglet67, 2022-06-04:
It happens on Co Pro 24 with all versions of hognose (I tried alpha1, beta1 and rc0)

Just typing HELP and the -> prompt is suffient to hang it.

No clues on the debug console.

Also, it's not a hard crash; the debug console and the VDU continue running (so ARM interrupts are still working fine).

dp111, 2022-06-04:
Just looking at it now

hoglet67, 2022-06-04:
I've generated a trace of of this with the debugger (on Co Pro 16):
[image]

It's ~12,000 instructions:
[comal2.log](https://github.com/hoglet67/PiTubeDirect/files/8837533/comal2.log)

I can't immediately spot anything unusual, and there doesn't appear to be any self-modifying code (i.e. all of code is executing from "ROM").

It does make use of the (ind, X) addressing mode.

I'll be very interested to hear how you approach debugging this Dominic.

hoglet67, 2022-06-04:
It is using the "bit" instruction to skip the next opcode....
```
8979 b0 15    bcs 8990
8990 18       clc
8991 24 38    bit 38
8993 60       rts
```
That's skipping an SEC instruction.

Lots of use of this pattern.

dp111, 2022-06-04:
Here is where I am.

I've enabled jit debugging ( also fixed a bug that had crept into to the
debug code)

below is what the JITer is doing once in COMAL then
*GO 7FFE ( starts debug output )
>HELP

NB the below is only code that hasn't been jitted yet.

tube reset - copro 24
Jit : fa59 : pla
Jit : fa5a : sta EE
Jit : fa5c : sta F2
Jit : fa5e : pla
Jit : fa5f : sta EF
Jit : fa61 : sta F3
Jit : fa63 : pla
Jit : fa64 : rts
Jit : 855f : clc
Jit : 8560 : rts
Jit : 83bd : jmp 838F
Jit : 8b48 : cmp #2E
Jit : 8b4a : bne 8B88
Jit : 8b88 : jsr 8C8E
Jit : 8c8e : lda (2B),Y
Jit : 8c90 : jsr 896B
Jit : 896b : cmp #7B
Jit : 896d : bcs 8992
Jit : 896f : cmp #61
Jit : 8971 : bcs 8990
Jit : 8973 : cmp #5B
Jit : 8975 : bcs 8992
Jit : 8977 : cmp #41
Jit : 8979 : bcs 8990
Jit : 8990 : clc
Jit : 8991 : bit 38
Jit : 8993 : rts
Jit : 8c93 : bcs 8C99
Jit : 8c95 : ora #20
Jit : 8c97 : sta (2B),Y
Jit : 8c99 : rts
Jit : 8b8b : sta 67
Jit : 8b8d : bcc 8BD7
Jit : 8bd7 : lda #FD
Jit : 8bd9 : sta 6D
Jit : 8bdb : lda #7C
Jit : 8bdd : sta 29
Jit : 8bdf : lda #87
Jit : 8be1 : sta 2A
Jit : 8be3 : ldy #FF
Jit : 8be5 : iny
Jit : 8be6 : lda (2B),Y
Jit : 8be8 : jsr 896B
Jit : 8beb : bcs 8BEF
Jit : 8bed : and #DF
Jit : 8bef : eor (29),Y
Jit : 8bf1 : beq 8BE5
Jit : 8bf3 : cmp #80
Jit : 8bf5 : bne 8C3B
Jit : 8c3b : lda (2B),Y
Jit : 8c3d : cmp #2E
Jit : 8c3f : beq 8C07
Jit : 8c41 : dey
Jit : 8c42 : iny
Jit : 8c43 : lda (29),Y
Jit : 8c45 : bpl 8C42
Jit : 8c47 : tya
Jit : 8c48 : sec
Jit : 8c49 : adc 29
Jit : 8c4b : sta 29
Jit : 8c4d : bcc 8C51
Jit : 8c51 : dec 6D
Jit : 8c53 : lda (29,X)
Jit : 8c55 : bne 8BE3
Jit : 8c4f : inc 2A
Jit : 8c51 : dec 6D
Jit : 8c53 : lda (29,X)
Jit : 8c55 : bne 8BE3
Jit : 8c57 : ldy #00
Jit : 8c59 : lda 67
Jit : 8c5b : cmp #2F
Jit : 8c5d : beq 8C6F
Jit : 8c5f : jsr 89C4
Jit : 8c62 : jsr 8C8E
Jit : 8c65 : bcc 8C5F
Jit : 897b : bcc 8992
Jit : 8c67 : jsr 897D
Jit : 897d : jsr 896B
Jit : 8980 : bcc 8990
Jit : 8982 : cmp #5F
Jit : 8984 : beq 8990
Jit : 8c6a : bcc 8C5F
Jit : 8c6c : jmp 8B88
Jit : 8b8f : cmp #2F
Jit : 8b91 : beq 8BD7
Jit : 8b93 : cmp #0D
Jit : 8b95 : beq 8B61
Jit : 8b97 : cmp #26
Jit : 8b99 : beq 8BB5
Jit : 8b9b : jsr 8986
Jit : 8b9e : bcc 8BC1
Jit : 8ba0 : cmp #2E
Jit : 8ba2 : beq 8BC1
Jit : 8ba4 : cmp #22
Jit : 8ba6 : bne 8B83
Jit : 8b83 : jsr 89C4
Jit : 8b86 : ldy #00
Jit : 8b88 : jsr 8C8E
Jit : 89c8 : inc 2C
Jit : 89ca : lda (2B),Y
Jit : 89cc : rts
Jit : 8c6f : jmp 8B83
Jit : 8c07 : lda 6D
Jit : 8c09 : sta (2B,X)
Jit : 8c0b : jsr 8AB2
Jit : 8ab2 : cmp #ED
Jit : 8ab4 : beq 8AB8
Jit : 8ab6 : cmp #EE
Jit : 8ab8 : rts
Jit : 8c0e : bne 8C18
Jit : 8c18 : iny
Jit : 8c19 : lda (2B),Y
Jit : 8c1b : cmp #20
Jit : 8c1d : beq 8C18
Jit : 8c1f : dey
Jit : 8c20 : jsr 89C4
Jit : 8c23 : tya
Jit : 8c24 : clc
Jit : 8c25 : adc 2B
Jit : 8c27 : sta 29
Jit : 8c29 : lda #00
Jit : 8c2b : tay
Jit : 8c2c : adc 2C
Jit : 8c2e : sta 2A
Jit : 8c30 : lda (29),Y
Jit : 8c32 : sta (2B),Y
Jit : 8c34 : iny
Jit : 8c35 : cmp #0D
Jit : 8c37 : bne 8C30
Jit : 8c39 : beq 8C72
Jit : 8c72 : dey
Jit : 8c73 : lda 6D
Jit : 8c75 : cmp #CE
Jit : 8c77 : bcc 8C7D
Jit : 8c79 : cmp #D5
Jit : 8c7b : bcc 8C88
Jit : 8c7d : cmp #8C
Jit : 8c7f : beq 8C8B
Jit : 8c81 : cmp #93
Jit : 8c83 : beq 8C8B
Jit : 8c85 : jmp 8B86
Jit : 8bf7 : lda (2B),Y
Jit : 8bf9 : jsr 897D
Jit : 8bfc : bcs 8C07
Jit : 8bfe : iny
Jit : 8bff : lda (2B),Y
Jit : 8c01 : jsr 897D
Jit : 8c04 : dey
Jit : 8c05 : bcc 8C43

Then appears to hang.

In the disassembly of COMAL there is quite a lot of ldx #&FF TXS which
could perhaps upset things  ( the one at &838F) is of interest

hoglet67, 2022-06-04:
What did you type in at the -> prompt?

dp111, 2022-06-04:
*GO 7FFE
HELP

hoglet67, 2022-06-04:
I understand I need to be areful interpretting your trace, as it represents the first time each instruction is hit (when it's jitted).

The first point of divergence looks like:
```
Jit : 8b8f : cmp #2F
Jit : 8b91 : beq 8BD7
Jit : 8b93 : cmp #0D
Jit : 8b95 : beq 8B61
Jit : 8b97 : cmp [#26]   <<<<< I never hit 8b97
Jit : 8b99 : beq 8BB5
Jit : 8b9b : jsr 8986
```
My trace (also of HELP) contains 8ab8 (rts) but not 8c0e
```
8b8f c9 2f    cmp #2F
8b91 f0 44    beq 8BD7
8b93 c9 0d    cmp #0D
8b95 f0 ca    beq 8B61
8b61 c6 2c    dec 2C
8b63 e6 2b    inc 2B
```
I think 8b97 is the first instruction in your trace that isn't present in mine.

hoglet67, 2022-06-04:
Actually, looking right at the start or your trace, there is 855f clc - I never hit this instruction.

Can you get a more complete list of every jitted instruction?

dp111, 2022-06-04:
This is what I did
*go 7FFE
*COMAL
HELP

these are just address as the disassembly is too slow.

Jit : fa59
Jit : fa5a
Jit : fa5c
Jit : fa5e
Jit : fa5f
Jit : fa61
Jit : fa63
Jit : fa64
Jit : 8fae
Jit : 8fb0
Jit : 8fb2
Jit : 8fb3
Jit : 8fb4
Jit : 8fb6
Jit : 8fb8
Jit : 8fbb
Jit : 8fc1
Jit : 8fc3
Jit : 8fc5
Jit : f9f0
Jit : f9f2
Jit : f9f5
Jit : fa20
Jit : fa22
Jit : fa25
Jit : f9c7
Jit : f9c9
Jit : f9cc
Jit : f9ce
Jit : f9d0
Jit : f9d3
Jit : f9d4
Jit : f9d6
Jit : f9d8
Jit : f9da
Jit : fa28
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
Jit : fdec
Jit : fdef
Jit : fdf0
[... 489 lines cut ...]
Jit : 8c30
Jit : 8c32
Jit : 8c34
Jit : 8c35
Jit : 8c37
Jit : 8c39
Jit : 8c72
Jit : 8c73
Jit : 8c75
Jit : 8c77
Jit : 8c79
Jit : 8c7b
Jit : 8c7d
Jit : 8c7f
Jit : 8c81
Jit : 8c83
Jit : 8c85
Jit : 8bf7
Jit : 8bf9
Jit : 8bfc
Jit : 8bfe
Jit : 8bff
Jit : 8c01
Jit : 8c04
Jit : 8c05

hoglet67, 2022-06-04:
Here's a reference trace processed to show the first hit of each instruction:
[ref.txt](https://github.com/hoglet67/PiTubeDirect/files/8837814/ref.txt)
Here's your trace above:
[dp11.txt](https://github.com/hoglet67/PiTubeDirect/files/8837815/dp11.txt)
It's useful to diff them with kdiff3.

I think it's safe to ignore differences in the client ROM.

8993 is jitted twice because of the jumping into the middle of a BIT instruction, which I think the jitter handles correctly.

Any idea why 8c51 is jitted twice?

First significant difference is my reference trace includes 898a; yours doesn't. Then soon after there is a lot of divergence.

dp111, 2022-06-04:
Thanks for the traces

8c51 gets jitted twice because the first time there is a jump to 8c51
second time it is 8c4f which flows into 8c51 jitting only stops when a
branch /jump/ rts/rti occurs. Its not worth checking if an instruction has
already been jitted as it is very to rejit and most of the time
rejitting doesn't occur.

hoglet67, 2022-06-04:
Can you force somehow every instruction to be re-jitted, so you then get a complete trace?

dp111, 2022-06-04:
Unfortunately not. When code gets jitted the arm instruction overwrites the
table which contains branches to the jitting code

hoglet67, 2022-06-04:
What about dumping the A, X, Y, S and P in the jit log?

Then we can see when the machine state diverges.

hoglet67, 2022-06-04:
So really the only atypical pattern here is the use of BIT <zp> to skip a 1-byte opcode.

Looking at the code I see this TODO flagged:
```
// Instructions that get used in smaller chunks to save a jump ( overlapping )
// BIT &0000 so B on first byte
// BIT &00 //TODO ?
// NOP6s &0000
// NOP6s &00
```
Is it possible the Jitter copes with this pattern with BIT <absolute> but not bit <zp>?

dp111, 2022-06-04:
> So really the only atypical pattern here is the use of BIT to skip a
> Is it possible the Jitter copes with this pattern with BIT but not bit ?

I think you have it , let me play. I was just adding the start of a
debugger.

dp111, 2022-06-04:
I've pushed a fix to the indigo branch.

hoglet67, 2022-06-04:
I'd really like to include this fix in hognose, because I suspect it's a a pattern used quite widely at Acornsoft.

Looking at the fix on indigo-dev there are changes to JITDEBUG as well as to the BIT instruction. Are they a necessary part of this as well?

Also, there seems to be a mix of #if JITDEBUG and #ifdef JITDEBUG. Is this intentional?

dp111, 2022-06-04:
I think it should be on hognose too.

The JITDEBUG stuff fixes a bug that has been in for a while . also JITDEBUG
is now always #if and not #ifdef

hoglet67, 2022-06-05:
Dominic, I've cherry-picked 26b4bbb4 into hognose-dev and it seems to resolve this issue.

Are you happy with that as the fix?

Dave

dp111, 2022-06-05:
I'm happy

mincebert, 2022-06-05:
Thanks -- that seems to have fixed it for me (at least briefly testing)!

As an aside, there appears to be a bug in src/framebuffer/teletext.c at line 298 introduced in commit 1db0a41485bf8ce9b55342953922852abc754132 preventing it from compiling — there's a 'default' in a switch block but without any block after it.  I think the 'default:' needs removing or a 'break;' adding, if you want to leave it there

hoglet67, 2022-06-05:
I've just pushed a fix for that compile error.

What version of GCC ARM are you using?

I don't see any errors with 11.2

Dave

mincebert, 2022-06-05:
I’m using 10.2.1, compiling on Raspberry Pi OS 11, so a little older.  I’ve no idea what the C standard has to say about this — my knowledge of that froze in about 1994!

Anyway it compiles OK now and COMAL seems fine, thanks!

hoglet67, 2022-06-08:
Fix included in Hognose RC6 release.

closed 2022-06-08 by hoglet67