hoglet67/PiTubeDirect #162: 186 emulation.

stuaxo, 2022-06-09:
186 emulation would be interesting on this (and I guess from there it might be possible to emulate x86 processors that never saw the tube).

hoglet67, 2022-06-09:
The existing x86 Co Pro uses Fake86 which already includes all the 80186 specific instructions.

Do you have any examples of programs that fail to run, which work on the original Acorn 80186 Co Processor?

Are you aware of any other emulation issues?

stuaxo, 2022-06-09:
Nope, apologies it's aspirational for me to get a Beeb and PiTube at the moment... I'll definitely give it a go when I do though, I guess emulation opens up the possibility of running things that didn't run on the original, like apps for the 286 for instance.

closed 2022-06-09 by stuaxo

hoglet67, 2022-06-10:
No need to apologise - I just wanted to check that you hadn't hit a specific issue, as we are very close to a new release. If you are interested in the BBC Micro, it's definitely worth joining the startot forums:
https://stardot.org.uk/forums/

stuaxo, 2022-06-10:
Cheers - I do go there and lurk from time to time, not as often as I'd like :)