hoglet67/PiTubeDirect #165: Debugging

zolbatar, 2022-10-29:
I'm working on my JIT code and getting odd behaviour writing to some registers. I'm presuming some error is being raised it freezes.

Is there a way to set the HDMI output to display the console (rather than PiVDU) so I can see errors without plugging anything else in?

hoglet67, 2022-10-29:
> I'm working on my JIT code and getting odd behaviour writing to some registers. I'm presuming some error is being raised it freezes.

Which registers?

> Is there a way to set the HDMI output to display the console (rather than PiVDU) so I can see errors without plugging anything else in?

Sorry, this is not currently possible. The only way to view the console is via the serial port.

zolbatar, 2022-10-29:
I'm not sure which one I have, I'll take a look later. It came all prebuilt anyway. I don't honestly remember seeing a pin header though.

I think the register is most irrelevant, as I think the problem is elsewhere but I just get a lockup so it's difficult to diagnose. The wonders of doing JIT/AOT stuff!

closed 2023-06-21 by hoglet67