hoglet67/PiTubeDirect #169: GCC patches for PDP-11 backend issues?

gameblabla, 2023-03-23:
I was reading this [PDP 11 Co Pro Notes](https://github.com/hoglet67/PiTubeDirect/wiki/PDP-11-Co-Pro-Notes#chapter-3---compiling-with-the-gcc-pdp-11-cross-compiler), which was very interesting and helped me setup a cross compiler for PDP11.
However you mentioned how you managed to fix several GCC bugs but it's not exactly clear how you did it and you didn't send them upstream so i was wondering if you could explicitely share them ?
I encountered some issues with GCC myself with code generation and stuck with O0 for now.

hoglet67, 2023-03-23:
There was some discission on the Retro Computing Forum with Paul Koning (who maintains the PDP11 backend) about including these fixes: 
https://retrocomputingforum.com/t/modern-pdp-11-c-compilers/2329/19?u=hoglet

Diffs for the two changes I made are included in that thread.

I'm not sure if Paul ever pulled these changes in, but it would be in the gcc v13 development branch. I just did some digging, and could not find anything.

Dave

gameblabla, 2023-03-23:
Yeah it doesn't seem like he went through with the fixes, a shame. Perhaps i should consider opening up a bug report... Thanks for the patches !