hoglet67/PiTubeDirect #170: Frame Buffer with Native ARM copro: stuck when accessing filing systems

mincebert, 2023-06-10:
I have a Pi 3A+ running Hognose final release in a Master with the frame buffer enabled.

When I have the 6502 coprocessor enabled I can access filing systems without any trouble, even if I enable the frame buffer with `CALL &300`, it all works fine.

If, however, I switch to the Native ARM coprocessor (number 15) with the frame buffer enabled with `*PIVDU 2`, the screen just scrolls printing blank lines.  This does NOT occur when only using the native BBC video or, oddly, with `*PIVDU 3`.  If I type something like `*HELP COPROS`, I get the normal output but then things scroll off the screen quickly with blank lines.

I have various filing systems installed but I get the same results with MMFS (the Master version), ANFS and ADFS.

Otherwise the system works fine, as long as I stay away from the filing system (for example, running a program in `*ARMBASIC`).

hoglet67, 2023-06-10:
I can reproduce this and know what's going on...

I have no idea how we missed this, as I think it was introduced in hognose-rc5, three months before we released!

I've just uploaded an new release that should address this:
https://github.com/hoglet67/PiTubeDirect/releases/tag/hognose-fixes

It's currently tagged as a pre-release, as there is one more (unrelated) thing I want to add.

Please me me know how you get on....

Dave

mincebert, 2023-06-10:
Thanks for the usual prompt service!  That certainly fixes the problem I was having — I can't comment on the other two changes!

hoglet67, 2023-06-10:
Fixed in hognose-fixes release

closed 2023-06-10 by hoglet67