hoglet67/PiTubeDirect #172: ARM Native copro *DIS command calculates negative offset ADR addresses incorrectly?

mincebert, 2023-06-20:
I'm not an expert on ARM but I think the *DIS command in the ARMv7 Native Coprocessor calculates the address used in an ADR instruction incorrectly, when the offset is negative.

For example, if I assemble (in BBC BASIC V) at &9018:

```
.data
EQUW &87654321
ADR R1,data
```

Then I do `*DIS 9018`, I get:

```
>*DIS 9018
00009018 87654321 STRBHI r4, [r5, -r1, LSR #6]!
0000901C E24F100C ADR r1, &9030
```

... the ADR instruction is a SUB with an offset of &C, minus 8 for the pipeline so should be &9018; I think the offset is always being added and not subtracted, when the instruction is determined to be an ADR.

By comparison, `*MEMORYI` on RISC OS gives (assembled at &901C):

```
>*MEMORYI 901C
0000901C : !Ceá : 87654321 : STRHIB  R4,[R5,-R1,LSR #6]!
00009020 : ..O‚ : E24F100C : ADR     R1,&0000901C
```

... same encoding but the address is correctly shown as &901C (in this case).

hoglet67, 2023-06-21:
Disassembling this with [an Online Disassembler](https://shell-storm.org/online/Online-Assembler-and-Disassembler/?opcodes=87654321%0D%0AE24F100C&arch=arm&endianness=big&baddr=0x00009018&dis_with_addr=True&dis_with_raw=True&dis_with_ins=True#disassembly) I get:
```
0x0000000000009018:  87 65 43 21    strbhi r4, [r5, -r1, lsr #6]!
0x000000000000901c:  E2 4F 10 0C    sub    r1, pc, #0xc
```
But I agree there is something wrong here.

dp111, 2023-06-21:
The issue is the disassembler is using banch calculations in the table
[I_ADR] = {"cdb"}," in armv7-tbl.c which is only good for forward address
calculations.

dp111, 2023-06-21:
So I think in the 'b' case in darm.c ( line 463)  the d->U flag needs to be
used to select if the (8+ d->imm) is subtracted instead of added.

hoglet67, 2023-06-21:
OK, I'll fix this now in hognose-fixes and merge across to indigo-dev.

Dominic, can I also merge indigo-dev-dp back to indigo-dev?

dp111, 2023-06-21:
Yes I'm happy for indigo-dev-dp to be merged into indigo-dev

hoglet67, 2023-06-21:
OK, this should now be fixed in hognose-fixes:
https://github.com/hoglet67/PiTubeDirect/releases/tag/hognose-fixes

closed 2023-06-21 by hoglet67