hoglet67/PiTubeDirect #174: Frame Buffer: COPY cursor reverts to host screen after running TWIN

mincebert, 2023-07-02:
If using the ARM Native copro and having the frame buffer selected with `*PIVDU 2`, after running the TWIN editor (from the ARM Evaluation discs), moving the cursor does not appear to do anything but, when pushing COPY, seems to copy from the host screen.  On the host screen, a block is printed but you can't see the copy cursor.

Running `*PIVDU 2` fixes things again.

I suspect there's probably nothing you can do about this and it's just TWIN mucking about with the vectors, but I thought I'd report it!

hoglet67, 2023-07-03:
Does *FX 4,1 also fix it?

mincebert, 2023-07-03:
Yes, it does and everything is fixed: the copy cursor moves around on the Pi frame buffer and copies from it.

hoglet67, 2023-07-03:
I guess the VDU driver could be much more aggressive in doing *FX 4,1 every line, but that seems a bit excessive.

So I'm inclined to leave it as it is.

Dave

hoglet67, 2023-09-06:
Leaving as is....

closed 2023-09-06 by hoglet67