hoglet67/PiTubeDirect #176: Native ARM crashes when CALLing OSCLI from ARMBASIC if the string is a HELP command

beebmaster, 2023-09-04:
As here:
[image]

There appears to be a case, after a reset, where an error is returned instead:
[image]

RetroClinic, 2023-09-05:
I noticed this too, but wasn't it fixed with the Hognose update?

hoglet67, 2023-09-05:
Mark, I don't actually recall this issue, or fixing it, but it seems you are right.

I can reproduce this with Gecko, but not with Hognose or Indigo. 

BeebMaster, can you confirm what version of PiTubeDirect you are running?

beebmaster, 2023-09-05:
Yes, you are quite right!  It works in Hognose, I just tested on Master 128 Station 112, which has a fairly new Pi Tube, with internal Retroclinic adapter, and it works.

Station 201 however is till on Gecko, where the problem exists.  The trouble there is that I have one of Ken's external adapters plugged into the Tube and it's stuck so I can't get it out to update the SD card!

hoglet67, 2023-09-06:
Fixed in Hognose release.

closed 2023-09-06 by hoglet67