hoglet67/PiTubeDirect #204: Issue with 2022 version RPi 3A+

RetroClinic, 2023-10-13:
Hi.

I've taken delivery of some new RPi 3A+, and these seem to be slightly different, having a (C)2022 instead of (C)2018, and have some slight differences on the PCB.

[image]

However, none of the new ones will boot the firmware? I get the usual single flash of the green LED, followed by 4 long flashes and 7 quick flashes. I presume this is an error code of some description?

I've looked through the issues and new files, but can't seem to see any reference to an issue like this arising so far? I've tried it with the latest Indigo and it's doing the same thing on the 2022 version, but works fine on the 2018 one?

Thanks as always, Mark.

hoglet67, 2023-10-13:
I'm not aware of a new 3A+ model, maybe Dominic is?

The 4-long 7-short error code is "Power Failure Type B":
https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes

Is the 3.3V output from the Pi present?

Can you boot Raspberry Pi OS using an official PSU?

RetroClinic, 2023-10-13:
Ok, that's a worrying error, as all of them (tested 8 so far) are doing the same thing. There is 3.31v where it should be, so that seems ok.

I'll try booting the Pi OS a bit later, thanks.

dp111, 2023-10-13:
The firmware does need to be updated to support this board. This should go
into the indigo release.

hoglet67, 2023-10-13:
Seems there is a new revision of the 3A+
[PCN-Raspberry-Pi-3A-Revision-2-PCB.pdf](https://github.com/hoglet67/PiTubeDirect/files/12892817/PCN-Raspberry-Pi-3A-Revision-2-PCB.pdf)

hoglet67, 2023-10-13:
OK, I was planning a new indigo alpha shortly any way.

Any suggestion of a specific firmware revision, or should I just grab the latest?

Dominic, I'll merge you branch as well.

Dave

dp111, 2023-10-13:
Just the latest should be fine.

RetroClinic, 2023-10-13:
Yes, latest version of RPi OS boots just fine, so hopefully you guys have got it in hand, as always! I'll put them aside and wait for the new release and let you know.

Thanks. Mark.

hoglet67, 2023-10-13:
Here's an updated Indigo build that includes the latest firmware blob:
https://github.com/hoglet67/PiTubeDirect/releases/tag/indigo-alpha4

Let us know if this resolves the issue.

closed 2023-10-18 by hoglet67