hoglet67/PiTubeDirect #207: Native ARM: framebuffer *PIVDU "Bad command" in indigo-alpha5

mincebert, 2024-03-02:
Thanks for fixing the CoPro 24 framebuffer MODE crash but, sorry to bother you again, there's a regression in indigo-alpha5...

Switching to the ARM Native CoPro 15 and executing `*PIVDU 2` works, and the new status of the Beeb and Pi VDU drivers is displayed, but then you always get a `Bad command` error.

Again, MOS 3.50, Pi 3A+ with vdu=1, filesystem is ADFS but I get the same error in MMFS.

dp111, 2024-03-02:
should be fixed on Indigo-dev-dp. Also fixed another stack alignment issue

hoglet67, 2024-03-02:
Thanks for the quick fix Dominic, and appologies for missing this on testing yesterday (I missed the stack alignment fixes also covered the Native ARM).

Here's an alpha6 release including the Bad Command fix:
https://github.com/hoglet67/PiTubeDirect/releases/tag/indigo-alpha6

closed 2024-03-02 by hoglet67