hoglet67/PiTubeDirect #211: Indigo release incompatible with certain Pi 1 models

hoglet67, 2024-05-26:
This was picked up by JHG and replicated by myself.

The Indigo release doesn't work properly on certain Pi 1 models.

Here's a list of all known Pi 1 Models:
 [image]

It works correctly on Revision 0002 (the original Pi B 1.0 256KB model)

Ir doesn't work correctly on Revision 000F (the {Pi B 2.0 512KB model)

I think the Pi B 2.0 has a slightly different
 GPIO pin assignment, see:
https://www.raspberrypi-spy.co.uk/2012/06/simple-guide-to-the-rpi-gpio-header-and-pins/

Here are some boot logs:
[board_2.txt](https://github.com/hoglet67/PiTubeDirect/files/15447175/board_2.txt)
[board_f.txt](https://github.com/hoglet67/PiTubeDirect/files/15447176/board_f.txt)

Looking at the DEBUG_GPU stuff at the start, R3 is the same in both cases, and it shouldn't be.

I think this commit broke things: b487abbb

I'm going to attempt to fix this now.

hoglet67, 2024-05-26:
That commit also broke the activity LED on the Pi 1 models.

It  relies on the firmware inserting the bcm2709.disk_led_gpio property into the command line properties.

That doesn't seem to work on the Pi 1 Models I have.
```
      BOARD_REVISION : 00000002
            CMD_LINE : bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbdepth=16 dma.dmachans=0x7ff5 bcm2708.boardrev=0x2 bcm2708.serial=0xb27317b4 bcm2708.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:73:17:B4 vc_mem.mem_base=0xfa00000 vc_mem.mem_size=0x10000000  copro=24 copro1_speed=3 copro3_speed=4 tube_delay=0 elk_mode=0 vdu=1
```
```
      BOARD_REVISION : 0000000f
            CMD_LINE : bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbdepth=16 dma.dmachans=0x7ff5 bcm2708.boardrev=0xf bcm2708.serial=0x669b0030 bcm2708.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:9B:00:30 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  copro=24 copro1_speed=3 copro3_speed=4 tube_delay=0 elk_mode=0 vdu=1
```
Not sure if this is a regression with the newer RPI firmware blob we are using, or whether this has never worked. 

I'll fix this as well.

hoglet67, 2024-05-26:
Dominic, do you recall which original Pi models were not working, that b487abb fixed? I want to make sure I haven't caused any further regressions.

dp111, 2024-05-26:
I don't recall. I'm afraid. Sorry about the regression.

Dominic

closed 2024-06-16 by hoglet67