hoglet67/PiTubeDirect #212: Emulated BBC Micro on one Pi, with Emulated Tube, to talk to PiTubeDirect processors on a second Pi?

gtoal, 2024-07-13:
Has anyone modified any BBC Micro emulator that runs on a Pi, so that a second processor emulation could talk to it by wiring a GPIO to GPIO cable (ie no level shifters between the two Pies - just wires) and allow an emulated second processor to talk to an emulated Beeb over a virtual Tube, so we could enjoy these emulations even if we don't own a real and working BBC Micro?

I think all that would be needed would be for a BBC Micro emulator to implement the Tube interface by talking to the GPIO ports directly, much as say the 6502 second processor will be doing for the other side of the Tube interface already.

Those of us in the US where it's hard to find a working Beeb would really appreciate a system like that if it exists and I wonder how much work would it entail for a BBC Micro emulator author to modify an existing emulator to do this, if it doesn't already exist?

Graham

hoglet67, 2024-07-13:
Hi Graham,

Sorry, I'm not aware of a software emulator that does this., I don't think anyone has developed a BBC Emulator that is adapted specifically for the Raspberry Pi.

The nearest thing I can think of is the Spectrum Next port of BeebFPGA. This supports a directly connected PiTubeDirect (i.e. no level shifters) running on the in-build Pi Accelerator:
https://github.com/hoglet67/BeebFpga/wiki/Spectrum-Next#co-processor-support

Dave

dp111, 2024-07-13:
The closest thing I know of :
https://stardot.org.uk/forums/viewtopic.php?t=21985

Dominic